espressif/esp-idf #6310: Build errors due to missing dependencies in IDF components

boarchuz, 2020-12-22:
## Environment

- Module or chip used:  ESP32
- IDF version: master (v4.3-dev-2136-gb0150615d)
- Build System: CMake
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- IDE: VSCode + Espressif extension

## Problem Description

Some IDF components are missing dependencies (ie. not set in REQUIRES/PRIV_REQUIRES) causing "No such file or directory" compiler errors. The log errors below show a few I've encountered.

## Debug Logs

> /home/matt/Espressif/IDF/master/esp-idf/components/log/log_noos.c:17:10: fatal error: soc/cpu.h: No such file or directory
>  #include "soc/cpu.h"  // for esp_cpu_get_ccount()

log requires esp_hw_support

> /home/matt/Espressif/IDF/master/esp-idf/components/bootloader_support/src/bootloader_utility.c:61:10: fatal error: soc/cpu.h: No such file or directory
>  #include "soc/cpu.h"

bootloader_support requires esp_hw_support

> /home/matt/Espressif/IDF/master/esp-idf/components/esp_hw_support/port/esp32/rtc_clk.c:30:10: fatal error: hal/gpio_ll.h: No such file or directory
>  #include "hal/gpio_ll.h"

esp_hw_support requires hal

Alvin1Zhang, 2020-12-23:
Thanks for reporting and sorry for inconvenience, we will look into.

renzbagaporo, 2021-01-19:
Hi @boarchuz , by default all components depend on `esp_hw_support` and `hal`, so these are not expressed explicitly in the component CMake file. Internally, we have not encountered such issues. Is this on one of the examples on your local ESP-IDF copy?

boarchuz, 2021-01-20:
@renzbagaporo I didn't realise that, thanks. This was in a large project which is now building successfully (without above changes) so I'm not sure why I was seeing those errors earlier. I'll reopen if it happens again.

closed 2021-01-20 by boarchuz

TGiles1998, 2022-01-13:
This has happened to me when compiling a large project: https://github.com/mocleiri/tensorflow-micropython-examples/issues/60

Any way to prevent this?

Thank you!