espressif/esp-idf #6311: Download of tools extreamly slow

Hero9909, 2020-12-22:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version :v4.3-dev-2136-gb0150615d
- Build System:         not applicateable
- Compiler version : unfineshed installation
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

the download is extreamly slow(50-80kbps) while other progressing with 4/mbps

### Expected Behavior

faster?

### Actual Behavior

see above

### Steps to reproduce

1. run install.sh

// If possible, attach a picture of your setup/wiring here.
[image]

KaeLL, 2020-12-22:
I've had that happen to me multiple times as well, but it's a rather intermittent issue.

Hero9909, 2020-12-22:
Correct and I've to admit that this only occurs on one of my oldest installations of ubuntu. Currently, I can reproduce this issue consistently. Anything you would like me to check?

KaeLL, 2020-12-22:
I just assumed it was a normal thing and moved on. I think the espressif people will say something similar, but who knows..

GuyMinibrew, 2020-12-23:
Usually it took a minute to get all the tools, now my download is never complete (tried 10 times in the last 24 hours), I can't compile anything.

Hero9909, 2020-12-23:
I've done some research. The result is that the server is the bottleneck. I've tried to download the files via get and curl on Android (termux), with mobile connection. And on 2 ubuntu towers. Same result even in different networks. Limit is at 80kbps. Therefor this might be a case for espressif support and not a issues related to the script

igrr, 2020-12-24:
@Hero9909 Sorry for the inconvenience. We are moving the tool download URLs to Github Releases, which will provide faster downloads. We will update this issue once these changes are released.

Hero9909, 2021-03-06:
hi, its been a few month and ive done a install yesterday without issues. could you confirm that your changes are done?
if not i would be happy to hear some little infos about the progress.

igrr, 2021-03-07:
@Hero9909 Some changes have been done: release/v4.3 and later branches download most tools from Github Releases, not dl.espressif.com. We are also moving dl.espressif.com to a hosting with better availability, but this is not finished yet. We will update this issue when the process is complete.

closed 2021-12-14 by tom-borcin