espressif/esp-idf #6314: BLE Mesh network parameters support

mrzant, 2020-12-23:
Hi there,

I am currently developing a BLE Mesh application with latest ESP-IDF stable version (4.2). However, I would like to know where is it possible to add a random delay (0-10 ms) between consecutive transmissions within an advertising event as I am trying to analyze its effect on packet loss and average delay.

The current implementation allows for changing network and publish retransmission interval and counts which is working pretty fine. However, regarding relay retransmissions, many articles propose adding a random back-off time to avoid collisions between relays, which unfortunately is not implemented or I could not find it. As a temporary solution, I was able to add it in the `bt_mesh_net_relay()` function defined at the network layer.

Another interesting feature that could be added from your side is an option for changing scanning interval and window duration. For now, I could only change it statically with `MESH_SCAN_INTERVAL` and `MESH_SCAN_WINDOW` which are defined in `adv.c`.

Thank you in advance for any help. 

Cheers :)

Alvin1Zhang, 2020-12-24:
Thanks for raising this feature request.

InfiniteYuan, 2020-12-28:
Thanks for your feedback.

Current, The random delay is done by the bottom layer. Relay also includes random delay.

mrzant, 2020-12-30:
Hi @InfiniteYuan,

Thank you for your prompt response. 

I have not observed that random delay is added anywhere. However,  based on a pre-configured number of retransmissions and interval, gap "mesh" advertisement duration and interval is defined as shown in the snippet below from `mesh_core/adv.c`. This is further handled by mesh bearer. 

```c
static inline int adv_send(struct net_buf *buf)
{
...
    BT_DBG("type %u len %u: %s", BLE_MESH_ADV(buf)->type,
        buf->len, bt_hex(buf->data, buf->len));

#if CONFIG_BLE_MESH_SUPPORT_BLE_ADV
    if (BLE_MESH_ADV(buf)->type != BLE_MESH_ADV_BLE) {
#endif

        adv_int = MAX(adv_int_min,
                      BLE_MESH_TRANSMIT_INT(BLE_MESH_ADV(buf)->xmit));

        duration = (BLE_MESH_TRANSMIT_COUNT(BLE_MESH_ADV(buf)->xmit) + 1) *
                   (adv_int + 10);

        BT_DBG("count %u interval %ums duration %ums",
               BLE_MESH_TRANSMIT_COUNT(BLE_MESH_ADV(buf)->xmit) + 1, adv_int,
               duration);
...
}
```

Therefore, I would appreciate if you could be more specific where exactly, a random delay is added. Regarding randomization between channel advertisement intervals, this probably should somehow be added from the controller point of view, correct me if I am wrong. Just to be on the same page, I am using NimBLE host rather than Bluedroid as it is more appropriate for my case.