espressif/esp-idf #6315: esp_wifi_get_country() returned/set wifi_country_t structure doesn't follow the documented behaviour

BennyE, 2020-12-23:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-S2 / featherS2 (UnexpectedMaker)]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-2136-gb0150615d
- Build System:         [CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I was (again) working on adding the "country code" to CircuitPython, including a workaround to fix #6267 for the time being and I found that the wifi_country_t struct of esp_wifi_get_country() does not work as expected (and not as documented).

According to the documentation wifi_country_t.cc would be [3] == "CC\0" or "DE\0", but I found it to be a length of 6 instead.

The corresponding pull request is here: https://github.com/adafruit/circuitpython/pull/3868

I found my Python Object to be the following:

```python
>>> wifi.radio.ap_info.country
'DE \x01\r\x14'
```
I captured a Beacon of my Alcatel-Lucent Enterprise Stellar Wireless Access Point and suddenly the above looks familiar:
[image]

D (0x44) E (0x45) " " (0x20 Whitespace, which is actually "ANY Environment") -> conclusion: .cc includes the full wifi_country_t element instead of just the Country Code.

### Expected Behavior

esp_wifi_get_country() should fill the wifi_country_t struct as documented where the .cc struct member should be "CC\0".

### Actual Behavior

esp_wifi_get_country() creates a wifi_country_t struct where the .cc struct member has a length of six instead of three.

### Steps to reproduce

To ensure that I didn't do something wrong in CircuitPython, I have simply done the same in your espressif/esp-idf example code. See below

```cpp
// esp-idf -> examples -> getting_started -> station -> code
// with the following additions
#include "esp_wifi_types.h"

// inserted somewhere before main()

void string2hexString(char* input, char* output)
{
    int loop;
    int i; 
    
    i=0;
    loop=0;
    
    while(input[loop] != '\0')
    {
        sprintf((char*)(output+i),"%02X", input[loop]);
        loop+=1;
        i+=2;
    }
    //insert NULL at the end of the output string
    output[i++] = '\0';
}

// .. inserted after this ..
//    } else {
//        ESP_LOGE(TAG, "UNEXPECTED EVENT");
//    }

    memset(&apinfo, 0, sizeof(apinfo));
    esp_wifi_sta_get_ap_info(&apinfo);
    esp_wifi_get_country(&apinfo.country);

    int len = strlen(apinfo.country.cc);
    char hex_str[(len*2)+1];
    
    //converting ascii string to hex string
    string2hexString((char *)apinfo.country.cc, hex_str);    

    ESP_LOGI(TAG, "RSSI: %d", apinfo.rssi);
    ESP_LOGI(TAG, "Country (CC) strlen(): %d", strlen(apinfo.country.cc));
    ESP_LOGI(TAG, "Country (CC): string %s", (char *)apinfo.country.cc);
    ESP_LOGI(TAG, "Country (CC): hex 0x%s", hex_str);
    //ESP_LOGI(TAG, "Country: %s", (char*)&apinfo.country);
    ESP_LOGI(TAG, "Channel: %d", apinfo.primary);
    ESP_LOGI(TAG, "SSID: %s", apinfo.ssid);

```

## Debug Logs

Please note that Country (CC): string works, but the length is wrong and when displayed as hex it becomes evident that this is the full wifi_country_t struct element.
 
```
I (4449) wifi station: RSSI: -48
I (4449) wifi station: Country (CC) strlen(): 6
I (4459) wifi station: Country (CC): string DE 
I (4459) wifi station: Country (CC): hex 0x444520010D14
I (4469) wifi station: Channel: 1
I (4469) wifi station: SSID: Stellar-DSPSK
```

YouDONG-ESP, 2020-12-24:
Hi @BennyE, strlen() will start to count till it hit a '\0'. The country code is actually 'DE ' instead of 'DE\0' so it will count to `policy`  witch is eaqual to 0 and let it return 6. You can check the packet you captured, 44 45 20 is 'DE**space**'. I just checked the protocol, that's the correct form.
Meanwhile,we found that our softap generate a country code with 'CN\0' instead of 'CN '. So that's a bug for us, we will fix it, thanks.

BennyE, 2020-12-24:
Hi @YouDONG-ESP, thanks for your helpful reply. You are correct, thanks again that you took the time to share the feedback with me. I had a look at the "dot11CountryString OBJECT-TYPE" in IEEE 802.11d-2001 and also at the espressif documentation on country code [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-country-code) - I didn't know and learned something, what a good day :).

closed 2020-12-24 by BennyE