espressif/esp-idf #6316: protocomm_httpd randomly maps to incorrect session_id during SoftAp provisioning

william-ferguson-au, 2020-12-25:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32s2
- IDF version: v4.3-dev-2136-gb0150615d
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: CLion but building and launching via cmd line
- Power Supply:         Battery and/or external

## Problem Description

Trying to configure using SoftAp provisioning. 
Have built and flashed the SoftAp provisioning example and it successfully launches the AP and waits for the prov client. 

I use the ESP-IDF Android SoftAp Android app from Google Play (and I have built latest from source and tried that too). It rarely ever completes provisioning. It randomly falls over during one of the provisioning steps. Looking at the ESP-IDF side it fails because protocomm security for the HTTP session is lost. 

Looking closer and it becomes clear that the same session_id is not always being returned (protocomm_httpd.c line 54) for requests from the provisioning client. And if that happens after protocomm_security has been configured (and is expected on the session) then the next provisioning request will fail.

### Expected Behavior

All requests from the same provisioning client would be handled within the same HTTP session. and the provisioning would deterministically complete.

### Actual Behavior

Provisioning behaviour is non-deterministic but almost always fails because one of the prov requests gets given a different session_id

### Steps to reproduce

1. ESP-IDF Android SoftAp Provisioning app
2. Click on Provision new device
3. Click on I don't have a QR code
4. Click on Connect
5. When the Wifi Assistant is displayed, I select the AP published by the ESP device
6. Click on the back button
7. In the Proof of Possession page enter 'abcd1234'; and click Next
8. On the Select Wifi Network page note "Failed to get WifiScan-List" Toast and empty list once the wait icon disappears.
9. If you manage to get this far and it hasn't failed then click on the Provision button and note the failure to provision

### Code to reproduce this issue

wifi_prov_mgr example

To make the change in session_id clear I added the following to line 54 of protocomm_httpd.c
```cpp
    int cur_session_id = httpd_req_to_sockfd(req);
    ESP_LOGI(TAG, "common_post_handler uri=%s session_id=%d req_session_id=%d", req->uri, session_id, cur_session_id);
```

## Debug Logs

When it fails it will eventually end up with something like this:
```
I (33480) protocomm_httpd: common_post_handler uri=/proto-ver session_id=-1 req_session_id=57
I (39550) protocomm_httpd: common_post_handler uri=/prov-session session_id=57 req_session_id=58
I (40580) protocomm_httpd: common_post_handler uri=/prov-session session_id=58 req_session_id=58
I (42100) protocomm_httpd: common_post_handler uri=/prov-scan session_id=58 req_session_id=58
I (46090) protocomm_httpd: common_post_handler uri=/prov-scan session_id=58 req_session_id=59
E (46200) security1: Secure session not established.
E (46210) protocomm: Decryption of response failed for endpoint prov-scan
E (46220) protocomm_httpd: Data handler failed.
```

Alvin1Zhang, 2020-12-25:
Thanks for reporting, we will look into.

william-ferguson-au, 2020-12-26:
I'm curious. How does the ESPIDF http server expect to maintain a session with the client when it doesn't set a session_id in the response header?

shahpiyushv, 2020-12-28:
@william-ferguson-au The session id in this context is the socket number. We have observed that the Android networking layer does not give much control over sockets being used for HTTP connections. It is expected that the entire provisioning workflow (starting with first establishing a secure session) happens on the same socket session. This is often what happens in most cases. However, sometimes, Android opens a new socket session midway, giving errors like "E (46200) security1: Secure session not established". We made some attempts to fix the issue, but since the problem itself wasn't getting reproduced, it was hard to know if the fix is working.

Since you are seeing this issue often, can you test with the attached apk which has the speculative fix?

[ESP_SoftAP_Provisioning.apk.zip](https://github.com/espressif/esp-idf/files/5746630/ESP_SoftAP_Provisioning.apk.zip)

william-ferguson-au, 2020-12-28:
OK, firstly I can confirm that the linked APK doesn't solve the problem.

Second, assuming that the entire provisioning workflow is a really bad assumption.  While HTTP 1.1 allows for persistent connections, you cannot guarantee that a HTTP client will maintain a TCP connection. It may close it for many reasons including resource constraints, error handling etc. 

The only way that you can ensure a session over HTTP is for the server to provide a session cookie to the client that the client sends back in its requests. 

I have a rudimentary solution that I'll provide as a pull request for you to mull over.

shahpiyushv, 2020-12-29:
@william-ferguson-au , can you confirm what your observations with the new apk are? The session is supposed to get re-established even if the socket changes midway. Is that not happening?

Meanwhile, it is not an assumption, but a requirement that the entire communication happens on the same socket, to ensure better security. It is expected that the clients adhere to the requirement. This gets ensured in iOS automatically and we will check how to handle it better in Android. We may have to use the low level socket calls if required.

Since this is not a remote connection, this is quite an acceptable requirement and provisioning failing in case of resource constraints, error handling, connection timeout etc. is also expected.

Alvin1Zhang, 2020-12-29:
https://github.com/espressif/esp-idf/pull/6330

william-ferguson-au, 2020-12-29:
I have created a pull request that provides a rudimentary fix. https://github.com/espressif/esp-idf/pull/6330
The http client will need to enable cookies, but that is common and a one line correct to the Android app.

The real problem here is the assumption that a socket_fd == http_session. That is just not how web servers work.
Typically nr_open_sockets is hundreds or thousands of times less than the number of open sessions. 
The 2 things are most definitely not the same.

To be really blunt, *requiring* a http client to have the entire provisioning conversation run on the same socket means that you require a specific implementation of http, no matter what the client runtime : Android, iOS, Java, PHP, C etc.

Please don't try and reinvent HTTP in a limited form, just adhere to it's specs.

william-ferguson-au, 2020-12-29:
@shahpiyushv to answer your question about the APK. The session is determined on the server side, so no the session is not re-established when the socket changes. As per the code in httpd_txtrx httpd_req_to_sockfd() it takes the new socket FD and calls it a session, so we get a new session.

shahpiyushv, 2020-12-29:
@william-ferguson-au , what I mean is, if the Android app encounters an error (because of new socket being used), it should re-start the process to establish the security session on the new socket. In case you have the http server logs enabled, can you check if that is happening. Can you share the ESP32 side logs if available?

Meanwhile, you are right. If we want to keep the http client completely in our control, modifying it to ensure that the communication happens on same socket would make sense. But since this is an open spec and you should ideally be able to use any client, we will surely consider the cookies approach you have suggested in the PR.

william-ferguson-au, 2020-12-29:
@shahpiyushv No, as I said the modified APK still fails.
Here is the ESP32 log https://gist.github.com/william-ferguson-au/03fe09d9e93af5e4b4d970e32c3b3bc7

The Android app never managed to show any networks from the device.

At first glance it looks like the the socket fd has flipped *during* the 2 part setup of the security context.

william-ferguson-au, 2020-12-30:
Question: why is protobuf being used for message encoding instead of something simple like Json or XML?

It makes adding a custom provisioning endpoint significantly harder. 
It also impedes debugging issues like this.

shahpiyushv, 2020-12-30:
@william-ferguson-au , protobuf was chosen as it is a compact representation and can support binary data much more easily. This is quite suitable especially for BLE provisioning which has smaller MTU.

For custom endpoints, you can use any encoding of your choice as you can see in the wifi_prov_mgr example [here](https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/examples/provisioning/wifi_prov_mgr/main/app_main.c#L106).

william-ferguson-au, 2020-12-31:
Thanks @shahpiyushv I missed that in the example.

shahpiyushv, 2021-01-07:
@william-ferguson-au , we will check the PR and try to merge it soon. However, since it is a significant change, it may take time and may not be backported to older releases as it is sort of a new feature. Instead, I have added a simple cookie logic in the protocomm_httpd layer (which is used in SoftAP provisioning) so that the scope stays limited and easier to test and backport to older releases. A patch (for idf master) and an apk has been attached here. Can you try and check if it solves the issue?
[SoftAP_Prov_with_cookies.zip](https://github.com/espressif/esp-idf/files/5782511/SoftAP_Prov_with_cookies.zip)

william-ferguson-au, 2021-01-08:
@shahpiyushv my PR does no more than what your patch does. It only attempts to splice in a Cookie into the protcomm_httpd layer. So go with your patch. It looks fine.

shahpiyushv, 2021-01-08:
@william-ferguson-au did you try this with the new APK? Did it work fine for you?

william-ferguson-au, 2021-01-09:
@shahpiyushv Yesy. Ran it through half a dozen times and works all times.
Looking at the log the http session_id appears to be getting transmitted and used as the key.

Just one weirdness in the logs when the session is first constructed
`
D (465093) protocomm_httpd: Creating new session: 1859346111
D (465103) protocomm_httpd: New socket session ID: 57
D (465103) httpd_txrx: httpd_req_recv: remaining length = 3
D (465113) httpd_txrx: httpd_recv_with_opt: requested length = 3
D (465113) httpd_txrx: httpd_recv_with_opt: pending length = 3
D (465123) httpd_txrx: httpd_req_recv: received length = 3
D (465123) protocomm: VER_EP Req handler returned 0
D (465133) protocomm_httpd: Setting cookie session=1859346111
D (465143) httpd_txrx: httpd_resp_set_hdr: new header = Set-Cookie: session=1859346111
D (465153) httpd_txrx: httpd_send_all: sent = 62
D (465153) httpd_txrx: httpd_send_all: sent = 10
D (465153) httpd_txrx: httpd_send_all: sent = 2
D (465163) httpd_txrx: httpd_send_all: sent = 18
D (465163) httpd_txrx: httpd_send_all: sent = 2
D (465183) httpd_txrx: httpd_send_all: sent = 2
D (465183) httpd_txrx: httpd_send_all: sent = 67
D (465183) httpd_sess: httpd_sess_process: httpd_req_delete
D (465183) httpd_sess: httpd_sess_process: success
D (465193) httpd: httpd_server: doing select maxfd+1 = 58
D (465503) httpd: httpd_server: processing listen socket 54
D (465503) httpd_sess: httpd_sess_close_lru: fd = 57
D (465503) httpd: httpd_server: doing select maxfd+1 = 58
D (465503) httpd: httpd_server: processing ctrl message
D (465513) httpd: httpd_process_ctrl_msg: work
D (465513) httpd_sess: httpd_sess_delete: fd = 57
W (465523) protocomm_httpd: Resetting socket session id as socket 57 was closed
`

Note the last line with the socket being reset. Only occurs when the session is first created.

shahpiyushv, 2021-01-11:
@william-ferguson-au , I am not sure what the reason could be, but probably this could be happening as the client first tries to get some information on /proto-ver endpoint and then starts establishing the secure session. After getting the /proto-ver information, rest of the communication must be happening on a new socket, thereby causing the earlier socket session id to get reset. Meanwhile, even if the socket session id gets reset, if the correct cookie is received in the next request, the session is continued. This socket session id logic is kept just to maintain backward compatibility.

closed 2021-01-15 by espressif-bot