espressif/esp-idf #6317: UDP Client | Loop exit does not work

Oleg-Perevyshin, 2020-12-25:
Hello everyone!
I took as a basis the project of their examples udp_client.

I need to send requests to the broadcast address not constantly, but once every 5 seconds and then wait 30 seconds. This part works provided that there is a device on the network that responds to requests.
The essence of the work is as follows:
- I wrapped the data receiving block in a loop while (1)
- I create a task **taskListenExit** to change the flag **receiverLoopFlag** (in order to exit the reception cycle)
- I send a broadcast request and for 2 seconds I am in a receive cycle in order to receive responses from all devices
- after 2 seconds I have to exit the receive loop and send the next request

The problem is, I cannot get out of the receive loop if there are no devices on the network (no response). Operation **break** does not do this.
Tell me, am I doing something wrong?
```
/* Receive data cycle status (for forced exit if no responses) */
bool receiverLoopFlag = 1;

/* The task of exiting the loop waiting for responses */
static void taskListenExit() {
  while (1) {
    vTaskDelay(SMALL_PERIOD_SCAN * 1000 / portTICK_PERIOD_MS);
    receiverLoopFlag = 0;
  }
}

static void taskUDPClient(void * pvParameters) {
  char rx_buffer[1024];
  int addr_family = 0;
  int ip_protocol = 0;
  while (1) {
    struct sockaddr_in dest_addr;
    dest_addr.sin_addr.s_addr = inet_addr(HOST_IP_ADDR);
    dest_addr.sin_family = AF_INET;
    dest_addr.sin_port = htons(PORT);
    addr_family = AF_INET;
    ip_protocol = IPPROTO_IP;

    int sock = socket(addr_family, SOCK_DGRAM, ip_protocol);
    if (sock < 0) {
      break;
    }
    int iRequest = NUM_REQUESTS;
    /* Cycle of periodically sending a packet to search for devices */
    while (1) {
      int err = sendto(sock, SEARCH_PHRASE, strlen(SEARCH_PHRASE), 0, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
      if (err < 0) {
        break;
      }
      receiverLoopFlag = 1;
      /* Waiting cycle for responses */
      while (1) {
        struct sockaddr_in source_addr;
        socklen_t socklen = sizeof(source_addr);
        int len = recvfrom(sock, rx_buffer, sizeof(rx_buffer) - 1, 0, (struct sockaddr *)&source_addr, &socklen);
        if (len < 0) {  /* Ошибка при получении */
          break;
        }
        if (len > 0) {  /* Данные получены, забираем серийный номер */
          /* Package processing */
        }
        if (receiverLoopFlag != 1) {
          break;
        }
      }
      iRequest--;
      if (iRequest > 0) {
        vTaskDelay(SMALL_PERIOD_SCAN * 1000 / portTICK_PERIOD_MS);
      } else {
        send_uart("UDPc", "LONG_PERIOD_SCAN", NULL);
        vTaskDelay(LONG_PERIOD_SCAN * 1000 / portTICK_PERIOD_MS);
        iRequest = NUM_REQUESTS;
      }
    }
    if (sock != -1) {
      shutdown(sock, 0);
      close(sock);
    }
  }
  bool taskStatus = true;
  while (taskStatus) {
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    vTaskDelete(NULL);
    if (udpClientInit() == ESP_OK) {
      taskStatus = false;
    }
  }
}

// Инициализация UDP Client
esp_err_t udpClientInit() {
  /* Создать json объект с серийным номером текущего устройства */
  cJSON * root = cJSON_CreateObject();
  cJSON_AddStringToObject(root, "0", getDeviceSN(FuncBuffer));
  devSNList = cJSON_PrintUnformatted(root);
  cJSON_Delete(root);
  /* Основная задача UDP клиента */
  if (xTaskCreate(taskUDPClient, "taskUDPClient", 4096, NULL, 5, NULL) != pdPASS) {
    return ESP_FAIL;
  }
  if (xTaskCreate(taskListenExit, "taskListenExit", 4096, NULL, 4, NULL) != pdPASS) {
    send_uart("UDPc", "taskListenExit", "Not Сreated");
    return ESP_FAIL;
  }
  return ESP_OK;
}
```

negativekelvin, 2020-12-26:
https://stackoverflow.com/questions/15941005/making-recvfrom-function-non-blocking

Oleg-Perevyshin, 2020-12-26:
I have read the link. The task matches mine!
But there is very little information and I did not understand anything how to implement it.

Is there an example or a more detailed explanation of how to do this?

Oleg-Perevyshin, 2020-12-26:
Thanks, figured it out!

closed 2020-12-26 by Oleg-Perevyshin