espressif/esp-idf #6319: Build does not work - python 3.9.0 and wrong magic number

bbulkow, 2020-12-26:
This environment works perfectly with Python 2.7 which I have been using for many years, but with each version you complain my python is too old so I tried using Python 3.

With a new download and build, I get the following error:

-- Project sdkconfig file /mnt/c/Users/x/dev/esp/FastLED-idf/sdkconfig
Traceback (most recent call last):
  File "/mnt/c/Users/x/dev/esp/esp-idf/tools/kconfig_new/confgen.py", line 27, in <module>
    import kconfiglib
ImportError: bad magic number in 'kconfiglib': b'\x03\xf3\r\n'
CMake Error at /mnt/c/Users/x/dev/esp/esp-idf/tools/cmake/kconfig.cmake:228 (message):
  Failed to run confgen.py
  (/home/x/.espressif/python_env/idf4.3_py3.9_env/bin/python;/mnt/c/Users/x/dev/esp/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/mnt/c/Users/x/dev/esp/esp-idf/Kconfig;--sdkconfig-rename;/mnt/c/Users/x/dev/esp/esp-idf/sdkconfig.rename;--config;/mnt/c/Users/x/dev/esp/FastLED-idf/sdkconfig;--env-file;/mnt/c/Users/x/dev/esp/FastLED-idf/build/config.env).
  Error 1
Call Stack (most recent call first):
  /mnt/c/Users/x/dev/esp/esp-idf/tools/cmake/build.cmake:443 (__kconfig_generate_config)
  /mnt/c/Users/x/dev/esp/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

This is with master as of today 12/25/20.

I have tried this with deleting the build directory and deleting sdkconfig and whatnot. I don't know what kconfiglib is an why it would have the 'wrong magic number'

dobairoland, 2020-12-26:
Hi @bbulkow. I'm sorry for the inconvenience. Please try to run `idf.py python-clean`. It should solve the problem.

dobairoland, 2020-12-29:
I assume this has been solved. Please let me know if not.

closed 2020-12-29 by dobairoland

Grufbert, 2021-11-18:
Worked for me

timmyhadwen, 2022-01-19:
Worked for me too!

malachib, 2022-04-08:
Worked for me too