espressif/esp-idf #6320: 4.2版本Uart2写入报文实际输出时，顺序错误

libinliu2013, 2020-12-26:
如题， **4.2版本， Uart2,** 写入如0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0xE, 0x0F
但接收程序收到的是：0x0D, 0x0E, 0x0F,    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C
已通过逻辑分析仪查看实际报文， 确认接收端程序没有问题。

**4.1版本和 3.6.1版本测试没有该问题， 仅在4.2版本测试出该问题。** 

**初始化代码为：**

`     //串口配置结构体
	uart_config_t uart2_config = {
		.baud_rate = 9600,					   //波特率
		.data_bits = UART_DATA_8_BITS,		   //数据位
		.parity = UART_PARITY_DISABLE,		   //校验位
		.stop_bits = UART_STOP_BITS_1,		   //停止位
		.flow_ctrl = UART_HW_FLOWCTRL_DISABLE, //硬件流控
	};
	//注册串口服务即使能+设置缓存区大小
	ESP_ERROR_CHECK(uart_driver_install(UART_NUM_2, 1024, 1024, 0, NULL, 0));

	//设置串口2
	ESP_ERROR_CHECK(uart_param_config(UART_NUM_2, &uart2_config));

	//IO映射
	ESP_ERROR_CHECK(uart_set_pin(UART_NUM_2, TX_AUDIO_PIN, RX_AUDIO_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
`
**发送代码为**

`
       char cmd[15];
        //填充cmd数据
       ....
	uart_write_bytes(UART_NUM_2, cmd, 5);
`

Alvin1Zhang, 2020-12-28:
@libinliu2013 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

libinliu2013, 2020-12-28:
谢谢，我们现在使用的是4.1版本， 一切正常。 
4.2版本问题的测试环境如下； 编译过程中未发现错误， 如上串口初始化、uart_write_bytes发送数据返回值等都显示一切正常， us, sdkconfig文件仅调整了ota选项和freertos的频率为1000Hz, 其他如默认配置，未做改动。

## Environment

- Module or chip used:  ESP32-S  Ai-Thinker
- IDF version: v4.2-47-g2532ddd9f
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- Using an IDE?:  Yes(Vscode)
- Power Supply:        USB / external 5V / external 3.3V

libinliu2013, 2020-12-28:
另，issue列表中 ， 有搜索到过类似的问题描述， 但他们是较老的版本中出现的，原因是fifo溢出造成的， 我们现在发送的数据很小16byte, 且版本最最新稳定版本。

Ethan-CYQ, 2020-12-31:
Hi, @libinliu2013 
Sorry I can't reproduce your problem.
Could you try to use `Uart1` and tell us if this problem still exist?

libinliu2013, 2021-01-02:
见谅，假期未上线。
我们的项目同时使用了串口0,1,2, 该问题仅在4.2版本的串口2出现。 
已测试工作正常的版本是3.6.1和4.1,  我们现在使用的就是4.1版本, 工作是正常的。

Alvin1Zhang, 2021-01-04:
@libinliu2013 感谢更新及反馈，版本 3.6.1 能提供下相关的链接么？谢谢。

Ethan-CYQ, 2021-01-05:
Hi, @libinliu2013 
你可以尝试一下在`release/v4.2`删除`components/driver/uart.c`中`uart_param_config`接口的tx_fifo复位代码`uart_hal_txfifo_rst(&(uart_context[uart_num].hal));`，看一下能不能解决你的问题。

libinliu2013, 2021-01-06:
> @libinliu2013 感谢更新及反馈，版本 3.6.1 能提供下相关的链接么？谢谢。

您指的是3.6.1版本sdk的下载链接？ 应该github的历史版本里面能够找到吧？ 我们是之前一直用这个版本，项目建立后， 一直未升级，所以之前用的一直是这个版本。 
另，我们之所以升级4版本， 是因为3.6.1版本，我们在使用时，存在STA和AP模式同时工作时， AP的热点经常性出现连接不上的问题， 所以更新的4版本.

libinliu2013, 2021-01-06:
> Hi, @libinliu2013
> 你可以尝试一下在`release/v4.2`删除`components/driver/uart.c`中`uart_param_config`接口的tx_fifo复位代码`uart_hal_txfifo_rst(&(uart_context[uart_num].hal));`，看一下能不能解决你的问题。

非常感谢， 我找时间测试下， 这段时间用的是4.1版本， 暂时来看， 一切正常。

Alvin1Zhang, 2021-08-05:
@libinliu2013 您好，这个问题还有其他更新么？谢谢。

libinliu2013, 2021-08-05:
后续没时间测试了， 见谅。  现在使用的还是4.1版本， 一直很稳定。

closed 2021-08-05 by libinliu2013

Alvin1Zhang, 2021-08-06:
@libinliu2013 感谢提供更新。