espressif/esp-idf #6321: UDP Server | Memory leak when using cJSON

Oleg-Perevyshin, 2020-12-26:
I am using UDP Server based on example.
I want to return JSON.
I found that there is a memory leak that is slightly larger than the data size.
This is part of the task code.
```c
while (1) {
      struct sockaddr_in6 source_addr;
      socklen_t socklen = sizeof(source_addr);
      int len = recvfrom(sock, udpsBuffer, sizeof(udpsBuffer) - 1, 0, (struct sockaddr *)&source_addr, &socklen);
      if (len < 0) {  /* Ошибка при получении */
        sendUART("UDPs", "Error Receiving", NULL);
        break;
      }
      if (len > MIN_LENGTH_PACK) {  /* Данные получены */
        /* Определть IP-адрес отправителя */
        if (source_addr.sin6_family == PF_INET) {
          inet_ntoa_r(((struct sockaddr_in *)&source_addr)->sin_addr.s_addr, addr_str, sizeof(addr_str) - 1);
        }
        udpsBuffer[len] = 0;
        /* Определить совпадение с searchPhrase */
        if (strcmp(udpsBuffer, searchPhrase) == 0) {
          /* Сформировать пакет для отправки */
          cJSON * root = cJSON_CreateObject();
          cJSON_AddStringToObject(root, "DevID", configDev.DevID);
          cJSON_AddStringToObject(root, "DevSN", configDev.DevSN);
          cJSON_AddStringToObject(root, "DevAddr", configDev.DevAddr);
          const char * jsonString = cJSON_PrintUnformatted(root);
          len = strlen(jsonString);
          int err = sendto(sock, jsonString, len, 0, (struct sockaddr *)&source_addr, sizeof(source_addr));
          cJSON_Delete(root);
          if (err < 0) {
            sendUART("UDPs", "Error occurred during sending", NULL);
            break;
          }
        }
      }
    }
```
In this loop, I get the data and compare against the phrase searchPhrase.
Next, I form a JSON packet and send it back.
If delete the line with `const char * jsonString = cJSON_PrintUnformatted(root);` and just return the received packet (echo), then there is no leak.
I rechecked everything line by line - this is the place that creates the memory leak.
How can you solve this?

negativekelvin, 2020-12-27:
https://stackoverflow.com/questions/27394364/should-the-return-value-of-cjson-print-be-freed-by-the-caller

Oleg-Perevyshin, 2020-12-27:
Yes it helped!

closed 2020-12-27 by Oleg-Perevyshin