espressif/esp-idf #6323: Instructions on how to get ESP-IDF should be on top of release info

rogerdahl, 2020-12-27:
The instructions on how to get ESP-IDF for a given release from GitHub is currently the bottom of the release info, underneath the (very long) changelog. It would be better to have it on top, as the first thing you see when opening a release.

Alvin1Zhang, 2020-12-28:
Thanks for raising this feature request.

projectgus, 2020-12-29:
Hi @rogerdahl,

Yes, agree this seems reasonable. The steps have been at the end for a long time now, since back when the changelogs were shorter.

We won't edit the existing releases, as editing a release sends a new round of emails to everyone who is subscribed. But we can change this starting from the next release.

Alvin1Zhang, 2021-08-05:
Thanks for your suggestion, we have used this approach in our recent releases. Will close the issue.

closed 2021-08-05 by Alvin1Zhang