espressif/esp-idf #6324: UDP Cient | CORRUPT HEAP: multi_heap.c:540 detected at 0x3ffc8e7c

Oleg-Perevyshin, 2020-12-27:
Faced strange application behavior.
The first time I send a request, everything goes well, I get a complete list of all responses.
On the second run to send the request, the application crashes.

```c
// Database of serial numbers and correspondences of IP addresses found on the network
char * devsList;

/* Create json with serial number of current device and IP address */
cJSON * root = cJSON_CreateObject();
cJSON_AddStringToObject(root, getDeviceSN(commonBuffer), ipAddr);
char * jsonString = cJSON_PrintUnformatted(root);
int len = strlen(jsonString);
devsList = (char *) malloc(len + 1);
devsList = cJSON_PrintUnformatted(root);
cJSON_free(jsonString);
cJSON_Delete(root);
```

```c
// Checking the received package and adding new devices to the list
static void checkDevSN(const char * udpcBuffer, const char * addressString) {
  cJSON * rxData = cJSON_Parse(udpcBuffer);
  if (rxData != NULL) {
    const char * currentDev = cJSON_GetObjectItem(rxData, "DevSN")->valuestring;
    if (validDevSN(currentDev) == true) {
      cJSON * devsList_t = cJSON_Parse(devsList);
      cJSON * item = NULL;
      int isExist = false;
      cJSON_ArrayForEach(item, devsList_t) {
        if (strcmp(item->string, currentDev) == 0) {
          isExist = true;
          break;
        }
      }
      cJSON_Delete(item);
      if (isExist == false) {
        // ToDo: Проверить IP адрес, если отличается - обновить запись
        cJSON_AddStringToObject(devsList_t, currentDev, addressString);
        char * newDevsList = cJSON_PrintUnformatted(devsList_t);
        cJSON_Delete(devsList_t);
        int len = strlen(newDevsList);
        devsList = (char *) realloc(devsList, len + 1);
        strcpy(devsList, newDevsList);
        cJSON_free(newDevsList);
      } else {
        cJSON_Delete(devsList_t);
      }
    }
  }
  cJSON_Delete(rxData);
}
```

```c
// Sending requests to find devices (Task UDC Client)
static void taskUDPClient(void * pvParameters) {
  char udpcBuffer[128];
  char addressString[128];
  int addressFamily = 0;
  int ipProtocol = 0;
  /* Timeout for recvfrom */
  struct timeval read_timeout;
  read_timeout.tv_sec = 3;
  while (1) {
    struct sockaddr_in dest_addr;
    dest_addr.sin_addr.s_addr = inet_addr(UDPC_HOST_IP_ADDR);
    dest_addr.sin_family = AF_INET;
    dest_addr.sin_port = htons(UDP_PORT);
    addressFamily = AF_INET;
    ipProtocol = IPPROTO_IP;
    int sock = socket(addressFamily, SOCK_DGRAM, ipProtocol);
    if (sock < 0) {
      sendUART("ER!", "UDP Client Task", "Unable to create socket");
      break;
    }
    setsockopt(sock, SOL_SOCKET, SO_RCVTIMEO, &read_timeout, sizeof read_timeout);
    int requestCounter = 3;

    while (1) {
      int err = sendto(sock, UDP_SEARCH_PHRASE, strlen(UDP_SEARCH_PHRASE), 0, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
      if (err < 0) {
        sendUART("ER!" , "UDP Client Task", "Error Sending");
        break;
      }
      /* Waiting for responses */
      while (1) {
        struct sockaddr_storage source_addr;
        socklen_t socklen = sizeof(source_addr);
        int len = recvfrom(sock, udpcBuffer, sizeof(udpcBuffer) - 1, 0, (struct sockaddr *)&source_addr, &socklen);
        if (len < 0) break;
        /* Get the sender's IP address */
        inet_ntoa_r(((struct sockaddr_in *)&source_addr)->sin_addr, addressString, sizeof(addressString) - 1);
        /* Data received */
        if (len > 0) {
          udpcBuffer[len] = 0;
          checkDevSN(udpcBuffer, addressString);
        }
      }
      if (--requestCounter == 0) {
        /* Displaying an updated database of serial numbers */
        sendUART("UDPc", "devsList", devsList);
        vTaskDelay(10 * 1000 / portTICK_PERIOD_MS);
        requestCounter = 3;
      }
    }
    if (sock != -1) {
      sendUART("ER!" , "UDP Client Task", "Restarting");
      shutdown(sock, 0);
      close(sock);
    }
  }
  udpClientDeInit();
}
```
On the first request, I receive and process everything correctly `UDPc devsList  {"FF008-6D1FC8C-F2DC0":"192.168.1.107","FD8A0-61DF5EC-22900":"192.168.1.108","F7C9E-6BDFE3C-07180":"192.168.1.109"} `
On the second occurrence (after the delay vTaskDelay (10 * 1000 / portTICK_PERIOD_MS);) the application crashes with an error:
```c
CORRUPT HEAP: multi_heap.c: 540 detected at 0x3ffc8e94

abort () was called at PC 0x4008bb99 on core 0

Backtrace: 0x40087d77: 0x3ffcc180 0x40088415: 0x3ffcc1a0 0x4008ef7a: 0x3ffcc1c0 0x4008bb99: 0x3ffcc230 0x4008141a: 0x3ffcc250 0x4008efa5: 0x3ffcc270 0x400d771e: 0x3ffcc290 0x400d770d: 0x3ffcc2b0 0x400d6638: 0x3ffcc2d0 0x400d675a: 0x3ffcc2f0 0x4008841e: 0x3ffcc450

ELF file SHA256: 8d42d0b59f82caa2
```

negativekelvin, 2020-12-27:
Possible overflow here `strcpy(devsList, newDevsList);`. Otherwise check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html

This is not the best place to get help writing your code, posting here should be reserved for when you find a bug in esp-idf code.  You can post at esp32.com forum or other places.

Alvin1Zhang, 2020-12-28:
Thanks for the notes @negativekelvin . Thanks for reporting, will close now, please help create a post on our [forum](https://esp32.com). Thanks.

closed 2020-12-28 by Alvin1Zhang