espressif/esp-idf #6325: Error sending command to eMMC device

911c, 2020-12-27:
Hello. Сould you please document the emmc driver in more detail. Especially the structure of the command description.
```c++
typedef struct {
        uint32_t opcode;            /*!< SD or MMC command index */
        uint32_t arg;               /*!< SD/MMC command argument */
        sdmmc_response_t response;  /*!< response buffer */
        void* data;                 /*!< buffer to send or read into */
        size_t datalen;             /*!< length of data buffer */
        size_t blklen;              /*!< block length */
        int flags;                  /*!< see below */                        <<<<<<<<<<<<<<<?????????
/** @cond */
#define SCF_ITSDONE      0x0001     /*!< command is complete */
#define SCF_CMD(flags)   ((flags) & 0x00f0)
#define SCF_CMD_AC       0x0000
#define SCF_CMD_ADTC     0x0010
#define SCF_CMD_BC       0x0020
#define SCF_CMD_BCR      0x0030
#define SCF_CMD_READ     0x0040     /*!< read command (data expected) */
#define SCF_RSP_BSY      0x0100
#define SCF_RSP_136      0x0200
#define SCF_RSP_CRC      0x0400
#define SCF_RSP_IDX      0x0800
#define SCF_RSP_PRESENT  0x1000
/* response types */
#define SCF_RSP_R0       0 /*!< none */
#define SCF_RSP_R1       (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX)
#define SCF_RSP_R1B      (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX|SCF_RSP_BSY)
#define SCF_RSP_R2       (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_136)
#define SCF_RSP_R3       (SCF_RSP_PRESENT)
#define SCF_RSP_R4       (SCF_RSP_PRESENT)
#define SCF_RSP_R5       (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX)
#define SCF_RSP_R5B      (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX|SCF_RSP_BSY)
#define SCF_RSP_R6       (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX)
#define SCF_RSP_R7       (SCF_RSP_PRESENT|SCF_RSP_CRC|SCF_RSP_IDX)
/* special flags */
#define SCF_WAIT_BUSY    0x2000     /*!< Wait for completion of card busy signal before returning */
/** @endcond */
        esp_err_t error;            /*!< error returned from transfer */
        int timeout_ms;             /*!< response timeout, in milliseconds */
} sdmmc_command_t;
```
Can't figure out either this problem is with hardware or with code.  I have a circuit board esp-wroom-32. I want to read register extcsd. But the transmission attempt always ends ESP_ERR_TIMEOUT.  I'm one hundred percent sure of the quality of the connections. The card is being detected.
```c++
void setup()
{
    Serial.begin(115200);
    Serial.println("==Hello world==");
    sdmmc_slot_config_t slot_config = {SDMMC_SLOT_NO_CD, SDMMC_SLOT_NO_WP, 1, 0};
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.flags = SDMMC_HOST_FLAG_1BIT;
    host.max_freq_khz = SDMMC_FREQ_DEFAULT;
    sdmmc_card_t card;

    Serial.println("sdmmc_host_init");
    esp_err_t lerr = sdmmc_host_init();
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_host_init_slot");
    lerr = sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_card_init");
    lerr = sdmmc_card_init(&host,&card);
    Serial.println(esp_err_to_name(lerr));
    
    sdmmc_card_print_info(stdout, &card);
}
```
Log:
```
==Hello world==
sdmmc_host_init
ESP_OK
sdmmc_host_init_slot
ESP_OK
sdmmc_card_init
ESP_OK
Name: MMC04GX
Type: MMC
Speed: 20 MHz
Size: 3688MB
CSD: ver=3, sector_size=512, capacity=7553024 read_bl_len=9
```
But as soon as I try to send a command to read the register, everything ends in failure.
```c++
   uint8_t ecsd[EXT_CSD_MMC_SIZE] = {};

    sdmmc_command_t cmd;
    cmd.opcode = MMC_SEND_EXT_CSD;
    cmd.arg = 0;
    cmd.response[0] = 0;
    cmd.response[1] = 0;
    cmd.response[2] = 0;
    cmd.response[3] = 0;
    cmd.data = &ecsd;
    cmd.datalen = EXT_CSD_MMC_SIZE;
    cmd.blklen = 1;
    cmd.flags = SCF_RSP_R1 | SCF_CMD_ADTC;     \\<<<???  not sure
    cmd.error = 0;
    cmd.timeout_ms = 1000;

    Serial.println("sdmmc_host_do_transaction");
    lerr = host.do_transaction(SDMMC_HOST_SLOT_1, &cmd);
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_command_t error field");
    Serial.println(esp_err_to_name(cmd.error));

    Serial.println("sdmmc_command_t response field");
    for(int i = 0; i<=3; i++) {
        Serial.println(cmd.response[i],BIN);
    }
```
Log:
```
sdmmc_host_do_transaction
ESP_OK
sdmmc_command_t error field
ESP_ERR_TIMEOUT
sdmmc_command_t response field
100100000000
0
0
0
```
What am I doing wrong? is it some kind of error on the esp32 side?

igrr, 2020-12-27:
@911c You are pretty close to getting it to work, I think you are only missing SCF_CMD_READ bit flag — it indicates to the underlying driver that the command will return some data. You can see the code which sends this command here:
https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/components/sdmmc/sdmmc_mmc.c#L203-L215

The meaning of command flags (AC, ADTC, BC, etc) are given by the SD and eMMC specifications. Usually you need to look up the command in the spec to determine its type and the response format (R1, R2, etc). For example, for SEND_EXT_CSD command, you will find:
[image]

Some of the EXT_CSD fields are made available in sdmmc_card_t::ext_csd — if you need any which aren't present there, please let us know and we can add them.

911c, 2020-12-28:
@igrr thanks for responding. Now the answer comes ... but the output seems to be incorrect data. they just repeat one through three zeros.  I also used "assert" to check the array in DMA memory.  I think this is some kind of error in my code ... could you please point out what I'm doing wrong?
```c++
void setup()
{
    Serial.begin(115200);
    Serial.println("==Hello world==");
    sdmmc_slot_config_t slot_config = {SDMMC_SLOT_NO_CD, SDMMC_SLOT_NO_WP, 1, 0};
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.flags = SDMMC_HOST_FLAG_1BIT;
    host.max_freq_khz = SDMMC_FREQ_DEFAULT;
    sdmmc_card_t card;

    Serial.println("sdmmc_host_init");
    esp_err_t lerr = sdmmc_host_init();
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_host_init_slot");
    lerr = sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_card_init");
    lerr = sdmmc_card_init(&host,&card);
    Serial.println(esp_err_to_name(lerr));
    
    sdmmc_card_print_info(stdout, &card);
    
    uint8_t ecsd[EXT_CSD_MMC_SIZE] = {};
    assert(esp_ptr_dma_capable(ecsd));

    sdmmc_command_t cmd;
    cmd.opcode = MMC_SEND_EXT_CSD;
    cmd.arg = 0;
    cmd.response[0] = 0;
    cmd.response[1] = 0;
    cmd.response[2] = 0;
    cmd.response[3] = 0;
    cmd.data = ecsd;
    cmd.datalen = EXT_CSD_MMC_SIZE;
    cmd.blklen = 1;
    cmd.flags = SCF_CMD_ADTC | SCF_RSP_R1 | SCF_CMD_READ;
    cmd.error = 0;
    cmd.timeout_ms = 1000;

    Serial.println("sdmmc_host_do_transaction");
    lerr = host.do_transaction(SDMMC_HOST_SLOT_1, &cmd);
    Serial.println(esp_err_to_name(lerr));

    Serial.println("sdmmc_command_t error field");
    Serial.println(esp_err_to_name(cmd.error));

    Serial.println("sdmmc_command_t response field");
    for(int i = 0; i<=3; i++) {
        Serial.println(cmd.response[i],BIN);
    }
    Serial.println();
    Serial.println("sdmmc_command_t data field");
    for(int i = 0; i<512; i++){
      Serial.println(ecsd[i], BIN);
    }

    Serial.println();
    Serial.println("==DONE==");
      
}

void loop() {}
```
Log:
```
==Hello world==
sdmmc_host_init
ESP_OK
sdmmc_host_init_slot
ESP_OK
sdmmc_card_init
ESP_OK
Name: MMC04GX
Type: MMC
Speed: 20 MHz
Size: 3688MB
CSD: ver=3, sector_size=512, capacity=7553024 read_bl_len=9
sdmmc_host_do_transaction
ESP_OK
sdmmc_command_t error field
ESP_FAIL
sdmmc_command_t response field
100100000000
0
0
0

sdmmc_command_t data field
11111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110001111111000111111100011111110000000000000000000
==DONE==
```

911c, 2020-12-28:
@igrr In general I wanted to read the boot data ... and if it is wrong edit it for further boot from this device.
[image]

911c, 2020-12-28:
@igrr It's strange but if I put a flag
```c++
host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
```
getting
```
sdmmc_command_t data field
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

911c, 2020-12-28:
Ohh...it seems to work...
```c++
cmd.datalen = EXT_CSD_MMC_SIZE;
cmd.blklen = EXT_CSD_MMC_SIZE;
```

closed 2020-12-29 by 911c