espressif/esp-idf #6329: tcp keepalive issue in AP mode

AxelLin, 2020-12-29:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.3-dev-2136-gb0150615dff5
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I notice sometimes when tcp keepalive is enabled, the tcp server got
KEEPALIVE timeout. Aborting. then *all* servers on the device stop working.
This issue can be reproduced when device running in AP mode.
(Also observe when device in STA mode, but not easy to reproduce it).

### Expected Behavior
TCP keepalive timeout should abort the socket, but it should continue accept new connection.
(BTW, the tcp keepalive timeout should not happen at all as attached wireshark shows my device send back ACK but esp32 does not receive it)

### Actual Behavior
Device no longer accept new connection.
In additional, when the issue happened all servers on the device stop working.

### Steps to reproduce

I modify the examples/protocols/sockets/tcp_server example code as attached file.
So the example code will star an AP (TEST-AP, 12345678 as password).
It also start a tcp server listen on port 3333, with tcp keepalive enabled.

Copy tcp_server.c.txt to examples/protocols/sockets/tcp_server/main/tcp_server.c

Build and run the code.
I use my laptop to join TEST-AP, and connect to the tcp server.
 $ netcat 192.168.4.1 3333
 type something and got echo.
Then just wait for several seconds and observe if the tcp keep alive abort issue will happen or not.

To make it easier to know if tcp keep alive abort issue happen or not,
I add a debug print to show pcb->keep_cnt_sent (https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/tcp.c#L1346)
When pcb->keep_cnt_sent > 1, then it's likely this issue happen. then wait unitl hit ABORT.

It does not happen everytime. so if it does not happen, restart the esp32 and re-start the test again.

I also has wireshark log attached, (the esp32 softap is 192.168.4.1, my client device is 192.168.4.2).
In the wireshark, I can see the device ACK the tcp keepalive but it looks like esp32 does not receive it.

BTW, I mentioned that when this issue happen then *all* servers on the device stop working.
It's not showing in this example code, but in my real application which also has httpd running,
I notice httpd stop working once I hit this issue.

[tcp_server.c.txt](https://github.com/espressif/esp-idf/files/5749481/tcp_server.c.txt)
[tcp_keepalive_abort_issue.pcapng.txt](https://github.com/espressif/esp-idf/files/5749490/tcp_keepalive_abort_issue.pcapng.txt)

Alvin1Zhang, 2020-12-29:
Thanks for reporting and letting us know, we will look into.

david-cermak, 2021-01-07:
@AxelLin Is the issue reproducible when running just the tcp_server example with your modification above? No other server, no httpd, etc... I wasn't able to reproduce it, must be missing something, maybe `sdkconfig` (had all options per default).

If I understand the report correctly, the keepalive packet is accepted by the client, the response gets lost (which still might happen) and the connection is aborted (still the correct action), but then after that, the server itself (and other active servers)  is/are not able to receive any other connection. Correct?

I've tested the scenario, and even made some of the keepalive responses dropped to match exactly your attached packet capture. Got the connection aborted, but the server was capable of accepting new connections.
I've also added another `tcp_server` on port `3334` to check if this usecase breaks/invalidates unrelated connections (to mimic your httpd issues). Still seems to work okay when made the lwip ignore keepalive replies, it just aborts the specific connection with no influence to the other server.

AxelLin, 2021-01-07:
> @AxelLin Is the issue reproducible when running just the tcp_server example with your modification above? No other server, no httpd, etc... I wasn't able to reproduce it, must be missing something, maybe `sdkconfig` (had all options per default).

Yes, it should be able to reproduce with the modified tcp_server example. (although it;s not 100% reproduceable)
I didn't modify other sdkconfig. just run idf.py build

> If I understand the report correctly, the keepalive packet is accepted by the client, the response gets lost (which still might happen) and the connection is aborted (still the correct action), but [...]

I think your understanding is correct.

> 
> I've tested the scenario, and even made some of the keepalive responses dropped to match exactly your attached packet capture. Got the connection aborted, but the server was capable of accepting new connections.

Actually, I did the same test as well, but it cannot reproduce the issue this way.
My experience is try the modified tcp_server example multiple times. (and add debug print for pcb->keep_cnt_sent )
If pcb->keep_cnt_sent keeps showing 1, it does not happened.
Usually I just check a few minutes, if it does not happen then reboot and check again.

david-cermak, 2021-01-08:
>  it;s not 100% reproduceable

How often are you able to reproduce it? I've checked many times, always seeing the `pcb->keep_cnt_sent` as `1` in the `tcp.c`. Even written a script which restarts the ESP32 if this value is 1 and reinitiates a new connection after the reset. This runs for several hour already, but no issue found so far (didn't change the default timeouts, so it only waits for about 35 seconds for the first occurrence of `keep_cnt_sent++`, meaning the test ran 500-1000times so far), which makes me think we must have something set differently in the environment...

> Actually, I did the same test as well, but it cannot reproduce the issue this way.

This could mean, that the keepalive ACK is actually received by the ESP32. Could you please take another look and e.g.  enable the `TCP_INPUT_DEBUG`, to see if the packet were actually accepted by the lwip?

david-cermak, 2021-01-08:
> I add a debug print to show pcb->keep_cnt_sent (https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/tcp.c#L1346) When pcb->keep_cnt_sent > 1, then it's likely this issue happen. then wait unitl hit ABORT.

btw, how could this `keep_cnt_sent` be ever greater than 1 after the restart and the initial connection? Is not this the only place where it's modified?

AxelLin, 2021-01-08:
> btw, how could this `keep_cnt_sent` be ever greater than 1 after the restart and the initial connection? Is not this the only place where it's modified?

I mean reboot the esp32 device and re-test again.
i.e.
esp32 running as softap and run a tcp server.
My linux laptop join the esp32 softap and run netcat to connect tcp server.
Then wait to see if the keep_cnt_sent increases.
(which means esp32 does not receive the keepalive ack from netcat)

As I attached the wireshark log, I'm sure the netcat sent back ack but esp32 still increase keep_cnt_sent (which means it does not receive the ack).

david-cermak, 2021-01-08:
> I mean reboot the esp32 device and re-test again.

Yes, this is exactly what I was doing. Thanks, just wanted to double-check with you that we're on the same page, since I always read the `keep_cnt_set` as `1`, unless I force the keepalive ack packet to get lost. And even then, after the connection aborts, the server still accepts connections.

I've also tried to simulate the packet getting lost before it's posted to the stack, so not dropping it in the `tcp_in` but simply not receiving it -- with exactly the same result. Connection aborts, but the ESP32 is ready accepting new connections.

AxelLin, 2021-01-11:
Hi @david-cermak 

Now I sync my esp-idf to v4.3-dev-2398-g2bfdd036b2db, and re-test again.
I test 3 times, hit the issue 2 times.
Then I enable TCP_INPUT_DEBUG, test 2 times and I hit the issue at the second time test.

Attached my test logs.

(I guess you probably stop testing after got first pcb->keep_cnt_sent=1. You have to wait longer time, e.g. a few minutes)

BTW, when I check the 20210111.debuglog1.txt, I'm wondering why it shows negative values below:
tcp_receive: ACK for -1625747489, unacked->seqno -1625747490:-1625747489
tcp_receive: removing -1625747490:-1625747489 from pcb->unacked
I (7951) example: Socket accepted ip address: 192.168.4.2

[20210111.log1.txt](https://github.com/espressif/esp-idf/files/5793712/20210111.log1.txt)
[20210111.log2.txt](https://github.com/espressif/esp-idf/files/5793713/20210111.log2.txt)
[20210111.debuglog1.txt](https://github.com/espressif/esp-idf/files/5793714/20210111.debuglog1.txt)

AxelLin, 2021-01-11:
> BTW, when I check the 20210111.debuglog1.txt, I'm wondering why it shows negative values below:
> tcp_receive: ACK for -1625747489, unacked->seqno -1625747490:-1625747489
> tcp_receive: removing -1625747490:-1625747489 from pcb->unacked
> I (7951) example: Socket accepted ip address: 192.168.4.2

Below line causes printing the negative seqno values that needs fix:
https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/arch/cc.h#L67

AxelLin, 2021-01-11:
I'm a little surprised it's so easy to reproduce the issue now.
I change to another test device, still can reprdouce it.
Attached the sdkconfig.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5794272/sdkconfig.txt)

Note, my test device already has enabled flash encryption.
so you may need to disable it with the sdkconfig file.

Below is what I did:
$ idf.py erase_flash
$ idf.py encrypted-flash monitor (run the first time test, this usually fine)
  Make my laptop join esp32 softap, and execute "netcat 192.168.4.1 3333"
$ idf.py monitor (test again, this will restart the module, this time usually hit issue)
  I did "sudo iwlist scan", then the esp32 shows a station joined. Then execute "netcat 192.168.4.1 3333".

david-cermak, 2021-01-11:
Hi @AxelLin 

Thanks for the additional details!
I have switched to the 2bfdd036b2, as well, reverted all my debug changes and applied your `sdkconfig` (without the flash encryption). And made sure to wait for several minutes, to get the `pcb->keep_cnt_sent = 1` printed at least 10 times, but still the same result.

> 20210111.debuglog1.txt

Thanks for the debug log! I'm sorry, I meant to ask you for enabling also the `TCP_DEBUG`  to have the packet headers printed, too, but still we can clearly see that after your first occurrence of `pcb->keep_cnt_sent=1`, no packet is coming to the `tcp_input()` -> very interesting and alarming

> Below line causes printing the negative seqno values that needs fix:

Thanks, will fix!

With that said, I think we are still missing something, so asking for 2 more details:
1) It seems like the keepalive ACKs went out from your laptop (per your wireshark log) but didn't arrive to the `tcp_process()` on the ESP32. Could we somehow check where the packet(s) got lost?
* Could you please try to capture the Air packets on some other device/laptop to see if the ACK packets are there.
* Could you please add this debug log to the `wlanif_input()` to see if the packet arrive into the netif inputs:
``` c
    if (len > 47) {
        const uint8_t my[] = { 0xc0, 0xa8, 0x04, 0x02, 0xc0, 0xa8, 0x04, 0x01};
        int match = 0;
        for (int i=0; i<sizeof (my); ++i) {
            if (((uint8_t*)buffer)[26+i] == my[i]) {
                match++;
            }
        }
        if (match == sizeof (my) && ((uint8_t*)buffer)[36] == 0x0d && ((uint8_t*)buffer)[37] == 0x05) {
            ESP_LOG_BUFFER_HEXDUMP("IN", buffer, len, ESP_LOG_INFO);
        }
  }
```
this would just print  out packets between `192.168.4.1` and `192.168.4.2` on port `333` 
please paste here:

https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/wlanif.c#L186

2) Could you please share the version of your `netcat`? (not expecting any issue, just based on previous customer reports, there might be subtle differences in the tools..., so wanted to use exactly the same environment when trying to reproduce)

AxelLin, 2021-01-11:
> Below is what I did:
> $ idf.py erase_flash
> $ idf.py encrypted-flash monitor (run the first time test, this usually fine)
> Make my laptop join esp32 softap, and execute "netcat 192.168.4.1 3333"
> $ idf.py monitor (test again, this will restart the module, this time usually hit issue)
> On my laptop, I did "sudo iwlist scan", then the esp32 shows a station joined. Then execute "netcat 192.168.4.1 3333".

I figure out why it's so easy to reproduce.
In above steps, I didn't leave the esp32 AP, and just run "sudo iwlist scan" then my laptop re-connect to esp32.
If I make the laptop join to other AP then re-join esp32 AP, it's not easy to reproduce.
This is strange because I'm sure netcat indeed connect to esp32 and I esp32 shows the words I sent via netcat.
then it becomes unreachable latter.
I attached the log.
[keepalive_debug_log.txt](https://github.com/espressif/esp-idf/files/5794835/keepalive_debug_log.txt)

Above is a different issue (I didn't hit such issue when test with non-esp32 platforms so I think it's still an issue), and if I re-join esp32 I still can connect to the tcp server.

I will change my test steps to make my laptop join other AP then re-join esp32 AP.
It seems not easy to reproduce this way, I will update once I hit the issue again.

btw, my netcat is the standard package from ubuntu-20.04 ( netcat-openbsd   1.206-1ubuntu1)

AxelLin, 2021-01-19:
Hi @david-cermak 

Thanks for your support.

After testing with different laptops:
1. Test with different Windows laptops, cannot repoduce the issue.
2. Test with my colleague's laptop which is running Linux, cannot reproduce this issue.
I found this issue only happened on my laptop running Linux, maybe it's not esp32 issue. Close it now.

closed 2021-01-19 by AxelLin