espressif/esp-idf #6333: esp32-s2: i2s, dma and dac -- supported or doc bug?

jepler, 2020-12-29:
[ESP-IDF Documentation states](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/dac.html)
> The DAC channels can also be driven with DMA-style written sample data, via the I2S driver when using the “built-in DAC mode”.

However, the related functions are not available via the ESP-IDF when compiling for the esp32-s2, because SOC_I2S_SUPPORTS_ADC_DAC is not defined in `components/soc/esp32s2/include/soc/soc_caps.h`.

Is there a the supported way to DMA to the DAC on ESP32-S2?

jepler, 2020-12-29:
@ladyada

ginkgm, 2020-12-30:
Hi @jepler ,
    This is an documentation issue. On S2 there is still DAC-DMA feature (though the API hasn't be released), but no longer depend on the I2S. The digital controller can make use of the DMA and generate the period required by DAC.

jepler, 2020-12-30:
Thank you for the clarification.  Is support planned in a future esp-idf?

ginkgm, 2021-01-13:
Hi @jepler ,

Yes, it will be supported in the future. Currently you can refer to the unit test code, if it's urgent for you. https://github.com/espressif/esp-idf/blob/master/components/driver/test/dac_dma_test/test_esp32s2.c

jepler, 2021-01-13:
Thank you @ginkgm , we'll take a look!

Can you also answer whether PDM functionality is available on the S2?

closed 2021-01-15 by espressif-bot

jepler, 2021-01-25:
@ginkgm Thanks again, I successfully built the example on Kaluga and on Adafruit's Metro ESP32-S2 and it seems to work properly. However, while adapting the code to CircuitPython I have a strange outcome where there are spurious edges in the waveforms, even in simple sawtooth waves:
[image]

Do you have any advice on how to continue investigating this problem?  What further information can I furnish?

jepler, 2021-02-08:
@ginkgm Just a gentle ping -- do you have any guidance or is there another place more appropriate to raise the issue of the audio waveforms shown above?