espressif/esp-idf #6337: Add support to flash external flash

tsctrl, 2020-12-30:
This is to keep track the future enhancement request that was asked in the esp32 forum.
[https://www.esp32.com/viewtopic.php?f=13&t=18727](Unable to flash FAT FS on External Flash)

Problem statement: 
Not able to preprogram external flash with predefined content. Flashing external spi flash that was supported by espressif using similar way like internal flashing using mkspiffs and esptool or any equivalent approach to provide such functionality.

This could be added in the road map to be deliver in Q1 2021.

igrr, 2020-12-30:
@tsctrl thanks for posting this feature request. One part (generating FATFS images on host) is tracked in #5785. The other part (flashing something to external flash) is already possible with esptool.py, by specifying `--spi-connection` argument. We will consider how and whether to integrate this part into `idf.py`.

CmManuES, 2021-12-02:
It will be a nice feature to save data directly in an external SPI flash during bootloading?
I would like to save graphic elements, e.g. bitmaps and fonts, in a SPIFFS partition in an external SPI flash.
How can I save the data directly in the external flash at flashing or bootloading?
Comparable to a partition in main flash, which I can define in partitions.csv.
Is of course also interesting for a FAT file system

closed 2024-04-17 by espressif-bot