espressif/esp-idf #6340: LAN8720 stop at "Ethernet Started"

Grufbert, 2020-12-30:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version: v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-47-g2532ddd9f-dirty
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE: VSCODE
- Power Supply: USB

## Problem Description

Currently I try to get the ethernet basic example running. I figured out some issues but now I stuck and have no clue what to do next.

I boot up the example basic ethernet example with adjusted settings to LAN9720 and clock out on gpio 7. MDC and MDIO is working fine and looks ok on oscilloscope. The clock also looks good. If I disconnect the clock (output von GPIO17) I get a timeout error as expected. Valid vendor ID is pulled from the LAN8720 registers. PHY address is correctly double checked. But the last serial output is always "Ethernet Started".

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (374) system_api: Base MAC address is not set
I (384) system_api: read default base MAC address from EFUSE
I (384) BEYOND: Found PHY address: 1
I (414) esp_eth.netif.glue: 30:ae:a4:47:7a:b7
I (414) esp_eth.netif.glue: ethernet attached to netif
I (424) eth_example: Ethernet Started
```

The "ETHERNET_EVENT_CONNECTED" event aka "Ethernet Link Up" is never fired. 

Who is responsible for the "Ethernet Link Up" event? The PHY or the MAC? I tried different Ethernet cables / Ethernet Routers and even a complete other setup (own PCB with ESP32 + LAN8720) same behavior.

Any Idea what I can try?

Look forward to your help!

Thanks!

### Hardware changes

1.5k pull-up on MDIO
1.5k pull-up on CRS

### Expected Behavior

"Ethernet Link Up" event fire

suda-morris, 2020-12-30:
> Who is responsible for the "Ethernet Link Up" event

PHY.

> Any Idea what I can try?

Check the PHY datasheet, make sure your LAN9720 driver is reading the correct register which indicates the link status.

Grufbert, 2020-12-30:
Hi @suda-morris, thanks for your fast reply.

The LAN8720 [datasheet](http://ww1.microchip.com/downloads/en/devicedoc/00002165b.pdf) shows on "4.2.2 Basic Status Register" page 42 that bit 2 provide the link status. 0 = down, 1 = up.

The readout from this register ETH_PHY_BMSR_REG_ADDR (0x01) result in 30729 (dec) aka 0111 1000 000 10**0**1 (bin).

Unfortunately there is no change in the ETH_PHY_BMSR_REG_ADD register, even if I disconnect or connect a cable.

More ideas?

nathanRamaNoodles, 2020-12-30:
Try this? [https://sautter.com/blog/ethernet-on-esp32-using-lan8720/](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/)
Also, make sure you set the reset pin to which ever pin you soldered to the eth's clock. (so pin 7 in this case?)
I am using gpio 0 as input clk, which means I never edited the default configuration for the ethernet component in `menuconfig`.
Also, if it helps, try setting the `sw_reset_timeout_ms` to 2000.
```
eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;

mac_config.smi_mdc_gpio_num = CONFIG_EXAMPLE_ETH_MDC_GPIO;
mac_config.smi_mdio_gpio_num = CONFIG_EXAMPLE_ETH_MDIO_GPIO;
mac_config.sw_reset_timeout_ms = 2000;   // try this?
```

Grufbert, 2021-01-01:
I double measured all my signals. Found a broken TX wire between PHY and transformer jack. Fixed this issue and now the LAN8720 works fine. Thank you guys!

closed 2021-01-01 by Grufbert