espressif/esp-idf #6342: Bluetooth Classic Panic when switching to SNIFF Mode

JL1946, 2020-12-30:
## Environment

- Development Kit: N/A
- Kit version N/A
- Module or chip used:  ESP32-WROVER-IE
- IDF version : v4.3-dev-2136-gb0150615d
- Build System:         -- CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc8_4_0-esp-2020r3-macos
- Operating System:     macOS
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Eclipse 2020-12 with ESP-IDF Plugin -- Segger Jlink 
- Power Supply:         USB
## Problem Description
 esp-idf --> Bluetooth a2dp_source example modified as follows:
1. test.wav and silent.wav file in SPIFFS
2. GPIO interrupt (IRAM_ATTR) on I/O Pin 23 that plays the test.wav file on connected bluetooth speaker when pulled low
3. Timer task (replaces Heartbeat task) to periodically (120 seconds) send the silent.wav file to keep the Speaker from disconnecting itself from the ESP32.
4. Bluetooth Controller & Bluedroid running on Core 1.
I have 2 issues:
1. [MAJOR] - Intermittent "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled." The Exception always occurs after sending the file and the system switches to SNIFF Mode in the gap callback function. The debugger always sets a breakpoint and stops at line 712 of btm_pm.c (p_cb = p_acl_cb->p_pm_mode_db;)
2. [MINOR] While the ESP32 is sending the wav file, I get many of "BT_L2CAP: l2cab is_cong_cback_context" errors printed to the monitor while the wav file is playing properly. After the function completes the file read/send, I get many "BT_APPL: Media task Scheduled after Suspend" Warnings printed to the monitor (see below printouts). The errors don't affect the sound on the speaker.

[elf file with sdkconfig.zip](https://github.com/espressif/esp-idf/files/5754997/elf.file.with.sdkconfig.zip)

### Expected Behavior
Switch from Active Mode to Sniff Mode occurs without panic
### Actual Behavior
Switch from Active Mode to Sniff Mode occurs intermittently with panic result
### Steps to reproduce
The read/send cycle starts with the "BT_APP_KEEP_ALIVE_EVT" generated by the 120-second timer function and ends with a switch back to the SNIFF mode in the GAP Callback Function

Normal Printout from Monitor:
I (17098497) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_STOPPED
<<<<<----- s_MEDIA_process Value: APP_MEDIA_IDLE_PROCESS
W (17098507) BT_APP_KEEP_ALIVE_EVT: --> --> --> --> --> --> --> --> KEEP ALIVE EVENT NUMBER: 140
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_STOPPED
----->>>>> s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (17098537) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_STOPPED
<<<<<----- s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (17098567) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY
I (17098577) ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY: --> --> --> SUCCESS
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_READY
----->>>>> s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
W (17099067) GAP CALLBACK: <<<<<<------ Incoming Event: ESP_BT_GAP_MODE_CHG_EVT
<<<<<----- s_GAP_state Value: APP_GAP_STATE_IDLE
<<<<<----- s_POWER_mode Value: POWER_MODE_SNIFF
I (17099077) GAP CALLBACK: ESP_BT_GAP_MODE_CHG_EVT mode: POWER_MODE_ACTIVE
----->>>>> s_GAP_state Value: APP_GAP_STATE_IDLE
----->>>>> s_POWER_mode Value: POWER_MODE_ACTIVE
I (17099107) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

I (17099107) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_READY
<<<<<----- s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (17099137) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_START
I (17099137) ESP_A2D_MEDIA_CTRL_START: --> --> --> SUCCESS
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
[... 1729 lines cut ...]

Name : fpreserved20
	Hex:0x0
	Decimal:0
	Octal:0
	Binary:0
	Default:0

Name : fpreserved20a
	Hex:0x0
	Decimal:0
	Octal:0
	Binary:0
	Default:0

Name : fpreserved5
	Hex:0x0
	Decimal:0
	Octal:0
	Binary:0
	Default:0
---------------------------------------------------------------------------
## Other items if possible
sdkconfig and elf file attached as a zip file
coredump not available

JL1946, 2020-12-30:
I switched to Release 4.2 --> Results:
1. The Power Mode switch from SNIFF to ACTIVE and back to SNIFF does not seem to occur.  I have no idea if the switch occurs internally and is not visible in the GAP Callback function as the "ESP_BT_GAP_MODE_CHG_EVT" is not in the GAP Event callback list in version 4.2. ---> So far, no panics have occurred.
2. The multiple "BT_L2CAP: l2cab is_cong_cback_context" errors are not occurring. 
3. I am seeing some occurrences of "BT_APPL: un-acked a2dp cmd: 3" in the A2DP Callback function (see printout below)
4. Continue to see multiple occurrences of " BT_APPL: Media task Scheduled after Suspend" After the ESP_A2D_AUDIO_STATE_EVT "STOPPED" is addressed in the code
5. I am seeing the following in the monitor just after the EOF is reached in the SPIFFS wav files;
              W (3220577) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 512 ###
              W (3220577) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 3, -1
              E (3220587) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffbbcb8
              W (3220607) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 513 ###
              W (3220607) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 14, -2
              W (3220607) BT_APPL: un-acked a2dp cmd: 3
6. THE SECTION OF CODE THAT READS/SENDS THE WAV FILE IS:
static int32_t bt_A2DP_Data_Send_Callback(uint8_t *data, int32_t len)
{

    if (len < 0 || data == NULL) {
        return 0;
    }

    int bytes_read = fread(data, 1, len, m_sample);
    if (bytes_read < len)                                                 // read up to EOF
    {
    	fclose(m_sample);
    	gpio_intr_enable(GPIO_INPUT_IO_0);
	xTimerStart(keep_alive_timer, portMAX_DELAY);                  // Reset the 120 second timer
        s_AUDIO_state  = APP_AV_AUDIO_STATE_STOPPING;         // Set the audio-state variable
	esp_a2d_media_ctrl(ESP_A2D_MEDIA_CTRL_STOP);           // Send STOP command
        return -1;
    }
    return len;
}

FULL TYPICAL MONITOR OUTPUT FOR 1 CYCLE:
***********************************************************************************************
I (2978387) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_STOPPED
<<<<<----- s_MEDIA_process Value: APP_MEDIA_IDLE_PROCESS
W (2978407) BT_APP_KEEP_ALIVE_EVT: --> --> --> --> --> --> --> --> KEEP ALIVE EVENT NUMBER: 24
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_STOPPED
----->>>>> s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
***********************************************************************************************
***********************************************************************************************
I (2978457) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_STOPPED
<<<<<----- s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (2978487) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY
I (2978497) ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY: --> --> --> SUCCESS
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_READY
----->>>>> s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
***********************************************************************************************
I (2978597) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

***********************************************************************************************
I (2978607) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPED
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_READY
<<<<<----- s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (2978637) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_START
I (2978637) ESP_A2D_MEDIA_CTRL_START: --> --> --> SUCCESS
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
[... 56 lines cut ...]
<<<<<----- s_MEDIA_process Value: APP_MEDIA_KEEP_ALIVE_PROCESS
I (2979267) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_STOP
I (2979277) ESP_A2D_MEDIA_CTRL_STOP: --> --> --> SUCCESS
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
----->>>>> s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPING
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_STOPPED
----->>>>> s_MEDIA_process Value: APP_MEDIA_IDLE_PROCESS
***********************************************************************************************
***********************************************************************************************
I (2979317) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_AUDIO_STATE_EVT
<<<<<----- s_A2DP_state Value: APP_AV_STATE_CONNECTED
<<<<<----- s_AUDIO_state Value: APP_AV_AUDIO_STATE_STOPPING
<<<<<----- s_AUDIO_state Value: APP_AV_MEDIA_STATE_STOPPED
<<<<<----- s_MEDIA_process Value: APP_MEDIA_IDLE_PROCESS
I (2979347) ESP_A2D_AUDIO_STATE_EVT: --> --> --> AUDIO STOPPED
----->>>>> s_A2DP_state Value: APP_AV_STATE_CONNECTED
[... cut ...] _STOPPED
----->>>>> s_MEDIA_state Value: APP_AV_MEDIA_STATE_STOPPED
----->>>>> s_MEDIA_process Value: APP_MEDIA_IDLE_PROCESS
***********************************************************************************************
W (2979387) BT_APPL: Media task Scheduled after Suspend
W (2979387) BT_APPL: Media task Scheduled after Suspend
W (2979397) BT_APPL: Media task Scheduled after Suspend
W (2979407) BT_APPL: Media task Scheduled after Suspend
W (2979407) BT_APPL: Media task Scheduled after Suspend

Alvin1Zhang, 2020-12-31:
Thanks for the very detailed report, and letting us know, we will look into.

JL1946, 2021-01-01:
Apparently, Release 4.2 is having some of the same issues with SNIFF, but, in this case the WROVER is disconnecting from the bluetooth speaker without a panic as with Release 4.3 demonstrated above. What follows is a printout showing the error and disconnect - LOOK FOR THE ARROWS ON THE RIGHT SIDE OF THE TEXT. (Note - I modified the code to show entry into and out of the GAP Callback and A2DP Callback):

[Test 11 elf wit sdkconfig.zip](https://github.com/espressif/esp-idf/files/5758641/Test.11.elf.wit.sdkconfig.zip)

************************************** ENTER A2DP CALLBACK **************************************
I (744757) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT
W (744757) BT_APP_KEEP_ALIVE_EVT: --> --> --> --> --> --> --> --> KEEP ALIVE EVENT NUMBER: 12
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (744787) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (744797) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY
I (744797) ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
I (744997) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8000

************************************** ENTER A2DP CALLBACK **************************************
I (744997) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (745007) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_START
I (745017) ESP_A2D_MEDIA_CTRL_START: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (745037) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_AUDIO_STATE_EVT
I (745047) ESP_A2D_AUDIO_STATE_EVT: --> --> --> AUDIO STARTED
************************************** LEAVE A2DP CALLBACK **************************************
EOF
W (746827) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 512 ###
W (746827) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 3, -1
E (746837) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffbb79c
W (746847) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 513 ###
W (746857) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 13, -2
W (746877) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 514 ###
W (746877) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -3
W (746907) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 515 ###
W (746907) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -4
************************************** ENTER A2DP CALLBACK **************************************
I (746927) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (746937) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_STOP
I (746937) ESP_A2D_MEDIA_CTRL_STOP: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (746967) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_AUDIO_STATE_EVT
I (746967) ESP_A2D_AUDIO_STATE_EVT: --> --> --> AUDIO STOPPED
************************************** LEAVE A2DP CALLBACK **************************************
E (778967) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7.                         <<<<<--------ERROR
************************************** ENTER A2DP CALLBACK **************************************
I (778967) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_CONNECTION_STATE_EVT
I (778977) ESP_A2D_CONNECTION_STATE_EVT: --> --> --> A2DP DISCONNECTED.         <<<<------ Disconnect
I (778987) DEVICE DISCOVERY: Starting device discovery...
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER GAP  CALLBACK **************************************
W (779007) GAP CALLBACK: <<<<<<------ Incoming Event: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (779017) GAP CALLBACK: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (779027) GAP CALLBACK: Discovery Started.
************************************** LEAVE GAP  CALLBACK **************************************
************************************** ENTER GAP  CALLBACK **************************************
W (804607) GAP CALLBACK: <<<<<<------ Incoming Event: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (804617) GAP CALLBACK: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (804617) GAP CALLBACK: Discovery Failed --> CONTINUE TO DISCOVER...
I (804627) DEVICE DISCOVERY: Starting device discovery...
************************************** LEAVE GAP  CALLBACK **************************************
************************************** ENTER GAP  CALLBACK **************************************
W (804647) GAP CALLBACK: <<<<<<------ Incoming Event: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (804657) GAP CALLBACK: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (804667) GAP CALLBACK: Discovery Started.
************************************** LEAVE GAP  CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (806827) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT
[... cut ...]
I (855907) DEVICE DISCOVERY: Starting device discovery...
************************************** LEAVE GAP  CALLBACK **************************************
************************************** ENTER GAP  CALLBACK **************************************
W (855937) GAP CALLBACK: <<<<<<------ Incoming Event: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (855937) GAP CALLBACK: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (855947) GAP CALLBACK: Discovery Started.
************************************** LEAVE GAP  CALLBACK **************************************
*

JL1946, 2021-01-03:
New Issue:
----------------------
I made some changes to my code to reconnect to the bt speaker when the switch to SNIFF mode error occurs. The reconnect functions properly, but I am now running into the following intermittent error:

E8192
ASSERT_ERR(0), in ld_fm.c at line 559

**Note: Panic or Exception restart does not occur**

below is a printout from monitor for the cycle that fails:
************************************** ENTER A2DP CALLBACK **************************************
I (2934998) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT
W (2934998) BT_APP_KEEP_ALIVE_EVT: --> --> --> --> --> --> --> --> KEEP ALIVE EVENT NUMBER: 46
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (2935028) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (2935038) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY
I (2935038) ESP_A2D_MEDIA_CTRL_CHECK_SRC_RDY: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
I (2935558) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x8000

************************************** ENTER A2DP CALLBACK **************************************
I (2935568) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (2935578) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_START
I (2935588) ESP_A2D_MEDIA_CTRL_START: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (2935608) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_AUDIO_STATE_EVT
I (2935618) ESP_A2D_AUDIO_STATE_EVT: --> --> --> AUDIO STARTED
************************************** LEAVE A2DP CALLBACK **************************************
EOF
W (2937068) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 512 ###
W (2937068) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 3, -1
E (2937078) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x3ffdf30c
W (2937098) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 513 ###
W (2937098) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 13, -2
W (2937128) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 514 ###
W (2937128) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -3
W (2937158) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 515 ###
W (2937158) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -4
W (2937188) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 516 ###
W (2937188) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -5
W (2937218) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 517 ###
W (2937218) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -6
W (2937248) BT_APPL: ### UNDERFLOW :: ONLY READ -1 BYTES OUT OF 518 ###
W (2937248) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, -7
************************************** ENTER A2DP CALLBACK **************************************
I (2937268) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_MEDIA_CTRL_ACK_EVT
I (2937278) ESP_A2D_MEDIA_CTRL_ACK_EVT: ESP_A2D_MEDIA_CTRL_STOP
I (2937288) ESP_A2D_MEDIA_CTRL_STOP: --> --> --> SUCCESS
************************************** LEAVE A2DP CALLBACK **************************************
************************************** ENTER A2DP CALLBACK **************************************
I (2937308) A2DP STATE MACHINE: <<<<<----- Incoming Event: ESP_A2D_AUDIO_STATE_EVT
I (2937318) ESP_A2D_AUDIO_STATE_EVT: --> --> --> AUDIO STOPPED
************************************** LEAVE A2DP CALLBACK **************************************
E8192
ASSERT_ERR(0), in ld_fm.c at line 559                                           **<<<---- Speaker disconnects here but No Restart or Panic**
************************************** ENTER A2DP CALLBACK **************************************
I (2997068) A2DP STATE MACHINE: <<<<<----- Incoming Event: BT_APP_KEEP_ALIVE_EVT.  <<<--- Next Timed Event
W (2997068) BT_APP_KEEP_ALIVE_EVT: --> --> --> --> --> --> --> --> KEEP ALIVE EVENT NUMBER: 47
E (3657078) BT_APP_CORE: bt_app_send_msg xQueue send failed    **<<<--- ESP_A2D_MEDIA_CTRL_ACK_EVT not received**
E (3717078) BT_APP_CORE: bt_app_send_msg xQueue send failed
E (3777078) BT_APP_CORE: bt_app_send_msg xQueue send failed

[Test 12 with sdkconfig.zip](https://github.com/espressif/esp-idf/files/5762157/Test.12.with.sdkconfig.zip)

Any help that you can give me would be appreciated. Thanks

JL1946, 2021-01-06:
I made a change to the project that has apparently cleaned up a lot of problems for me. Since I am not using WiFi in my project, I disabled the Bluetooth/wifi coexistence in the sdkconfig. From reading the Espressif docs, I realized that using coexistence imposes limits on the RF communication protocol. When I disabled the coexistence mechanism, things turned around for me. I have had a solid link with the speaker since then. The error has gone away and I am no longer getting the above mentioned error-induced disconnects. The other change made was to limit Log messages and printf statements in the A2DP Callback function.

The issue I am having now is that if I send nothing to the speaker while it is bonded to the ESP32, after about 5-6 minutes, it shuts itself off. My understanding was that the ESP32 Bluetooth code is supposed to send a keep-alive frame to the speaker every 20 mSeconds, so either that is not happening or the speaker itself does not recognize the frame and shuts down to conserve battery. It is an inexpensive Sony SRS-XB01.

My research into this issue has yielded nothing. If anyone has some info about this issue, please respond. Thanks...

Wth-Esp, 2021-01-20:
@JL1946 

I have a SONY's Bluetooth speaker too which named after ZR7. I noticed that when I use iPhone connecting to ZR7 with no audio transferred. The connection will power off after a while, maybe 15min or 20min. And on the Bluetooth side, when there are no data to send the connection is kept by NULL and POLL packet. So, I think it's a common battery strategy that SONY's speaker all have.

Thanks

Wth-Esp, 2021-01-20:
@JL1946 

The same thing happens on the Bluetooth headphones named WH-1000M2 and WH-1000M3. So... I think maybe sony's engineer start a timer when a2dp data shut down to close the connection and power off.

Thanks

JL1946, 2021-02-18:
@Wth-Esp 

Yes... I agree with you. I have noticed that with other speakers, I don't seem to have this intermittent issue. As a workaround, I run a timer function to send a 2-second silent wave file to the speaker every 5 minutes. This workaround is functioning to keep the connection, but I still get drop-outs every once in a while. When it occurs, the ESP32 reconnects to the bonded speaker... I guess that I will have to live with this issue when using Sony Bluetooth speakers. Thanks...

closed 2021-02-18 by JL1946

Wth-Esp, 2021-02-19:
@JL1946 You can download MusicCenter app from SONY. And close the automatically disconnect in the settings.