espressif/esp-idf #6344: simple_ota_example: Firmware upgrade failed

kaibin1111, 2020-12-31:
I (4708) simple_ota_example: Starting OTA example
I (7228) esp_https_ota: Starting OTA...
I (7228) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
D (9338) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128
I (9338) esp_https_ota: Connection closed
I (9338) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x05b2c ( 23340) map
E (9358) esp_image: invalid segment length 0xffffffff
E (9358) simple_ota_example: Firmware upgrade failed

mahavirj, 2020-12-31:
@kaibin1111 

Information you have provided is not sufficient, please follow issue template.

Please see if suggestion from below comment help you:
https://github.com/espressif/esp-idf/issues/5240#issuecomment-639292839

kaibin1111, 2020-12-31:
Thank you for your reply. The problem has been solved.

mahavirj, 2020-12-31:
@kaibin1111 Thanks for update, I will close the issue.

closed 2020-12-31 by mahavirj