espressif/esp-idf #6345: Invalid pointer error when building documentation

rdeterre, 2020-12-31:
## Environment

- IDF version (run ``git describe --tags`` to find it): v4.3-dev-2136-gb0150615d
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No

- Doxygen version (`doxygen --version`):  1.8.20
- Sphinx version: 2.3.1
- Breathe version: 4.14.1
- sphinx-idf-theme version: 0.2
- sphinx-notfound-page version: 0.5
- blockdiag version: 2.0.1
- recommonmark version: 0.7.1

## Problem Description

I am trying to build the ESP-IDF documentation on my Fedora Linux 33 computer, but the `./build_docs.py` command fails

### Expected Behavior

The `./build_docs.py` command should build the documentation and output without errors

### Actual Behavior

The ./build_docs.py` command fails to build the documentation, and instead outputs an error (see full log below)

### Steps to reproduce

1. Install Linux prerequisites for ESP-IDF as listed [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html)
2. Install doxygen
3. Get ESP-IDF (latest master branch)
4. Call `./install.sh` then `. ./export.sh` from the `esp-idf` directory
5. `cd docs`
6. `pip install -r requirements.txt`
7. `./build_docs.py -l en -t esp32`

## Debug Logs

Here is the end of the output of `./build_docs.py -l en -t esp32`. The full log file is
[log.txt](https://github.com/espressif/esp-idf/files/5757814/log.txt)

```
...
en/esp32: reading sources... [100%] versions
en/esp32: 
en/esp32: /home/rdeterre/Documents/docsets/esp-idf/docs/en/api-guides/ulp-legacy.rst:136: WARNING: Duplicate declaration, esp_err_t ulp_load_binary(uint32_t load_addr, const uint8_t * program_binary, size_t program_size)
en/esp32: /home/rdeterre/Documents/docsets/esp-idf/docs/en/api-guides/ulp-legacy.rst:142: WARNING: Duplicate declaration, esp_err_t ulp_run(uint32_t entry_point)
en/esp32: /home/rdeterre/Documents/docsets/esp-idf/docs/en/api-guides/ulp-legacy.rst:157: WARNING: Duplicate declaration, esp_err_t ulp_set_wakeup_period(size_t period_index, uint32_t period_us)
en/esp32: ../../../../components/ulp/README.rst:75: WARNING: Duplicate declaration, esp_err_t ulp_run(uint32_t entry_point)
en/esp32: inc/esp_ble_mesh_defs.inc:33: WARNING: Duplicate declaration, uint16_t esp_ble_mesh_model::model_id
en/esp32: inc/esp_spp_api.inc:44: WARNING: Error in type declaration.
en/esp32: If typedef-like declaration:
en/esp32:   Type must be either just a name or a typedef-like declaration.
en/esp32:   If just a name:
en/esp32:     Error in declarator or parameters and qualifiers
en/esp32:     Invalid definition: Expected identifier in nested name, got keyword: void [error at 4]
en/esp32:       void() esp_spp_cb_t(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
en/esp32:       ----^
en/esp32:   If typedef-like declaration:
en/esp32:     Error in declarator
en/esp32:     If pointer to member declarator:
en/esp32:       Invalid definition: Expected identifier in nested name. [error at 4]
en/esp32:         void() esp_spp_cb_t(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
en/esp32:         ----^
en/esp32:     If declId, parameters, and qualifiers:
en/esp32:       Invalid definition: Expected identifier in nested name. [error at 4]
en/esp32:         void() esp_spp_cb_t(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)
en/esp32:         ----^
[... 61 lines cut ...]
en/esp32: writing output... [ 15%] api-guides/linker-script-generation
en/esp32: writing output... [ 16%] api-guides/lwip
en/esp32: writing output... [ 16%] api-guides/mesh
en/esp32: writing output... [ 17%] api-guides/partition-tables
en/esp32: writing output... [ 17%] api-guides/romconsole
en/esp32: writing output... [ 18%] api-guides/thread-local-storage
en/esp32: writing output... [ 18%] api-guides/tools/idf-docker-image
en/esp32: writing output... [ 18%] api-guides/tools/idf-monitor
en/esp32: writing output... [ 19%] api-guides/tools/idf-tools
en/esp32: writing output... [ 19%] api-guides/tools/index
en/esp32: writing output... [ 20%] api-guides/ulp
en/esp32: writing output... [ 20%] api-guides/ulp-legacy
en/esp32: writing output... [ 21%] api-guides/ulp_instruction_set
en/esp32: writing output... [ 21%] api-guides/ulp_macros
en/esp32: writing output... [ 22%] api-guides/unit-tests
en/esp32: writing output... [ 22%] api-guides/unit-tests-legacy
en/esp32: writing output... [ 22%] api-guides/wifi
en/esp32: munmap_chunk(): invalid pointer
Local doxygen version 1.8.20 is newer than CI doxygen version 1.8.11. Local build may contain warnings that will not be raised when built by CI.
[('en', 'esp32', '/home/rdeterre/Documents/docsets/esp-idf/docs/_build/en/esp32', '/home/rdeterre/Documents/docsets/esp-idf/docs/en', 1, ['html'], [''])]
[-6]

The following language/target combinations failed to build:
language: en, target: esp32, errcode: -6
```

Alvin1Zhang, 2021-01-04:
Thanks for reporting and letting us know, we will look into.

ESP-Marius, 2021-01-27:
Seems like some issue with the underlying libraries on your system. I'll try to find the time to replicate your setup and test it, but it might take a while.

But thanks for the report!

rdeterre, 2021-01-28:
Thanks for the update. For information, I'm trying to build documentation to create a docset to make it usable with [Dash](https://kapeli.com/dash) and [Zeal](https://zealdocs.org/). Is there a docker image or another environment where I could build the documentation without having my system's libraries mess things up?

ESP-Marius, 2021-01-29:
Unfortunately the docker image we use for building the docs isn't published anywhere, but the IDF docker image should also work as a base if you install the additional python packages listed in `docs/requirements.txt`. 

Here's our public docker images: https://hub.docker.com/r/espressif/idf/tags?page=1&ordering=last_updated

ESP-Marius, 2021-08-05:
Closing this, the recommended approach if running into similar issues is using a ubuntu based Docker image for building.

closed 2021-08-05 by ESP-Marius

mickeyl, 2024-03-19:
@rdeterre Did you succeed in building a docket for Dash and Zeal? I'd like to use it, too.