espressif/esp-idf #6346: abort when using tickless idle mode with DEBUG log level

mickael9, 2020-12-31:
## Environment

- Module or chip used:  ESP32-S
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-2136-gb0150615d
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
Using tickless idle mode and having `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`, the system crashes when attempting to go to idle.

It seems to be caused by this line:
https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/components/esp_system/sleep_modes.c#L891

## Backtrace
```
#0  esp_system_abort (details=0x3ffb5b4b "abort() was called at PC 0x4008252e on core 0") at /data/sdks/esp-idf/components/esp_system/system_api.c:108
#1  0x4008acb1 in abort () at /data/sdks/esp-idf/components/newlib/abort.c:46
#2  0x40082531 in lock_acquire_generic (lock=<optimized out>, delay=4294967295, mutex_type=4 '\004') at /data/sdks/esp-idf/components/newlib/locks.c:140
#3  0x4008267c in _lock_acquire_recursive (lock=0x3ffaeb8c) at /data/sdks/esp-idf/components/newlib/locks.c:168
#4  0x400dc50d in _vfprintf_r (data=0x3ffb6160, fp=0x3ffaeb34, fmt0=0x3f403f98 "D (%d) %s: RTC_PERIPH: %s\033[0m\n", ap=...)
    at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853
#5  0x400df54c in vprintf (fmt=0x3f403f98 "D (%d) %s: RTC_PERIPH: %s\033[0m\n", ap=...) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
#6  0x400e3e08 in esp_log_writev (level=<optimized out>, tag=<optimized out>, format=0x3f403f98 "D (%d) %s: RTC_PERIPH: %s\033[0m\n", args=...) at /data/sdks/esp-idf/components/log/log.c:189
#7  0x4008ab68 in esp_log_write (level=ESP_LOG_DEBUG, tag=0x3f403f90 "sleep", format=0x3f403f98 "D (%d) %s: RTC_PERIPH: %s\033[0m\n") at /data/sdks/esp-idf/components/log/log.c:199
#8  0x400d50cd in get_power_down_flags () at /data/sdks/esp-idf/components/esp_system/sleep_modes.c:891
#9  0x400d5487 in esp_light_sleep_start () at /data/sdks/esp-idf/components/esp_system/sleep_modes.c:457
#10 0x40082b5f in vApplicationSleep (xExpectedIdleTime=<optimized out>) at /data/sdks/esp-idf/components/esp_pm/pm_impl.c:625
#11 0x40086981 in prvIdleTask (pvParameters=0x0) at /data/sdks/esp-idf/components/freertos/tasks.c:3878
#12 0x40088060 in vPortTaskWrapper (pxCode=0x40086938 <prvIdleTask>, pvParameters=0x0) at /data/sdks/esp-idf/components/freertos/port/xtensa/port.c:168
```

### Code to reproduce this issue

```cpp
#include "esp_pm.h"

void app_main(void)
{
    esp_pm_config_esp32_t pm_config = {
        .light_sleep_enable = true,
        .max_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
        .min_freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ,
    };
    ESP_ERROR_CHECK(esp_pm_configure(&pm_config));
}

```
### sdkconfig.defaults
```kconfig
CONFIG_PM_ENABLE=y
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
```

Alvin1Zhang, 2021-01-04:
Thanks for reporting, we will look into.

S0l0m4n, 2022-02-21:
@Alvin1Zhang Any update on this issue?

AxelLin, 2022-03-05:
@S0l0m4n 
Do you still hit the issue with latest update in v4.3 branch?
BTW, which module are you using?

S0l0m4n, 2022-03-09:
@AxelLin We use v4.2.2. in our project. Unfortunately, I can't compile it using v4.3. I get an error that `IRAM0 segment data does not fit`. We use the ESP32-D0WD-V3.

mickael9, 2022-03-19:
The problem is fixed in current master branch, it seems to be a duplicate of #7942

AxelLin, 2022-03-24:
> @AxelLin We use v4.2.2. in our project. 

Fixed in v4.2 branch: 84effb1

Alvin1Zhang, 2022-04-19:
Thanks for reporting, sorry for slow turnaround, fixes are 

* master branch: https://github.com/espressif/esp-idf/commit/a857f52107732a79136343e095f83f9f2b52e336
* release/4.4: https://github.com/espressif/esp-idf/commit/09a892a44f85321c875dc5dd2d154aa25b165ff5
* release/4.3: https://github.com/espressif/esp-idf/commit/76ddca045e1b89bddd3be76a3b3ea08d76bf2f79
* release/4.2: https://github.com/espressif/esp-idf/commit/84effb1708b93d33b8fa04a8bdac645cd84ca040
* release/4.1: https://github.com/espressif/esp-idf/commit/73f56bbbe18fb0ac3b892c3fbbe981f3895c0d59

Feel free to reopen.

closed 2022-04-19 by Alvin1Zhang