espressif/esp-idf #6348: No support for chain topology in manual mesh networking examples

MisterScience5, 2021-01-02:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used: ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): release/4.2
- Build System:        CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes - VS Code with ESP-IDF extension
- Power Supply:         Battery

## Problem Description

When employing chain topology, the networking IE is 62 bytes rather than 58 which does not match the packed struct mesh_assoc_t, causing the code example to be unable to extract assoc_cap, layer2_cap and possibly others causing a scan fail for chain topology AP's.

The example also does not identify chain topology networks as it employs esp_mesh_scan_get_ap_ie_len() to identify mesh AP's, and chain mesh AP's have a longer IE.

Chain topology does work with set_self_organized(true, x);

### Expected Behavior

Manual networking example should be able to identify and connect to chain AP's.

### Actual Behavior

### Steps to reproduce

1. Use code from manual networking to setup ESP-MESH, 
2. Change topology with esp_mesh_set_topology(MESH_TOPO_CHAIN);

### Code to reproduce this issue

void app_main()
{
esp_mesh_set_topology(MESH_TOPO_CHAIN);

mesh_assoc_t assoc;
esp_mesh_scan_get_ap_ie_len(&ie_len);
if (ie_len == sizeof(assoc)) { etc...
}

Alvin1Zhang, 2021-01-04:
Thanks for reporting.

MisterScience5, 2021-03-12:
Would it be possible to post what are the changes for tree topology mesh_assoc_t? With those details I should be able to make a workaround quite quickly:)

Alvin1Zhang, 2021-07-19:
@MisterScience5 Thanks for reporting, fix on release/4.2 is available https://github.com/espressif/esp-idf/commit/fd032425f144d3bda4cc9da35a5a869b8e6c6c32, feel free to reopen.

closed 2021-07-19 by Alvin1Zhang