espressif/esp-idf #6349: Memory leak: queue not freed in esp_mesh_stop()

MisterScience5, 2021-01-02:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): none
- Module or chip used: ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it): release/4.2
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):    xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes - VS Code with ESP-IDF extension
- Power Supply:         Battery

## Problem Description

There appears to be a queue created in esp_mesh_start() that is not freed on esp_mesh_stop(). In my application I periodically start and stop ESP-MESH to conserve battery, and therefore run esp_mesh_stop() and esp_mesh_start() repeatedly. However, on every cycle I lose 92 bytes of memory which appears to be queue created in esp_mesh_start() which is not freed, see trace.

### Expected Behavior

esp_mesh_stop() should free all allocated memory from esp_mesh_start().

### Actual Behavior

A queue is orphaned on each cycle of esp_mesh_stop()/esp_mesh_start().

### Steps to reproduce

1. Repeatedly run esp_mesh_start() and then esp_mesh_stop(). (In my case, I use the functions mesh_enable()/mesh_disable() in the code below).

### Code to reproduce this issue

`esp_err_t mesh_enable(void) {
  esp_err_t err = ESP_OK;
  if (!mesh_on) {
    if (!mesh_initialized) {
      ESP_LOGW(MESH_TAG, "Initializing mesh network");
    } else {
      ESP_LOGW(MESH_TAG, "Enabling mesh network");
    }
    if (!mesh_initialized) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_init());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_loop_create_default());NULL));");
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_netif_create_default_wifi_mesh_netifs(&netif_sta, NULL));
      wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_init(&config));&ip_event_handler, NULL));");
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler, NULL));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_storage(WIFI_STORAGE_RAM));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_ps(WIFI_PS_NONE));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_mode(WIFI_MODE_APSTA));
      if (mesh_settings.lr_mode == 1) {
        ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_LR));
        ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_LR));
      }
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_start());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_init());
    ESP_ERROR_CHECK_WITHOUT_ABORT(setup_root());    if (!mesh_initialized) {
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL));
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_topology(MESH_TOPO_CHAIN));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_max_layer(mesh_settings.max_layer));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_vote_percentage(1));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_xon_qsize(64));
    if (mesh_settings.pwr_save) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_enable_ps());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_assoc_expire(60));
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_announce_interval(5000, 30000));
    } else {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_disable_ps());
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_assoc_expire(10));
    }
    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
    memcpy((uint8_t *)&cfg.mesh_id, mesh_settings.id, 6);
    cfg.channel = mesh_settings.channel;
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_ap_authmode(mesh_settings.auth_mode));
    cfg.mesh_ap.max_connection = mesh_settings.max_conn;
    memcpy((uint8_t *)&cfg.mesh_ap.password, mesh_settings.ap_pass, strlen(mesh_settings.ap_pass));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_config(&cfg));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_start());
    mesh_on = true;
    if (!mesh_initialized) {
      ESP_ERROR_CHECK_WITHOUT_ABORT(esp_read_mac(mesh_self_addr.addr, 0));
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_set_self_organized(true, false));
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    if (!mesh_initialized) {
      if (esp_mesh_is_root()) {
        memcpy(&mesh_root_addr, &mesh_self_addr, sizeof(mesh_addr_t));
        memcpy(&node_status[0].address, &mesh_self_addr, sizeof(mesh_addr_t));
        node_status[0].exists = true;
        node_status[0].connected = true;
        node_status[0].current_version = SOFTWARE_VERSION;
        node_status[0].uptime = (uint32_t)(esp_timer_get_time() / 1000000ULL);
        node_status[0].mesh_layer = esp_mesh_get_layer();
      }
      mesh_initialized = true;
    }
    ESP_LOGW(MESH_TAG, "Mesh starts successfully for " MACSTR ", max. layer:%d heap:%d, %s<%d>%s, ps:%d\n", MAC2STR(mesh_self_addr.addr), esp_mesh_get_max_layer(), esp_get_minimum_free_heap_size(), esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed", esp_mesh_get_topology(), esp_mesh_get_topology() ? "(chain)" : "(tree)", esp_mesh_is_ps_enabled());
  } else {
    ESP_LOGW(MESH_TAG, "Mesh network already enabled!");
  }
  return err;
}

esp_err_t mesh_disable(void) {
  if (mesh_on) {
    ESP_LOGW(MESH_TAG, "Disabling mesh network, heap: %u", esp_get_free_heap_size());
    is_mesh_connected = false;
    if (esp_mesh_is_root()) {
      node_status[0].connected = false;
    }
    vTaskDelay(100 / portTICK_PERIOD_MS);
    if (!esp_mesh_is_root()) {
      esp_wifi_scan_stop();
    }
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_disconnect());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_flush_scan_result());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_flush_upstream_packets());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_stop());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_deinit());
    vTaskDelay(100 / portTICK_PERIOD_MS);
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_comm_p2p_stop());
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_stop());
    mesh_on = false;
  } else {
    ESP_LOGE(MESH_TAG, "Mesh network already disabled!");
  }
  return ESP_OK;
}``cpp

#include "esp_log.h"

`void app_main()
{
  for (uint8_t i = 0; i < 200; i++) {    
    heap_trace_start(HEAP_TRACE_LEAKS);
    mesh_enable();
    vTaskDelay(12000 / portTICK_PERIOD_MS);
    
    mesh_disable();
    vTaskDelay(3000 / portTICK_PERIOD_MS);
    heap_trace_stop();
    heap_trace_dump();
  }
}``cpp

## Debug Logs

For some reason, when using heap tracing stack depth 2, this queue is described as occupying 92 bytes every time, but with stack depth 4 it shows as 88, maybe it is not counting a pointer somewhere? I am losing 92 bytes every cycle, and the only things consistently in the trace are this and about 6 LWIP mallocs, however these are freed asynchronously.

```
88 bytes (@ 0x3ffe19b8) allocated CPU 0 ccount 0xa0555294 caller 0x4008f00e:0x4008f27b:0x400eece8:0x40112c00
0x4008f00e: xQueueGenericCreate at C:/Users/prebe/esp-idf/components/freertos/queue.c:391

0x4008f27b: xQueueCreateMutex at C:/Users/prebe/esp-idf/components/freertos/queue.c:498

0x400eece8: mutex_create_wrapper at C:/Users/prebe/esp-idf/components/esp_wifi/esp32/esp_adapter.c:301

0x40112c00: esp_mesh_start at ??:?
```

## Other items if possible

<script src="https://gist.github.com/MisterScience5/d52fba06f87e5ca2e5cfa7a82f58b809.js"></script>

Alvin1Zhang, 2021-01-04:
Thanks for reporting and letting us know, we will look into.

Satierfa, 2022-05-19:
Same issue here. Any news about it?

MisterScience5, 2022-07-20:
Did a recheck of this in 4.4 with ESP-MDF to see if there was any difference, but the problem still persists.

shenjun7, 2022-07-27:
Please provide a reproducible example．Is there a problem with mdf calling `mwifi_restart()`?

MisterScience5, 2022-07-27:
I dont know if mwifi_restart() elicits this behaviour, but I did a test with a vanilla fixed root example repeatedly calling mwifi_stop() and mwifi_start(), that does suffer from the memory leak in the same manner as esp_mesh_stop(), so I would think mwifi_restart() does as well if it calls esp_mesh_stop(). I will try to create an example tomorrow.

MisterScience5, 2022-08-08:
Sorry for the late reply, I have tested it with mwifi_restart(), which also elicits the same behaviour further supporting the hypothesis that the leak is due to improper cleanup in esp_mesh_stop(). My current test code:

`// Copyright 2017 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include "mdf_common.h"
#include "mwifi.h"
#include "mupgrade.h"
#include "esp_log.h"
#include "esp_partition.h"

// #define MEMORY_DEBUG
#define BUF_SIZE 512
#define SOFTWARE_VERSION (702)
#define MAX_NODES (200)

static const char *TAG = "no-router";
esp_netif_t *sta_netif;
const esp_partition_t *current_partition;
const esp_partition_t *ota_partition;
esp_image_metadata_t current_data;
esp_image_metadata_t ota_data;
TaskHandle_t root_read_handle = NULL;
TaskHandle_t node_read_handle = NULL;
TaskHandle_t ota_task_handle = NULL;
mesh_addr_t routing_table[MAX_NODES] = {0};
int routing_table_size = 0;
bool mesh_on = false;

void print_sha256(const uint8_t *image_hash, const char *label)
{
    char hash_print[32 * 2 + 1];
    hash_print[32 * 2] = 0;
    for (int i = 0; i < 32; ++i)
    {
        sprintf(&hash_print[i * 2], "%02x", image_hash[i]);
    }
    ESP_LOGI(TAG, "%s: %s", label, hash_print);
}

esp_err_t setup_partitions(void)
{
    // static bool is_partitions_setup = false;
    // if (!is_partitions_setup) {
    //   is_partitions_setup = true;
    current_partition = esp_ota_get_running_partition();
    assert(current_partition != NULL);
    ESP_LOGI("part", "Current partition: %s (%u)", current_partition->label, current_partition->size);
    ota_partition = esp_ota_get_next_update_partition(current_partition);
    assert(ota_partition != NULL);
    ESP_LOGI("part", "OTA partition: %s (%u)", ota_partition->label, ota_partition->size);
    uint8_t *current_sha256 = (uint8_t *)malloc(32);
    uint8_t *ota_sha256 = (uint8_t *)malloc(32);
    esp_partition_pos_t current_running_pos = {
        .offset = current_partition->address,
        .size = current_partition->size,
    };
    esp_partition_pos_t ota_running_pos = {
        .offset = ota_partition->address,
[... 589 lines cut ...]
            {
                first_memory_off_state = this_memory_off_state;
                last_memory_off_state = this_memory_off_state;
            }

            MDF_LOGE("Current memory on: %u, down %u since previous and %u since first", this_memory_on_state, last_memory_on_state - this_memory_on_state, first_memory_on_state - this_memory_on_state);
            MDF_LOGE("Current memory off: %u, down %u since previous and %u since first", this_memory_off_state, last_memory_off_state - this_memory_off_state, first_memory_off_state - this_memory_off_state);
            last_memory_on_state = this_memory_on_state;
            last_memory_off_state = this_memory_off_state;

            vTaskDelay(5000 / portTICK_PERIOD_MS);

            esp_wifi_start();
            mwifi_restart();

            while (mesh_on == false)
            {
                vTaskDelay(100 / portTICK_PERIOD_MS);
            }

            vTaskDelay(5000 / portTICK_PERIOD_MS);
        }
    }
}
`

MisterScience5, 2022-08-08:
I've attached my test code here which is a lightly modified version of the MDF no_router_example.

[no_router_example.zip](https://github.com/espressif/esp-idf/files/9280481/no_router_example.zip)

VitorBFreitas, 2022-09-15:
Noticed the same problem. Any one knows if this will have a fix any way soon?

pedrominatel, 2022-10-07:
Hi @Alvin1Zhang, do you have any update about this issue?

VitorBFreitas, 2022-11-10:
ESP-MDF says that any change in mesh configuration must be followed by mwifi_restart(). Otherwise, mwifi_restart() just do esp_mesh_stop() then esp_mesh_start(), wich has memory leak, so a lot of possible mesh applications are impracticable. Any update about this topic would be appreciated.

zhangyanjiaoesp, 2022-11-15:
@pedrominatel @MisterScience5 @Satierfa @VitorBFreitas 
We have reproduced this issue locally, and will find the root cause ASAP.

zhangyanjiaoesp, 2022-11-16:
The root cause has found, will backport the fix to v4.4 ASAP.

pedrominatel, 2022-11-16:
Thank you so much for your support, @zhangyanjiaoesp!

MisterScience5, 2022-11-19:
Awesome work, thanks @zhangyanjiaoesp :)

zhangyanjiaoesp, 2022-11-30:
The master branch is 7b35f80, and the v4.4 branch is 2302651

VitorBFreitas, 2022-11-30:
Thank you, @zhangyanjiaoesp.

closed 2023-04-25 by Alvin1Zhang