espressif/esp-idf #6352: Start in AP_STA mode with just an AP and then connect to STA later

bsalha1, 2021-01-03:
**Problem:**
I basically want to have an AP always running on my ESP32-S2 that runs a TCP server, accepting commands which alter the firmware, and then I want a command that connects to a STA. However, when I use AP_STA mode with esp-idf, it looks like I have to initialize both the STA and AP at the same time, but I initially don't have a STA to connect to, just an AP to start.

**Solution:**
Ability to start an AP, and later on be able to connect to a STA without restarting the AP.

**Alternatives:**
I started the ESP32-S2 in AP mode, and then when I wanted to connect to a STA, I stopped the wifi driver, and put it in AP_STA mode and remade the AP and then connected to the STA. However, remaking the AP aborts the TCP connection with my outside application, and I can't have that happen because upon connect to the STA, I want to send a packet back to the application consisting of the IP that the ESP32-S2 got when it connect to the STA.

**Note:**
I'm not sure if this is already possible or not - I haven't found any examples online.

bsalha1, 2021-01-03:
I actually fixed this by first initializing the netif's of the AP and STA, configuring the AP, starting the wifi driver, and then later on I could configure the STA and connect to it.

closed 2021-01-03 by bsalha1

Netskeh, 2021-01-14:
greetings, I have a similar application and you solution to this problem would be of interest. Could you paste some snippets of the solution you mentioned, along with the esp-idf version you are using?

bsalha1, 2021-01-15:
Hello, I'm using version 4.2, and here is the basic outline of how my code works: https://gist.github.com/bsalha1/2b8dcd0ccbf82499af2c4019e4463277

YouDONG-ESP, 2021-01-15:
@bsalha1 This code might gives you an idea. 

'''#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

static const char *TAG = "wifi ";
static esp_netif_t *netif_ap = NULL;
static esp_netif_t *netif_sta = NULL;
static EventGroupHandle_t wifi_event_group;

static void wifi_ap_mode(void)
{
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = "wifi1234",
            .ssid_len = 0,
            .max_connection = 4,
            .password = "12345678",
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
}

static void wifi_sta_mode(void)
{
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "wifi111",
            .password = ""
        }
     };

    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_APSTA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_connect() );
}

static void sta_connect_handler(void* arg, esp_event_base_t event_base,
                               int32_t event_id, void* event_data)
{
    printf("sta connected\n");
}

static void ap_connect_handler(void* arg, esp_event_base_t event_base,
                               int32_t event_id, void* event_data)
{
    printf("ap connected\n");
}

void initialise_wifi(void)
{
    ESP_ERROR_CHECK(esp_netif_init());
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_create_default() );
    netif_ap = esp_netif_create_default_wifi_ap();
    assert(netif_ap);
    netif_sta = esp_netif_create_default_wifi_sta();
    assert(netif_sta);
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        WIFI_EVENT_AP_STACONNECTED,
                                                        &ap_connect_handler,
                                                        NULL,
                                                        NULL));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        WIFI_EVENT_STA_CONNECTED,
                                                        &sta_connect_handler,
                                                        NULL,
                                                        NULL));                                                        

    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL) );
    ESP_ERROR_CHECK(esp_wifi_start() );
    wifi_sta_mode();
    wifi_ap_mode();
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    initialise_wifi();

}
'''

Netskeh, 2021-01-15:
thank you both. I got the general idea and indeed it works :)