espressif/esp-idf #6354: Using "FreeRTOS + POSIX" ?

joelguittet, 2021-01-03:
**Is your feature request related to a problem? Please describe.**

Today there is missing posix implementation, for example mqueue, etc. Support posix implementation have a great advantage  to build/use cross platform libraries and applications.

**Describe the solution you'd like**

FreeRTOS proposes https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_POSIX/index.html. Is it something you may consider to integrate in ESP-IDF ? Looks great.

**Describe alternatives you've considered**

I have not found other implementations to have posix support on ESP32.

I have tried to integrated the above source code myself but seems there is some part I do not clearly understand to integrate it without creating some conflicts.

**Additional context**

Nothing to add.

Alvin1Zhang, 2021-01-04:
Thanks for raising this feature request.

projectgus, 2021-01-04:
Hi @joelguittet,

Thanks for letting us know about this. ESP-IDF has native support for a lot of POSIX functionality already (including: time, pthreads, fcntl, errno) - some via newlib and some via ESP-IDF itself. We've heard some success stories of *nix-based firmware being ported without a lot of additional work needed.

Because there is already some functionality, integrating FreeRTOS+POSIX will be quite difficult without resolving a lot of conflicts - as you've no doubt found.

One thing we don't have at the moment is clear documentation on which parts of POSIX are or aren't supported in ESP-IDF, sorry about that. We will look at creating that document.

Apart from "mqueue" (which is not supported in ESP-IDF), is there anything else you've noticed is missing?

joelguittet, 2021-01-04:
Hello @projectgus 

You're welcome and thanks for the quick reply to this subject.

Actually, myself for a personal application I notice the following missing parts:

- "mqueue.h" : definition of "mqd_t", "mq_open", "mq_timedreceive", "mq_unlink", ...
- "semaphore.h" : definition of "sem_t", "sem_init", "sem_wait" ...
- and maybe some "fcntl.h" definitions: O_CREAT, O_RDWR (or they are not in the right header file)

Joel

projectgus, 2021-01-04:
Hi @joelguittet,

Thanks for the update.

Regarding mqueue.h & semaphore.h, we'll see what we can do about supporting these. I'm afraid I can't give you an ETA though.

`O_CREATE` & `O_RDWR` are defined in fcntl.h, you should be able to `#include <fcntl.h>` to use these. If you think this is a bug in ESP-IDF, please let us know more details and we'll try to fix it.

One thing I forgot to mention, FreeRTOS+POSIX is also part of Amazon FreeRTOS and ESP32 is a supported platform there as well. So another approach may be to switch to Amazon FreeRTOS.

Angus

joelguittet, 2021-01-05:
Hello @projectgus 

Regarding fcntl maybe I have something wrong, not sure on my side. The other dependencies are more important. i understand there is no ETA, but at least the point is mentioned and it's a great beginning. Will follow ESP-IDF evolutions in the future and in the meantime I will do a specific modification in my application.

Joel

0xjakob, 2022-03-21:
Linking #8594 here.

0xjakob, 2023-11-09:
The [POSIX API for unnamed semaphores](https://github.com/espressif/esp-idf/blob/master/components/newlib/platform_include/semaphore.h) has been added to IDF a while ago.

0xjakob, 2024-07-12:
@joelguittet It has been a long time... but we just merged POSIX message queues in our internal working repository. It should appear on master here on github very soon. POSIX unnamed semaphores have been working in ESP-IDF for a while now. For more information, please have a look at the [general POSIX documentation for ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/pthread.html).

joelguittet, 2024-07-12:
@0xjakob thanks for your message, it s always good to know this is progressing!
You can close the issue.

closed 2024-07-18 by Indastri