espressif/esp-idf #6355: WPA2 Enterprise example failing to connect

AReubens, 2021-01-03:
## Environment

- Development Kit:  None
- Module or chip used:  ESP32
- IDF version :   v4.2
- Build System:  CMake
- Compiler version:  8.4.0
- Operating System:  Linux
- Using an IDE?:  No
- Power Supply:  USB

## Problem Description

Running the example wpa2_enterprise _(examples/wifi/wpa2_enterprise)_ the module does no connect.
Looking at what the expected output is, it looks like there is something going on with the wpa since it's completely lacking in the logging and in it's place what appears to be something related to phy.

### Expected Behavior

Expecting a connection to be established with output akin to what is shown in the example:
#### Expected output
Here is an example of wpa2 enterprise(PEAP method) console output.

```
I (1352) example: Setting WiFi configuration SSID wpa2_test...
I (1362) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (1362) wifi: rx_ba=1 tx_ba=1

I (1372) wifi: mode : sta (24:0a:c4:03:b8:dc)
I (3002) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:11
I (3642) wifi: state: init -> auth (b0)
I (3642) wifi: state: auth -> assoc (0)
I (3652) wifi: state: assoc -> run (10)
I (3652) wpa: wpa2_task prio:24, stack:6144

I (3972) wpa: >>>>>wpa2 FINISH

I (3982) wpa: wpa2 task delete

I (3992) wifi: connected with wpa2_test, channel 11
I (5372) example: ~~~~~~~~~~~
I (5372) example: IP:0.0.0.0
I (5372) example: MASK:0.0.0.0
I (5372) example: GW:0.0.0.0
I (5372) example: ~~~~~~~~~~~
I (6832) event: ip: 192.168.1.112, mask: 255.255.255.0, gw: 192.168.1.1
I (7372) example: ~~~~~~~~~~~
I (7372) example: IP:192.168.1.112
I (7372) example: MASK:255.255.255.0
I (7372) example: GW:192.168.1.1
I (7372) example: ~~~~~~~~~~~
I (9372) example: ~~~~~~~~~~~
I (9372) example: IP:192.168.1.112
I (9372) example: MASK:255.255.255.0
I (9372) example: GW:192.168.1.1
I (9372) example: ~~~~~~~~~~~
```

### Actual Behavior
Appears that no connection is established
-Will continue to print IP, mask and Gateway as 0s for however long it's left
-Sporadic interjections of `wifi:state` and `wifi:new`

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (670) wifi:wifi driver task: 3ffc07c8, prio:23, stack:6656, core=0
I (670) system_api: Base MAC address is not set
I (670) system_api: read default base MAC address from EFUSE
I (690) wifi:wifi firmware version: 1865b55
I (690) wifi:wifi certification version: v7.0
I (690) wifi:config NVS flash: enabled
I (690) wifi:config nano formating: disabled
I (700) wifi:Init data frame dynamic rx buffer num: 32
I (700) wifi:Init management frame dynamic rx buffer num: 32
I (710) wifi:Init management short buffer num: 32
I (710) wifi:Init dynamic tx buffer num: 32
I (720) wifi:Init static rx buffer size: 1600
I (720) wifi:Init static rx buffer num: 10
I (730) wifi:Init dynamic rx buffer num: 32
I (730) wifi_init: rx ba win: 6
I (730) wifi_init: tcpip mbox: 32
I (740) wifi_init: udp mbox: 6
I (740) wifi_init: tcp mbox: 6
I (740) wifi_init: tcp tx win: 5744
I (750) wifi_init: tcp rx win: 5744
I (750) wifi_init: tcp mss: 1440
I (760) wifi_init: WiFi IRAM OP enabled
I (760) wifi_init: WiFi RX IRAM OP enabled
I (770) example: Setting WiFi configuration SSID ########...
I (880) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (880) wifi:mode : sta (4c:11:ae:a6:14:40)
I (1000) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (1000) wifi:state: init -> auth (b0)
I (1010) wifi:state: auth -> assoc (0)
I (1010) wifi:state: assoc -> run (10)
I (4880) example: ~~~~~~~~~~~
I (4880) example: IP:0.0.0.0
I (4880) example: MASK:0.0.0.0
I (4880) example: GW:0.0.0.0
I (4880) example: ~~~~~~~~~~~
...
```
## Other items if possible

I've double and triple checked credentials (SSID, username, identity and pass) and that it's using PEAP - All set from `idf.py menuconfig`
I've set up a fresh Ubuntu 20.04 vm and run the following commands which reproduces the error:
```
   1  git clone -b v4.2 --recursive https://github.com/espressif/esp-idf.git
   2  ls
   3  cd esp-idf/
   4  ls
   5  ./install.sh 
   6  . ./export.sh
   7  cd examples/wifi/wpa2_enterprise/
   8* 
   9  idf.py set-target esp32
   10  idf.py menuconfig
   11  idf.py build
   12  idf.py -p /dev/ttyUSB0 flash
   13  idf.py -p /dev/ttyUSB0 monitor
```

When programming the module through Arduino and uploading the WiFIClientEnterprise example the module connects.

Alvin1Zhang, 2021-01-04:
Thanks for reporting and letting us know, we will look into.

sagb2015, 2021-01-07:
@AReubens  Please help to provide radius server config.

AReubens, 2021-01-07:
@sagb2015 I'm afraid I can't, It's a university enterprise network (not eduroam)

I can provide captures and logs of successful connects with the ardiuno example and unsuccessful captures with the IDF example if that may help. Currently just checking if there is anything sensitive in them.

sagb2015, 2021-01-08:
@AReubens Okay please help to share as much information as can including sdkconfig, certs, captures since this is setup specific.

AReubens, 2021-01-10:
The capturing of what's going on seems to be inconsistent. While logging output is effectively the same, the Pcaps seem to differ a fair amount, the first attempt being the only one that looks like it makes some kind of attempt at PEAP.

[ESP-IDF.zip](https://github.com/espressif/esp-idf/files/5793367/ESP-IDF.zip)

nishanth-radja, 2021-02-10:
@AReubens ,Looks like the AP is not having any basic rates,Hence the STA fails to connect.Pls check the probe req from STA and Assoc response from AP.
In probe req we see the basic rates included (1,2,5.5 and 11) 
The assoc response from AP also does not include other basic rates like (1,2,5.5 and 11),Basic rate is set to 12Mbps.Can you pls retest with some other AP which supports at least one of the basic rates.

AReubens, 2021-02-14:
@nishanth-radja Thanks for looking into this, I've attached a few more captures, I moved about to get different APs in these captures, unfortunately it looks like all the AP's on this network use 12Mbps to communicate with this esp.
The initial probe requests sent out by the ESP indicate that 12Mbps is supported, is that for a different part of communication?

Captures 2 and 3 show some form of eap being initiated again, but again with no connection.

[Captures-Feb.zip](https://github.com/espressif/esp-idf/files/5977846/Captures-Feb.zip)

Also, side note, when capturing connection attempts from an esp32 to this network I can see a large amount of packet info, as shown in the captures, but when monitoring the connection attempt (and success) of an esp8266 I can only capture what is broadcast, I would have thought captures for both devices would be similar?

nishanth-radja, 2021-02-18:
@AReubens , Yes that is right,12Mbps is only for data communication after association,But basic Rates are mandatory as it will used for broadcast packets and other control frames.Without having at-least one common basic rate between AP and STA, The STA wont be able to successfully associate to the AP. After Association it can use higher data rates to transfer data.  

The EAP is being initiated by the AP,Since there is no common  basic rate,STA can't respond to it.Connection will not be successful without a common basic rate.

AReubens, 2021-02-21:
What gets me though is that older versions of idf are able to connect.
When programming through Arduino IDE using the default wpa2 sketch (WiFiClientEnterprise) from [arduino-esp32](https://github.com/espressif/arduino-esp32) the esp is able to connect.

I've been trying to use idf version 3.2 (which is what I think the arduino build it using?..) to connect, but I'm currently having an absolute mare with littlefs.

nishanth-radja, 2021-02-22:
@AReubens , can I get the capture of the successful enterprise connection with the older IDF you were using, We can check if the Network configurations were the same during the older successful connections and the present failure condition.

AReubens, 2021-02-22:
So I've attached two captures, the successful connection connected just now when programmed through Arduino.
The association responses look similar, only hardware info changing, but there is a difference in the association requests with the RSN and AMPDU values. 

[Comparision.zip](https://github.com/espressif/esp-idf/files/6022586/Comparision.zip)

nishanth-radja, 2021-02-23:
@AReubens I have gone through both the captures,I see that the STA has initially sent the deauth even in successful case and there are no EAP packets in the successful capture,If you had applied any filter can you remove them and share the packets during association along with the beacons of the AP.
I would like to know more about the network ,what is the model of the Cisco AP and is it controller based Network and what is the radius server you are using?

nishanth-radja, 2021-02-23:
@AReubens  Also can you enable the following logs using the Menuconfig in IDF and get the console logs for the failure case.
component config->log output->default log verbosity (to debug instead of info)
component config-> Enable Wifi debug log
Also can you pls attach the SDK config file for the failure case.

negativekelvin, 2021-02-26:
>support disable 11b rate

https://github.com/espressif/esp32-wifi-lib/commit/e1efaabef1ec5ea1240e04fce5c0c359281ec096

AReubens, 2021-03-05:
Attached a capture with the debug settings mentioned.

The oldsdkconfig folder contains a capture and log from a new device never seen before by the network, as is the arduino capture included. The debug capture was completed on a device that has attempted to connect to the network before - only noted because the amount of packets captured seems to be very different between the debug and non-debug-sdkconfig.

[DebugCaptures.zip](https://github.com/espressif/esp-idf/files/6092743/DebugCaptures.zip)

nishanth-radja, 2021-08-05:
@AReubens The fix for disabling the 11b rates are included in the latest master with this commit ID : commit 1e8d1344a1e548e9b6ffce5ebfa7883d7ad0e1b4 .
pls do a sub module update after moving to the latest master.

 Also pls enable these logs while trying the connection on the latest master.
1. Enable supplicant logs -> menuconfig->supplicant->print debug messages from wpa_supplicant
2. Log output to verbose -> menuconfig->log_output->Verbose
3. Also the sniffer capture of the full eap_tls handshake during the Adrino pass scenario.