espressif/esp-idf #6356: Upgrade json library

joelguittet, 2021-01-04:
**Is your feature request related to a problem? Please describe.**

I'm working on an application depending of cJSON.
The current ESP-IDF is fetching cJSON 1.7.12, but I need a later version because new API fonctions are available ("cJSON_GetNumberValue" in my example).

**Describe the solution you'd like**

Upgrade the library to the latest release which is v1.7.14 (instead of v1.7.12).

**Describe alternatives you've considered**

Forking ESP-IDF in my github and have the modification locally. I prefer this to be supported in the mainline (and it will benefit other people).

**Additional context**

Nothing more.

Alvin1Zhang, 2021-01-05:
Thanks for raising this feature request, thanks.

closed 2021-01-15 by espressif-bot

AxelLin, 2021-02-01:
Hi,
There are a couple null pointer dereference fixes after v1.7.14.
I think it worth update for the cjson component, please check below in cJson git tree:
$ git shortlog --no-merges 8e84db4c4e631c5578a1d2365bbcc9d814ab8a51..HEAD