espressif/esp-idf #6357: run unit-test in the order given in TEST_COMPONENTS

NikLeberg, 2021-01-05:
**Is your feature request related to a problem? Please describe.**

`TEST_COMPONENTS` in combination with `unity_run_all_tests()` runs tests in alphabetical file order.

I'm building an app with many custom components that I test with the provided unity-framework. e.g. based on [unit_test example](https://github.com/espressif/esp-idf/tree/master/examples/system/unit_test).
As I only want to test my custom components I set `TEST_COMPONENTS` to the components I'd like to test. As not all my component-tests are well separated from eachother, some rely on functionality of other components-under-test. I order the components-to-test in `TEST_COMPONENTS` accordingly.

**Describe the solution you'd like**

I'd like the unity-framework to run the testfiles in the order as given in `TEST_COMPONENTS`.
To do that the cmake skript [project.cmake](https://github.com/espressif/esp-idf/blob/39cbf2f7ed4810913031055065da08227020e8b4/tools/cmake/project.cmake#L252) in function `__project_init`, could be altered. Currently it loops over each component, looks up if it's in `TEST_COMPONENTS`. If it is, it's ./test folder is added as component. For doing what I propose one has to alter the skript to do kinda the reverse: loop over `TEST_COMPONENTS` and if it is a valid component, add the ./test folder als component. That way the order given in `TEST_COMPONENTS` is kept.

**Describe alternatives you've considered**

One could alter the way `TEST_CASE` registers test into unity e.g. use a priority number in the underlying call to `__attribute__((constructor))` but that is not straight forward.

**Additional context**

For my app I've changed [project.cmake](https://github.com/espressif/esp-idf/blob/39cbf2f7ed4810913031055065da08227020e8b4/tools/cmake/project.cmake#L252) with a dirty workaround:

```
    if(TESTS_ALL OR BUILD_TESTS OR TEST_COMPONENTS OR TEST_EXCLUDE_COMPONENTS)
        spaces2list(TEST_COMPONENTS)
        spaces2list(TEST_EXCLUDE_COMPONENTS)
        idf_build_get_property(component_targets __COMPONENT_TARGETS)
        foreach(test_component ${TEST_COMPONENTS})
            __component_get_property(component_dir ___idf_${test_component} COMPONENT_DIR)
            __component_get_property(component_name ___idf_${test_component} COMPONENT_NAME)
            __component_add(${component_dir}/test ${component_name})
            list(APPEND test_components ${component_name}::test)
        endforeach()
    endif()
```

Alvin1Zhang, 2021-01-06:
Thanks for raising this feature request.

0xjakob, 2021-01-06:
Hi @NikLeberg!

It sounds like you are talking about run-time dependencies, i.e., one test has to be run to put the component(s) into a state enabling the second test to run.

Speaking of unit tests in general, it is bad practice to have test cases depend on each other. If by any means possible, they should be independent from each other.
The way we usually solve this is by using "setup" and "tear-down" methods/macros.
You would define them once for a number of test cases and they would put the component into the desired state to test.
If you're using c++, you can also use a simple struct to let the destructor automatically call the tear down method.

Another alternative is, of course, putting all setup/tear down code into one test case.
The latter would also guarantee that code is run in the correct order.

I don't think we plan to guarantee the order of test cases at run time for the reason above.

But please let us know if "setup" and "tear-down" methods/macros aren't enough and you can't put all the code into one test case for whatever reason.

Thanks,
Jakob

NikLeberg, 2021-01-06:
Hey @0xjakob

I totaly get what you write and I agree that unit tests should be independent. But no, I'm not speaking of the run-time dependencies. For that I already use an setup/teardown construct.
My feature request is more a "visual enhancement", for the lack of a proper word.

My goal is, that the first test that fails, is the "real" fail. Like in gcc: the first error message is the most accurate and likely the "source", those errors that follow are often a side effect of the first one.
In the context of unit-tests I understand this as: If a component fails it should fail because of itself.

Now to my actual problem: Currently I see no way of mocking a custom component just for one single component-to-test (I think #5640 [IDFGH-3714] wants to implement that). If, lets say _componentA_ implements something that _componentB_ uses. I can without a problem test _componentA_ independently. But if I want to test _componentB_ I'd have to mock _componentA_ for the test to be independent. But I see no options to use the mocked-_componentA_ for the test of _componentB_, but at the same time use the "real"-_componentA_ for the tests of _componentA_.

Until that is possible, I use the ordering in `TEST_COMPONENTS` to implicitly test in the order of dependency. First _componentA_ and then _componentB_. That way, if _componentA_ has a failure, its at the top of the list and I know for sure it is at fault. Otherwise if _componentB_ would be tested first and fails because of a problem that _componentA_ had, its not as obvious who is to blame.

If there already is a way to enable mocked components on a "per component-to-test"-basis, please give me some pointers.

Have a great day!
Nik

0xjakob, 2021-01-07:
Hi @NikLeberg!

Thanks for the elaborate explanation.
I'm not too familiar with how the ordering works and whether there's an easy fix for that. I'm asking around internally...

> If there already is a way to enable mocked components on a "per component-to-test"-basis, please give me some pointers.

It's work in progress still, but a unit test for the `nvs_flash` component, mocking the `spi_flash` component is currently available, but only for a Linux host: https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/host_test/nvs_page_test.
It uses [CMock](https://www.throwtheswitch.org/cmock) to mock the `spi_flash` component.
You can see how the mocking works in this test's [source file](https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/host_test/nvs_page_test/main/nvs_page_test.cpp).

We don't have an example with CMock working on a target yet.
But we have an example with that kind of standalone (compared to the "usual" unit tests) test app which is based more purely on Unity: https://github.com/espressif/esp-idf/tree/master/components/esp_netif/test_apps.
I used mocking as a PoC in the latter test already but we won't add it since it wasn't well-designed. I can post a patch, however, if you're interested.

CMock has been integrated into IDF [as a component](https://github.com/espressif/esp-idf/tree/master/components/cmock) already. 

I'm also writing some documentation about CMock and unit tests on the host which will be merged into master during the next days/weeks.

If you want to follow the CMock approach, please let us know how far you get, what problems you encounter and where you get stuck. We definitely plan to use CMock and isolated unit testing more in the future. One problem with this is that we might change some aspects of the current mocking procedures in IDF when going to v5.0 as it's a young feature still under development. On the positive side, we can include feedback from you if you encounter issues.

Thanks,
Jakob

NikLeberg, 2021-01-20:
Hi @0xjakob

Thanks for providing this information.

Sadly I could not really follow how the mocking of `spi_flash` is done. Because it is very involved with many things that have to be set up.
My hope was that there was a way to "hide" a real-component or "overwrite" it on a per component basis. (See previous answer, third part).

Could you maybe show me the mocked PoC `esp_netif/test_apps` that you mentioned?

I'm looking forward to your documentation. If possible please hit me up when its online.

Thanks,
Nik

0xjakob, 2021-01-22:
@NikLeberg Please take this as a rough guideline. It should work, I rebased it on latest master. The way we introduce mocking might change a bit in the future, though.
[0001-nvs_flash-adding-mock-build.zip](https://github.com/espressif/esp-idf/files/5853927/0001-nvs_flash-adding-mock-build.zip)

NikLeberg, 2021-04-07:
Hi @0xjakob

I could sadly not get this to work with the mocking PoC you provided.
For now I use linker flags `-Wl,--wrap=functionToMock` to replace calls to other units of code that I dont want to test or that I want to mock. In the future I plan to replace unity with [cmocka](https://cmocka.org/). It has a very elegant mocking system. I might even get to separate the tests from the actual hardware.

Thanks

closed 2021-04-07 by NikLeberg

0xjakob, 2021-04-09:
@NikLeberg I'm sorry this didn't work out for you, but happy you found a solution. The testing/mocking in IDF still needs a lot of improvement and documentation. Let us know if you have any suggestions or encounter more problems.