espressif/esp-idf #6358: Random crashes in BT controller library

anszom, 2021-01-05:
## Environment

- Development Kit:      ESP32-PICO-Kit, various custom boards
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROOM-32, ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.2-beta1-202-g805a4d9b3, but see below
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Linux
- Using an IDE?: No; not relevant
- Power Supply:         USB, or custom board with a 12V->3.3V step-down

## Problem Description

I'm using ESP32 as a Bluetooth LE peripheral. I'm using nimble as the BT stack but it shouldn't matter. Under random circumstances, the BT controller library crashes.  I've reduced my code to a minimal example which is available [here](https://github.com/anszom/esp32-bt-crash-demo). To stress-test the BT interface, I'm using a custom Android application (that I unfortunately can't share), that randomly connects, disconnects, sends and receives data from the ESP32 BLE peripheral. Reproducing the bug takes up to several hours with my setup.

The issue doesn't occur with IDF v4.1. Using git bisect, I've identified the commit which introduces the problem - df301b89ece5325b5a0125135e96540156c890cb.

Below is an example crash log produced with ESP-IDF version df301b89ece5325b5a0125135e96540156c890cb

> ASSERT_PARAM(34240 1073429188), in ke_mem.c at line 409
> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
> 
> Core  0 register dump:
> PC      : 0x400848b9  PS      : 0x00060e34  A0      : 0x8001a04c  A1      : 0x3ffc6d80  
> 0x400848b9: r_assert_param at ??:?
> 
> A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000  
> A6      : 0x00000004  A7      : 0x3ffbdc08  A8      : 0x800848b4  A9      : 0x3ffc6d60  
> A10     : 0x00000000  A11     : 0x00000038  A12     : 0x00000014  A13     : 0xffffffff  
> A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
> EXCVADDR: 0x00000000  LBEG    : 0x400847ed  LEND    : 0x400847f4  LCOUNT  : 0x00000000  
> 0x400847ed: r_assert_param at ??:?
> 
> 0x400847f4: r_assert_param at ??:?
> 
> 
> Backtrace:0x400848b6:0x3ffc6d80 0x4001a049:0x3ffc6da0 0x4001ac01:0x3ffc6dd0 0x40045991:0x3ffc6df0 0x400855bd:0x3ffc6e10 0x40019d11:0x3ffc6e40 0x40055b4d:0x3ffc6e60 0x400e9977:0x3ffc6e80 0x400e9f35:0x3ffc6ea0 0x4008d73d:0x3ffc6ed0
> 0x400848b6: r_assert_param at ??:?
> 
> 0x400855bd: ke_task_schedule at ke_task.c:?
> 
> 0x400e9977: r_rw_schedule at ??:?
> 
> 0x400e9f35: btdm_controller_task at ??:?
> 
> 0x4008d73d: vPortTaskWrapper at /home/andrzej/Downloads/esp-idf-debug/components/freertos/xtensa/port.c:143
> 
> Core  1 register dump:
> PC      : 0x4011e1ae  PS      : 0x00060634  A0      : 0x800d3856  A1      : 0x3ffbcd30  
> 0x4011e1ae: esp_pm_impl_waiti at /home/andrzej/Downloads/esp-idf-debug/components/esp32/pm_esp32.c:486
> 
> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
> A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d37aa  A9      : 0x3ffbcd00  
> A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00060623  
> A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> Backtrace:0x4011e1ab:0x3ffbcd30 0x400d3853:0x3ffbcd50 0x4008e0e5:0x3ffbcd70 0x4008d73d:0x3ffbcd90
> 0x4011e1ab: esp_pm_impl_waiti at /home/andrzej/Downloads/esp-idf-debug/components/esp32/pm_esp32.c:484
> 
> 0x400d3853: esp_vApplicationIdleHook at /home/andrzej/Downloads/esp-idf-debug/components/esp_common/src/freertos_hooks.c:63
> 
> 0x4008e0e5: prvIdleTask at /home/andrzej/Downloads/esp-idf-debug/components/freertos/tasks.c:3386 (discriminator 1)
> 
> 0x4008d73d: vPortTaskWrapper at /home/andrzej/Downloads/esp-idf-debug/components/freertos/xtensa/port.c:143
>

anszom, 2021-01-05:
I forgot to mention - for the BT stress tests I'm using a Nexus 5 phone with the CrDroid ROM. The ROM has some bugs related to BT handling, and the BT controller on the phone occasionally crashes & restarts during the tests. It's possible that the BT crash on the phone triggers some unusual conditions which trigger the ESP32 bug - but I'm not sure if these are related. Anyway, it shouldn't be possible to crash the ESP32's bluetooth stack remotely.

Alvin1Zhang, 2021-01-06:
Thanks for reporting, we will look into.

h2zero, 2021-01-16:
I have a similar issue reported in #6417. 
Occurs when operating as a BLE client, so not likely related to the phone.