espressif/esp-idf #6359: ESP-IDF monitor not working properly

fahadalisarwar1, 2021-01-05:
I am trying to run the monitor command however i keep getting the following error. I am using VS code extension v6 with esp-idf version v4.2

PS D:\Coding\course-code\Chapter3-C_tutorial\example1-basics> C:\Users\fahad\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\fahad\esp\esp-idf\tools\idf.py -p COM3 monitor
Traceback (most recent call last):
  File "C:\Users\fahad\esp\esp-idf\tools\idf.py", line 788, in <module>
    main()
  File "C:\Users\fahad\esp\esp-idf\tools\idf.py", line 704, in main
    checks_output = check_environment()
  File "C:\Users\fahad\esp\esp-idf\tools\idf.py", line 68, in check_environment
    print_idf_version()
  File "C:\Users\fahad\esp\esp-idf\tools\idf.py", line 117, in print_idf_version
    version = idf_version()
  File "C:\Users\fahad\esp\esp-idf\tools\idf_py_actions\tools.py", line 54, in idf_version
    version = subprocess.check_output([
  File "C:\python38\lib\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\python38\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\python38\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\python38\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified

dobairoland, 2021-01-05:
Hi @fahadalisarwar1. Please try to run the following commands from both ESP-IDF Command Prompt and Vscode terminal.

```bash
idf.py --version
git --git-dir=$IDF_PATH/.git --work-tree=$IDF_PATH describe --tags --dirty
```

Asteliks, 2021-01-05:
this seems to be a duplicate of: espressif/vscode-esp-idf-extension#266

Asteliks, 2021-01-19:
has been fixed in v0.6.1 and can be closed

fahadalisarwar1, 2021-01-20:
Resolved in update v0.6.1

closed 2021-01-20 by fahadalisarwar1