espressif/esp-idf #6360: ESP IDF on VSCode Monitor Device bug

araafario, 2021-01-05:
After not using ESP-IDF for a while , i cant start monitoring device anymore , i think there is something wrong with the order of quotation mark , i have no idea to solve this one. I have tried to reinstalling ESP-IDF and delete its component , meanwhile downgrading to lower version doesnt solve this problem too . . . 

```
Windows PowerShell
PS D:\Coding\ESP32\bt_spp_initiator> $Env:Path = "
>> $Env:IDF_PATH = "
PS D:\Coding\ESP32\bt_spp_initiator> $Env:IDF_TARGET = "
>> $Env:PYTHON = "
PS D:\Coding\ESP32\bt_spp_initiator> $Env:IDF_PYTHON_ENV_PATH = "
>> cls
>> C:\Users\microsoft\.espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe D:\Users\microsoft\esp\esp-idf\tools\idf.py -p COM1 monitor
>> 
```

Asteliks, 2021-01-05:
this seems to be a duplicate of: https://github.com/espressif/vscode-esp-idf-extension/issues/266

closed 2021-01-06 by araafario

araafario, 2021-01-06:
My solve is just using command window and manually open

```

idf.py -p "port" monitor

```
If your CMD doesnt recognize idf.py , you should read this manual install https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html , just follow it all until monitor