espressif/esp-idf #6361: BLE Crash:"BT_BTC: btc_gatts_cb_param_copy_req 1 no mem"

BigSmileFace, 2021-01-06:
Environment

Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): release 3.3.4
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: win 10
Using an IDE: release 3.3.4
Power Supply: USB, or battery

Problem Description
I'm using ESP32 as a Bluetooth LE server.  an android device connects to the esp32. 
keep communication for 10 times /second;
and after about 30~300mins (the time is not fixed) ,esp32 will crashed.

` [0;31mE (572314) BT_BTC: btc_gatts_cb_param_copy_req 1 no mem		
 [0m		
 Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.		
 Core 0 register dump:		
 PC      : 0x401307f4  PS      : 0x00060c30  A0      : 0x8012b609  A1      : 0x3ffce2a0  		
 A2      : 0x3ffce2e4  A3      : 0x3fff15b0  A4      : 0x3ffc410c  A5      : 0x00000006  		
 A6      : 0x3ffc410c  A7      : 0x3ffb5bac  A8      : 0x00000000  A9      : 0x3ffce290  		
 A10     : 0x3ffce2a8  A11     : 0x3fff15b7  A12     : 0x3ffce2a0  A13     : 0x3ffce2ae  		
 A14     : 0x000000c0  A15     : 0x000000fd  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  		
 EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  		
 		`

Alvin1Zhang, 2021-01-06:
Thanks for reporting and letting us know, and sorry for inconvenience, we will look into.

BigSmileFace, 2021-01-06:
> Thanks for reporting and letting us know, and sorry for inconvenience, we will look into.

@Alvin1Zhang Thanks!!!

BigSmileFace, 2021-01-06:
and i tested only with lwip. keep sending data to server with lwip;
and the log shows that the memory changes :

> 15:26:11.957 ******TEST*****:Memory left = 94036 Largest_Free_Block: 47632
15:26:12.934 ******TEST*****:Memory left = 105544 Largest_Free_Block: 64800
15:26:13.907 ******TEST*****:Memory left = 100948 Largest_Free_Block: 64800
15:26:14.882 ******TEST*****:Memory left = 113812 Largest_Free_Block: 64800
15:26:15.863 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:16.841 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:17.819 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:18.795 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:19.764 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:20.735 ******TEST*****:Memory left = 116044 Largest_Free_Block: 64800
15:26:21.713 ******TEST*****:Memory left = 115920 Largest_Free_Block: 64800
15:26:22.686 ******TEST*****:Memory left = 102652 Largest_Free_Block: 64800
15:26:23.648 ******TEST*****:Memory left = 112328 Largest_Free_Block: 64800
15:26:24.618 ******TEST*****:Memory left = 112248 Largest_Free_Block: 64800
15:26:25.595 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:26.571 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:27.553 ******TEST*****:Memory left = 116000 Largest_Free_Block: 64800
15:26:28.532 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:29.506 ******TEST*****:Memory left = 116000 Largest_Free_Block: 64800
15:26:30.487 ******TEST*****:Memory left = 114420 Largest_Free_Block: 44928
15:26:31.470 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:32.454 ******TEST*****:Memory left = 83304 Largest_Free_Block: 53760
15:26:33.447 ******TEST*****:Memory left = 35696 Largest_Free_Block: 28792
15:26:34.433 ******TEST*****:Memory left = 104756 Largest_Free_Block: 64800
15:26:35.414 ******TEST*****:Memory left = 112244 Largest_Free_Block: 64800
15:26:36.387 ******TEST*****:Memory left = 113812 Largest_Free_Block: 64800
15:26:37.368 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:38.344 ******TEST*****:Memory left = 116036 Largest_Free_Block: 64800
15:26:39.325 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:40.299 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:41.277 ******TEST*****:Memory left = 116040 Largest_Free_Block: 64800
15:26:42.254 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:43.232 ******TEST*****:Memory left = 82636 Largest_Free_Block: 52932
15:26:44.219 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:26:45.193 ******TEST*****:Memory left = 93724 Largest_Free_Block: 64800
15:26:46.174 ******TEST*****:Memory left = 98064 Largest_Free_Block: 64800
15:26:47.157 ******TEST*****:Memory left = 109344 Largest_Free_Block: 64800
15:26:48.131 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:49.107 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:50.088 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:51.061 ******TEST*****:Memory left = 116044 Largest_Free_Block: 64800
15:26:52.041 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:53.020 ******TEST*****:Memory left = 116044 Largest_Free_Block: 64800
15:26:53.996 ******TEST*****:Memory left = 114560 Largest_Free_Block: 64800
15:26:54.974 ******TEST*****:Memory left = 92740 Largest_Free_Block: 61924
15:26:55.942 ******TEST*****:Memory left = 96704 Largest_Free_Block: 64800
15:26:56.919 ******TEST*****:Memory left = 111580 Largest_Free_Block: 64800
15:26:57.894 ******TEST*****:Memory left = 113688 Largest_Free_Block: 64800
15:26:58.877 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:26:59.856 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:00.836 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:01.820 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:02.799 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:03.779 ******TEST*****:Memory left = 116004 Largest_Free_Block: 64800
15:27:04.760 ******TEST*****:Memory left = 93624 Largest_Free_Block: 57508
15:27:05.744 ******TEST*****:Memory left = 80192 Largest_Free_Block: 48472
15:27:06.725 ******TEST*****:Memory left = 105528 Largest_Free_Block: 58016
15:27:07.703 ******TEST*****:Memory left = 101084 Largest_Free_Block: 64800
15:27:08.682 ******TEST*****:Memory left = 112228 Largest_Free_Block: 64800
15:27:09.657 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:10.636 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:11.604 ******TEST*****:Memory left = 116036 Largest_Free_Block: 64800
15:27:12.582 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:13.551 ******TEST*****:Memory left = 86972 Largest_Free_Block: 56636
15:27:14.526 ******TEST*****:Memory left = 116128 Largest_Free_Block: 64800
15:27:15.506 ******TEST*****:Memory left = 101492 Largest_Free_Block: 54888
15:27:16.489 ******TEST*****:Memory left = 79068 Largest_Free_Block: 49344
[... 26 lines cut ...]
15:27:43.020 ******TEST*****:Memory left = 98732 Largest_Free_Block: 64800
15:27:44.010 ******TEST*****:Memory left = 102648 Largest_Free_Block: 64800
15:27:44.991 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
[... cut ...]
15:28:00.645 ******TEST*****:Memory left = 116124 Largest_Free_Block: 64800
15:28:01.612 ******TEST*****:Memory left = 114552 Largest_Free_Block: 64800
15:28:02.585 ******TEST*****:Memory left = 108076 Largest_Free_Block: 64800
15:28:03.558 ******TEST*****:Memory left = 109344 Largest_Free_Block: 64800
15:28:04.535 ******TEST*****:Memory left = 110676 Largest_Free_Block: 64800
15:28:05.512 ******TEST*****:Memory left = 102652 Largest_Free_Block: 64800
15:28:06.499 ******TEST*****:Memory left = 110676 Largest_Free_Block: 64800

BigSmileFace, 2021-01-06:
Maybe the config of lwip can save memory?? 
could u give me some suggestion about the lwip config?

closed 2021-01-08 by BigSmileFace