espressif/esp-idf #6362: execute "./install.sh", but appear error， the error is : /usr/bin/env: "bash/r": No such file or directory

zhuolingmin, 2021-01-06:
I'm configuring esp32 environment in ubuntu, then i execute ./install.sh, the comman cann't execute , show follow error,
"/usr/bin/env: "bash/r": No such file or directory"
Does you  encountered this problem before， and you can solve this error?

dobairoland, 2021-01-06:
@zhuolingmin This looks like if you had Windows line endings on Linux. Are you using WSL or have you copied over ESP-IDF from a Windows machine or do you have configured `git` to transform line endings to `lfcr`?

A simple workaround could be to install `dos2unix` and run`dos2unix install.sh` but this doesn't solve the root cause.

zhuolingmin, 2021-01-06:
> @zhuolingmin This looks like if you had Windows line endings on Linux. Are you using WSL or have you copied over ESP-IDF from a Windows machine or do you have configured `git` to transform line [...]

so what i can do to solve this error?

dobairoland, 2021-01-06:
@zhuolingmin Please try `dos2unix` to confirm if that is the issue and answer my question. If you download it again on Linux then it might help. But if you have a wrong git config then you would end up with the same result.

zhuolingmin, 2021-01-06:
> @zhuolingmin Please try `dos2unix` to confirm if that is the issue and answer my question. If you download it again on Linux then it might help. But if you have a wrong git config then you would end up with the same result.

I have tested this method and it does not solve the problem

closed 2021-01-06 by dobairoland

zhuolingmin, 2021-01-06:
> I can reproduce your issue by transforming the install file to Windows line endings. The transformation back to Unix line endings solves the issue as shown in the following example.
> 
> ```shell
> $ unix2dos install.sh 
> unix2dos: converting file install.sh to DOS format...
> $ ./install.sh 
> /usr/bin/env: ‘bash\r’: No such file or directory
> $ dos2unix install.sh 
> dos2unix: converting file install.sh to Unix format...
> $ ./install.sh 
> Installing ESP-IDF tools
> Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
> ...
> ```
> 
> There are a couple of solutions.
> 
> 1. You can use `dos2unix` to fix your files. This might be inconvenient to do for the whole ESP-IDF.
> 2. You can download ESP-IDF again. You can do it from the releases page or by cloning the repo. If you used `git` when you got your copy of IDF then please read about `core.autocrlf` at https://www.git-scm.com/book/en/v2/Customizing-Git-Git-Configuration in order to avoid the issue again.

but i try to process it by this way,  but Just the same, the follow show what i execute:

`zlm@zlm:/mnt/hgfs/Share/esp-idf-c3$ dos2unix install.sh
dos2unix: 正在转换文件 install.sh 为Unix格式...
zlm@zlm:/mnt/hgfs/Share/esp-idf-c3$ ./install.sh
Installing ESP-IDF tools
/usr/bin/env: “python\r”: 没有那个文件或目录
zlm@zlm:/mnt/hgfs/Share/esp-idf-c3$ 
`

dobairoland, 2021-01-06:
> ./install.sh Installing ESP-IDF tools /usr/bin/env: “python\r”:

You fixed `install.sh` but now you have the same issue with the next file, I guess `tools/idf_tools.py`. Probably all of your files in ESP-IDF are in Windows format.