espressif/esp-idf #6363: WIFI: disable DHCP in AP mode raises : esp_netif_new: Failed to configure netif with config=...

chaabanemalki, 2021-01-06:
## Environment
- Module or chip used:     Custom board ESP32D2WDQ5 (revision 1)
- IDF version :      v4.1-352-g5d258af93c
- Build System:     idf.py
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows

## Problem Description

We are facing WIFI performance issues as well as WIFI connection stability and behaviors.
Our first step is to fix all warnings/ Errors regarding our WIFI code to setup WIFI.

We setup our WIFI to AP mode and stop dhcps to set a specific IP address then start WIFI. This works and I'm able to connect to my esp32 and do HTTP post/get commands, but I always get the following error:
` E (5404) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffd49b4 (config or if_key is NULL or duplicate key)` 

This might be the reason for loosing the specific IP address from time to time.

### Expected Behavior
No error about failing to setup the right config

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

    tcpip_adapter_init();
    esp_err_t res = esp_event_loop_create_default();

    esp_netif_t *wifi_ap = esp_netif_create_default_wifi_ap();
    res = esp_netif_dhcps_stop(wifi_ap);

    esp_netif_ip_info_t ip_info;

    IP4_ADDR(&ip_info.ip, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.gw, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

    res = esp_netif_set_ip_info(wifi_ap, &ip_info);
    ESP_LOGI(TAG, "esp_netif_set_ip_info(): 0x%X", res);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));

    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));

    wifi_config_t wifi_config = {
        .ap = {
            .max_connection = WIFI_AP_MAX_CONN_DEVICES,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK},
    };
    strcpy((char *)wifi_config.ap.ssid, _ssid);
    strcpy((char *)wifi_config.ap.password, _passwd);
    wifi_config.ap.ssid_len = strlen(_ssid);
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_set_protocol(ESP_IF_WIFI_AP, WIFI_PROTOCOL_11B));
    ESP_ERROR_CHECK_WITHOUT_ABORT(esp_wifi_start());
}

```

## Debug Logs

```
I (5064) wifi:wifi driver task: 3ffd6dfc, prio:23, stack:6144, core=0
I (5101) wifi:wifi firmware version: 04f00f3
I (5101) wifi:config NVS flash: enabled
I (5101) wifi:config nano formating: enabled
I (5102) wifi:Init data frame dynamic rx buffer num: 32
I (5106) wifi:Init management frame dynamic rx buffer num: 32
I (5112) wifi:Init management short buffer num: 32
I (5116) wifi:Init dynamic tx buffer num: 32
I (5121) wifi:Init static rx buffer size: 1600
I (5125) wifi:Init static rx buffer num: 10
I (5128) wifi:Init dynamic rx buffer num: 32
W (5133) event: handler already registered, overwriting
I (5264) wifi:mode : softAP (b4:e6:2d:80:13:9e)
I (5267) wifi:Total power save buffer number: 16
I (5268) wifi:Init max length of beacon: 752/752
I (5268) wifi:Init max length of beacon: 752/752
E (5271) esp_netif_objects: ifkey: WIFI_AP_DEF
V (5275) esp_netif_objects: esp_netif_next_unsafe 0
V (5280) esp_netif_objects: esp_netif_next_unsafe 0
E (5285) esp_netif_objects: ifkey: WIFI_AP_DEF
V (5290) esp_netif_objects: esp_netif_next_unsafe 0
V (5295) esp_netif_objects: esp_netif_next_unsafe 0
D (5300) esp_netif_objects: esp_netif_add_to_list 0x3ffdcb64
D (5306) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (5315) esp_netif_lwip: check: remote, if=0x3ffdcb64 fn=0x4010a19c
0x4010a19c: esp_netif_start_api at C:/DEV/esp/components/esp_netif/lwip/esp_netif_lwip.c:584

D (5321) esp_netif_lwip: esp_netif_start_api 0x3ffdcb64
D (5326) esp_netif_lwip: esp_netif_start_api Setting the lwip netif0x3ffdcbc4 UP
D (5335) esp_netif_lwip: DHCP server started successfully
V (5339) esp_netif_objects: esp_netif_next_unsafe 0
V (5344) esp_netif_objects: esp_netif_next_unsafe 0x3ffdcb64
V (5350) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffdcb64
D (5356) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (5362) WIFI_AP: WiFi AP Started
E (5366) esp_netif_objects: ifkey: WIFI_AP_DEF
V (5371) esp_netif_objects: esp_netif_next_unsafe 0
E (5376) esp_netif_objects: found the same ifkey
E (5381) esp_netif_lwip: esp_netif_get_handle_from_ifkey WIFI_AP_DEF
E (5388) esp_netif_objects: ifkey: WIFI_AP_DEF
V (5393) esp_netif_objects: esp_netif_next_unsafe 0
E (5398) esp_netif_objects: found the same ifkey // custom log that I added
E (5404) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffd49b4 (config or if_key is NULL or duplicate key)
```

Please, advice how to setup static IP address for WIFI AP.

Regards,
Chaabane

david-cermak, 2021-01-06:
Hi @chaabanemalki

> This error is due to the fact that the function `esp_netif_new()` finds the config already in place

Exactly! It was already created in the `tcpip_adapter_init()` call, together with other default interfaces available in the `tcpip_adapter` API.

The `tcpip_adapter` API, as well as the `esp-netif` API could be both used in the `v4.x` but they could not be combined. Please check the related documentation:
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html
* https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/tcpip_adapter_migration.html

Replacing the `tcpip_adapter_init();` with `esp_netif_init()` at the first line of your `app_main()` should fix the problem

chaabanemalki, 2021-01-07:
Thank you @david-cermak  for your quick response.
Changing tcpip_adapter_init() will take some effort and validation on our side since we used it with cellular device as well.

I was wondering if there is a way to stop the dhcps and change the IP address  and still keep  using tcpip_dapater_init(). Bascically doing this code :

    res = esp_netif_dhcps_stop(wifi_ap);

    esp_netif_ip_info_t ip_info;

    IP4_ADDR(&ip_info.ip, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.gw, 192, 168, 4, 1);
    IP4_ADDR(&ip_info.netmask, 255, 255, 255, 0);

    res = esp_netif_set_ip_info(wifi_ap, &ip_info);

david-cermak, 2021-01-07:
@chaabanemalki Yes, it is possible, but again, you have to use the `tcpip_adapter` API, not the `esp_netif` API , when initialising the stack with `tcpip_adapeter_init()`:
``` c
    tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);
    tcpip_adapter_ip_info_t ip_info;
    ip_info.ip.addr = ipaddr_addr("192.168.4.1");
    ip_info.gw.addr = ipaddr_addr("192.168.4.1");
    ip_info.netmask.addr = ipaddr_addr("255.255.255.0");
    tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &ip_info);
```

chaabanemalki, 2021-01-08:
Thank you @david-cermak for your quick help. I really appreciate it.

Now everything is working as expected. I'll close the issue

closed 2021-01-08 by chaabanemalki