espressif/esp-idf #6366: PCNT units 4-7 not available on esp32

TinkerBear, 2021-01-06:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     macOS
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: VSCode
- Power Supply:         USB

## Problem Description

PCNT units 4-7 are not available on esp32 after changes for esp32s2.  Number of units is defined in esp-idf/components/soc/soc/esp32/include/soc/pcnt_caps.h as SOC_PCNT_UNIT_NUM, which affects enum constants in esp-idf/components/soc/include/hal/pcnt_types.h

First problem: No other header file includes pcnt_caps.h, and it is not mentioned in documentation, so client-side code will never have the constants for all 8 units available.

Second problem: esp-idf/components/driver/pcnt.c includes pcnt_caps.h, but only after pcnt_types.h, so it has no effect, and the compiled code will not support units 4-7.  ("soc/pcnt_caps.h" must move to before "driver/pcnt.h".)

### Expected Behavior

That PCNT units 4-7 work as usual on esp32

### Actual Behavior

Compile or runtime errors for units 4-7

### Steps to reproduce

1) Open examples/peripherals/pcnt/main/pcnt_example_main.c
2) Replace "PCNT_UNIT_0" with "PCNT_UNIT_4"
3) Attempt to compile
4) Compile fails

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
PCNT_UNIT_4
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2021-01-07:
Thanks for reporting.

ESP-Marius, 2021-01-27:
@TinkerBear This has been merged into release/v4.2 with 10f16c5d727c548a2e07 and should be available the next time we sync with Github. The issue will be automatically updated when this happens.

I've also merged a similar commit to master which explicitly includes `x_caps.h` a bunch of places. 

Thanks for reporting this!

Alvin1Zhang, 2021-03-16:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/5e9783f4a6ffd96e016a6654a7bce2d12c5f6751, feel free to reopen. Thanks.

closed 2021-03-16 by Alvin1Zhang