espressif/esp-idf #6367: Win64 toolkit installer fails

bentterp, 2021-01-07:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): 4.2
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No|
- Power Supply:         USB

## Problem Description

The installer terminates with an error message

### Expected Behavior

The installer finishes without errors

### Actual Behavior

```
2021-01-07 04:57:08.973   Running command: C:\Users\bent\miniconda3\python.exe "C:\Users\bent\Desktop\esp-idf\tools\idf_tools.py" --idf-path C:\Users\bent\Desktop\esp-idf install
                          Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
                          Installing xtensa-esp32-elf@esp-2020r3-8.4.0
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\bent\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          ERROR: Failed to download, and retry count has expired
                          Traceback (most recent call last):
                            File "C:\Users\bent\Desktop\esp-idf\tools\idf_tools.py", line 1492, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\bent\Desktop\esp-idf\tools\idf_tools.py", line 1488, in main
                              action_func(args)
                            File "C:\Users\bent\Desktop\esp-idf\tools\idf_tools.py", line 1181, in action_install
                              tool_obj.download(tool_version)
                            File "C:\Users\bent\Desktop\esp-idf\tools\idf_tools.py", line 598, in download
                              raise DownloadError()
                          __main__.DownloadError
```
### Steps to reproduce

1. download the installer v2.3 and run it
2. select IDF verion 4.1 or release/4.2

### Code to reproduce this issue

N/A

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Note that I can open the url  https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip without problems and the file downloads as expected.

georgik, 2021-01-07:
@bentterp Thank you very much for detailed report.
The issue is caused by Anaconda Python which is not able to access HTTPS.
```
... C:\Users\bent\miniconda3\python.exe ...
...Download failure <urlopen error unknown url type: https>
```
Suggested solution:
Please install Python 3.7 from python.org and repeat the installation with Python 3.7.

Note: Python 3.7 for Windows is the most realiable right now and it has all necessary wheels for all major releases of IDF.

JosueCom, 2021-01-07:
Can an option be added to change which python interpreter used? I do a lot of work with machine learning so I cannot get rid of Anaconda.

JosueCom, 2021-01-08:
@bentterp 
It seems that installing the ESP-IDF via vscode seems to fix all the issues. It lets you chose the python interpreter unlike the toolkit

georgik, 2021-02-02:
@bentterp Update: The new versions of installers is available. It can co-exist with Anaconda environment. The solution also contains correction of HTTPS communication in case of missing root certificates.

- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

Alvin1Zhang, 2021-04-16:
Thanks for reporting, the fix is available here https://github.com/espressif/esp-idf/commit/b51a7e0cc7957a301fc4837b85677601acf81a14, feel free to reopen.

closed 2021-04-16 by Alvin1Zhang