espressif/esp-idf #6369: esp_wifi_80211_tx() completely stops working when called continuously in a loop

doragasu, 2021-01-07:
## Environment

- Development Kit: ESP-WROOM-32 NodeMCU
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I need to send raw packets with function `esp_wifi_80211_tx()` as fast as possible. Currently if I try sending in a loop as fast as possible, sending fails, the function `esp_wifi_80211_tx()` always returns error and does not recover, it does not send a single frame anymore.

Documentation for this function states:
```
* @attention 4. Only after the previous packet was sent, entered the sent callback,
* the next packet is allowed to send. Otherwise, wifi_send_pkt_freedom
* will return fail.
```

So I would expect that when called too fast, the function would fail and return error, but eventually work if I keep retrying. That is not happening, once this function fails one time, it is not able to send frames anymore, successive calls only returns with error.

### Expected Behavior

The function `esp_wifi_80211_tx()` returns error when called too fast, but retrying, it is able to eventually send the frame. Also I would expect the documentation to explain how to guess when the "sent callback" has been entered (in case this information is available).

### Actual Behavior

The funtion `esp_wifi_80211_tx()` returns error when called too fast, and does not recover, retries are never able to send the frame. Also it seems documentation does not explain how to know when the "sent callback" has been entered.

### Steps to reproduce

1. Configure the device as STA and associate to an AP. The AP shall be preferably configured in 802.11b mode with a rate of 1Mbps, in order for the frames to longer to be sent and reproduce the bug easier. NOTE: It should be easier to reproduce the issue by alternatively configuring the device as AP with no stations connected (this way you guarantee that frames will be sent in 802.11b mode at 1Mbps), but it seems currently this does not work, as I reported in [this issue](https://github.com/espressif/esp-idf/issues/6368).
2. Send frames as big as possible (e.g. 1300 byte-long) and as fast as possible (in a loop) using `esp_wifi_80211_tx()` function.
3. Check for the function return to verify if it works or has failed. When it fails once, it will always fail until the device is rebooted.

### Code to reproduce this issue

Code is a modified version of https://github.com/Jeija/esp32-80211-tx
The modifications are to configure in STA mode instead of AP, to use the new event handler, to increase the length of the sent frames to 1300 bytes and to comment out the delay to send as frequently as possible.

If using this code _make sure you change the SSID and password as appropriate_. Also configure the router in 802.11b mode and 1Mbps rate.

```c
#include "freertos/FreeRTOS.h"

#include "esp_system.h"
#include "esp_event.h"
#include "esp_wifi.h"

#include "nvs_flash.h"
#include "string.h"
#include "esp_log.h"

/*
 * This is the (currently unofficial) 802.11 raw frame TX API,
 * defined in esp32-wifi-lib's libnet80211.a/ieee80211_output.o
 *
 * This declaration is all you need for using esp_wifi_80211_tx in your own application.
 */
esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);

uint8_t beacon_raw[] = {
	0x80, 0x00,							// 0-1: Frame Control
	0x00, 0x00,							// 2-3: Duration
	0xff, 0xff, 0xff, 0xff, 0xff, 0xff,				// 4-9: Destination address (broadcast)
	0xba, 0xde, 0xaf, 0xfe, 0x00, 0x06,				// 10-15: Source address
	0xba, 0xde, 0xaf, 0xfe, 0x00, 0x06,				// 16-21: BSSID
[... 168 lines cut ...]
26 8 01 Never gonna give you up
E (944) raw tx: tx failed
27 8 02 Never gonna let you down
E (954) raw tx: tx failed
25 8 03 Never gonna run around
E (954) raw tx: tx failed
17 8 04 and desert you
E (964) raw tx: tx failed
27 8 05 Never gonna make you cry
E (974) raw tx: tx failed
26 8 06 Never gonna say goodbye
E (974) raw tx: tx failed
25 8 07 Never gonna tell a lie
E (984) raw tx: tx failed
15 8 08 and hurt you
E (984) raw tx: tx failed
26 8 01 Never gonna give you up
E (994) raw tx: tx failed
27 8 02 Never gonna let you down
E (1004) raw tx: tx failed
25 8 03 Never gonna run around
E (1004) raw tx: tx failed
```

If the delay in line 64 is restored, and set e.g. to 100 ms, the code does not fail anymore (but obviously rate is sub optimal).

Alvin1Zhang, 2021-01-08:
Thanks for reporting and sorry for the inconvenience, we will look into.

YouDONG-ESP, 2021-01-08:
@doragasu Thanks for reporting, It's a bug in our wifi lib. It has been fixed, a repair patch will be released soon.

doragasu, 2021-01-08:
Great, thanks for support!

Alvin1Zhang, 2021-02-02:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/f0b0418ed412c94114f0ed9176494f3e2c19a9a9, feel free to reopen.

closed 2021-02-02 by Alvin1Zhang