espressif/esp-idf #6374: OTA | Total image size + esp_restart

Oleg-Perevyshin, 2021-01-08:
Hello everyone!
I read the forum and couldn't find an answer.

Question 1. How can I get the size of the BIN file of the firmware when updating the OTA?
What I need is not the size at the moment esp_https_ota_get_image_len_read (https_ota_handle), but the total file size.
I want to see the percentage of the progress of the update on a similar calculation (currentSize * 100 / totalSize).

Question 2. The update is successful, but the esp_restart (); does not reset the chip. If you reboot with the Reset button, a new firmware is launched.

negativekelvin, 2021-01-10:
These questions should be posted on the forum, are you able to post there or are you waiting for your forum account to be approved?

1. try esp_http_client_get_content_length it depends on the if server provides it
2. Yes

Oleg-Perevyshin, 2021-01-12:
There is one thread on the forum, but it does not provide an answer on how to get the full image size.
It makes sense to help here, and when I debug and make a working version, I will add an answer on the forum.
This is my update function.
```c
void updateCore() {
  esp_err_t ota_finish_err = ESP_OK;
  esp_err_t ota_begin_err = ESP_OK;
  esp_http_client_config_t otaClientConfig = {
    .url = OTA_PATH_HTTPS_CORE,
    .cert_pem = (char *)server_cert_pem_start,
    .timeout_ms = 5000
  };
  esp_https_ota_config_t otaConfig = {
    .http_config = &otaClientConfig
  };
  esp_https_ota_handle_t https_ota_handle = NULL;
  ota_begin_err = esp_https_ota_begin(&otaConfig, &https_ota_handle);
  if (ota_begin_err != ESP_OK) {
    goto ota_end;
  }
  char strPercent[30];
  long int oldPercent = 0;
  long int newPercent = 0;
  while (1) {
    ota_begin_err = esp_https_ota_perform(https_ota_handle);
    if (ota_begin_err != ESP_ERR_HTTPS_OTA_IN_PROGRESS) {
      break;
    }
    newPercent = ((esp_https_ota_get_image_len_read(https_ota_handle) * 100) / 830000);
    if (newPercent != oldPercent) {
      oldPercent = newPercent;
      sprintf(strPercent, "%ld%%", newPercent);
      sendUART("OK!", "Update:", strPercent);
    }
  }
  if (esp_https_ota_is_complete_data_received(https_ota_handle) != true) {
    // OTA image was not fully received
  }

ota_end:
  ota_finish_err = esp_https_ota_finish(https_ota_handle);
  if ((ota_begin_err == ESP_OK) && (ota_finish_err == ESP_OK)) {
    sendUART("OK!", "OTA", "Update Successful! Rebooting...");
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    esp_restart();
  } else {
    sendUART("ER!", "OTA", "Failed Update");
  }
}
```
Suppose the server gives the image size.
How can I get it here and use it with `esp_http_client_get_content_length`?

AxelLin, 2021-01-12:
> There is one thread on the forum, but it does not provide an answer on how to get the full image size.
> It makes sense to help here, and when I debug and make a working version, I will add an answer on the forum.

No, this is not a good excuse to misuse the issue tracking system here.
But I understand it's frustrating if no response from the forum.

Oleg-Perevyshin, 2021-01-12:
I don't understand the purpose of the company is to make money. In order to earn them, you need to attract more customers. And if there is insufficient support, and it is not possible to solve the problem, then customers will start to turn away.
On the forum, however, there is no answer, but people ask a lot of questions. Many people need a solution to get the full file size.

Give a push at least to a solution, because the OTA part stops in development.

shubhamkulkarni97, 2021-01-12:
@Oleg-Perevyshin, I have pushed a branch to demonstrate use of esp_http_client_get_content_length from event_handler. Please refer to [this](https://github.com/shubhamkulkarni97/esp-idf/tree/feature/get_content_length) branch

Oleg-Perevyshin, 2021-01-12:
> @Oleg-Perevyshin, I have pushed a branch to demonstrate use of esp_http_client_get_content_length from event_handler. Please refer to [this](https://github.com/shubhamkulkarni97/esp-idf/tree/feature/get_content_length) branch

And where to look, can you tell?
Can't find an example of use.

shubhamkulkarni97, 2021-01-12:
> And where to look, can you tell?
> Can't find an example of use.

Please try running advanced_https_ota example from specified branch. You must be able to get content length in [content_len](https://github.com/shubhamkulkarni97/esp-idf/blob/52c16a7c9d5413e0571c5d0f058f6a0021a4e76c/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c#L40) variable in event handler

Oleg-Perevyshin, 2021-01-12:
Thank you so much!

closed 2021-01-12 by Oleg-Perevyshin