espressif/esp-idf #6375: Wifi connection got worse after switching from v4.0.1 to v4.0.2

scyomantion, 2021-01-09:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    v4.0.1 and v4.0.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0 (v4.0.1)
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0 (v4.0.2)
- Operating System:     macOS
- Using an IDE?: No (Sublime 3)
- Power Supply:         USB (iPhone charger)

## Problem Description

I have a custom board with ESP32-WROOM-32 module that is powered by LiPo or USB. For this test I powered it with USB. 
The code currently do following:

- reads the Si7021 temperature sensor
- connects to wifi
- connect MQTT over TCP
- sent the temperature, humidity and last online time
- disconnects MQTT and Wifi
- get online time with esp_timer_get_time() and save it in last online time
- go to deep sleep for 30 seconds. 

I build the project with idf.py. Following configuration values are changed from standard values:
```
CONFIG_PARTITION_TABLE_TWO_OTA=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE=y
CONFIG_OPTIMIZATION_LEVEL_RELEASE=y
CONFIG_ESP32_DEBUG_OCDAWARE=n
CONFIG_ESP32_DEBUG_STUBS_ENABLE=n
CONFIG_LWIP_DHCP_RESTORE_LAST_IP=y
CONFIG_LWIP_DHCP_DOES_ARP_CHECK=n
```
Furthermore the console logging is disabled completely to save time.

With version v4.0.1 I get very constant online times of around 400ms. After upgrading to v4.0.2 this time goes up to a few seconds every few connections. I visualized it in a graph comparing the two versions:

[image]

There are also very few spikes with V4.0.1 but not as much as with v4.0.2. 

Accesspoint is Unifi AP-AC-LR but the ESP32 is in the same room. Encryption is WPA2. 

### Expected Behavior

Wifi stability should be the same as with v4.0.1

### Actual Behavior

Longer wifi times

scyomantion, 2021-01-09:
I created a minimal example to reproduce it. You can find it at https://github.com/scyomantion/sensor-test

Simply checkout idf version v4.0.1 or v4.0.2, change wifi settings, mqtt address and build the example. 

There are 7 timers in this example
```
1: after initialize nvs
2: after wifi connect
3: after mqtt connect
4: after subscribed
5: after published message
6: after mqtt block
7: after disconnect mqtt + wifi
```

this is the result I get with v4.0.1
[image]

this is the result with v4.0.2
[image]

or side by side:
[image]

Alvin1Zhang, 2021-01-11:
Thanks for reporting and sorry for the inconvenience, we will look into.

scyomantion, 2021-01-16:
I also tried master branch yesterday and it seems to be the same.

YouDONG-ESP, 2021-02-26:
Hi @scyomantion, we have found the cause, it's the problem from the router's AMPDU strategy, the packet may be blocked in the wifi layer when the quantity is few which matches your scenario. I suggest you turn off AMPDU (menuconfig->component config-> wi-fi -> wifi AMPDU TX/RX) and this problem should be solved.

scyomantion, 2021-03-01:
Hi @YouDONG-ESP , I tested to disable the AMPDU setting and it seems that the problem is solved. Thank you very much!

closed 2021-03-01 by scyomantion