espressif/esp-idf #6376: Change BTA_GATTC_NOTIF_REG_MAX through the menuconfig

BillBernacchi, 2021-01-09:
BTA_GATTC_NOTIF_REG_MAX in file bta_gattc_int.h is currently set to 7. I would like to increase this value. My problem is that when future versions of the IDF are released, I will have to remember to modify this define. Adding it as a parameter to the menuconfig would solve this issue.

BillBernacchi, 2021-01-14:
In addition to the above request, I would also like the following defines to be added as well. I will need to change these as well and would have the same issue as described above.
ESP_GATT_ATTR_HANDLE_MAX in esp_gatt_defs.h
BTA_GATTC_MAX_CACHE_CHAR in esp_gatt_defs.h

xiewenxiang, 2021-03-21:
Hi @BillBernacchi

Because there are maybe many similar macro definitions in IDF, if they are all configured in menuconfig, it will be very complicated. 

If doing so makes your development easier, can you maintain a branch yourself? This branch adds support for these configuration items. If the IDF version is updated in the future, you can also rebase to the latest branch.

aircable, 2022-06-15:
I second this again as well. We have BLE devices that need more than 7 notifications enabled.
The line in the source shows, that is on the latest MASTER branch.
#define BTA_GATTC_NOTIF_REG_MAX     7//15
Why not simply keep it at 15 without commenting it out.

benedikt-bartscher, 2023-04-12:
I also have many BLE devices which do not work caused by this limit

Hamberthm, 2023-06-24:
> I also have many BLE devices which do not work caused by this limit

Hi, I'm porting a BLE project and also had this issue. But after investigating I can confirm that BTA_GATTC_NOTIF_REG_MAX is apparently now tied to BTA_GATTC_CONN_MAX, at least from ESP-IDF v4.4.

As in **bta_gattc_int.h**, line 323:

`#define BTA_GATTC_NOTIF_REG_MAX     BTA_GATTC_CONN_MAX`

I can confirm that this is configurable via SDKCONFIG. Changing "BT/BLE MAX ACL CONNECTIONS(1~9)" setting from the default "4" to "9" solved the issue.

Maybe ESPRESSIF could set the default to 9, I don't know why it is set to 4, but apparently it's as easy as set it to 9.

cmbpereira, 2023-08-28:
HI, I'm porting the BLE project and have problems because of the same change.

`#define BTA_GATTC_NOTIF_REG_MAX     BTA_GATTC_CONN_MAX`

And these are two different things:
`BTA_GATTC_CONN_MAX` Is the number max of peripherals we can connect to the server.
`BTA_GATTC_NOTIF_REG_MAX` Is the number of notifications each connection supports.

My suggestion is to create an independent configuration for the BTA_GATTC_NOTIF_REG_MAX and don't use the CONFIG_BT_ACL_CONNECTIONS to this macro.

pedromarinho94, 2023-10-04:
I was forced to create a fork of esp-idf due to the need to increase `BTA_GATTC_NOTIF_REG_MAX`.

Basically now this is attached to the number of connections, which does not make sense.

Is there already a PR to fix this?

Hamberthm, 2024-08-19:
This is still an issue. I'm having a hard time adding multi HID device support for my project, as I need a mouse + keyboard BLE combo connected at the same time.

The HID stack will connect to both and automatically try to subscribe to all notifications in both devices, this will rapidly overcome the subscription limit and leave one of the two devices unoperational.

Hamberthm, 2024-08-20:
Sorry, it seems to have been fixed last year and I didn't notice. Installed v5.3.0 and now I can config max notifications number in project settings. All works well now. (See https://github.com/espressif/esp-idf/commit/6f400e1b5d3b6db54620cbea6efbd2e40e8f55d9)

@BillBernacchi I think you should close this!