espressif/esp-idf #6377: Odd parity not available for Modbus RTU slave

devward, 2021-01-10:
Hi,

Odd parity is not supported in slave rtu mode (maybe master as well, I didn't check). When trying to set the parity to UART_PARITY_ODD the esp32 fails: 
`MB_CONTROLLER_SLAVE: mbc_serial_slave_setup(79): mb wrong parity option = (0x3)`

I saw in the last commit of the files `esp-idf/components/freemodbus/serial_slave/modbus_controller/mbc_serial_slave.c` and `esp-idf/components/freemodbus/port/portserial.c` the code checks that the parity is not higher than UART_PARITY_EVEN which would be 0x02 in lines 78 and 187 respectively. I modified the checks and ran the app again, got no error but it wouldn't work with odd parity.

Commit: `66fb5a29bbdc2482d67c52e6f66b303378c9b789 `

What is the reason for no odd parity?

alisitsyn, 2021-01-11:
Hi  @devward,

Thank you for this issue. This happens because of incorrect propagation of parity options from modbus controller to `xMBMasterPortSerialInit() (xMBPortSerialInit)` function due to changes in the UART driver. In order to keep compatibility with previous code and fix this issue I propose the patch attached:

[0002_fix_mb_controller_parity_ussue.patch.log](https://github.com/espressif/esp-idf/files/5795446/0002_fix_mb_controller_parity_ussue.patch.log)

Please check if this fixes the issue. Thanks.

alisitsyn, 2021-01-26:
Hi @devard,

Could you check the fix and confirm that it works in your case? 
The MR is reviwed and will be ready to be merged as soon as you confirm the result.

devward, 2021-01-26:
Hi @alisitsyn 

Sorry for my late reply, I only tested it on one device and it worked but as we remove odd parity from our latest release we haven't tested it properly.

Cheers

alisitsyn, 2021-01-27:
@devward ,
Thank you for update.

closed 2021-03-12 by espressif-bot