espressif/esp-idf #6379: BLE stopped working after using ESP for long.

ROYMURLI, 2021-01-11:
IDFv- 4.2
OS- Ubuntu

I have included BLE GATTC, GATTS and wifi in my project but the device BLE stops working if it is powered on for long (sometimes after an hour or 2 hour or 4 hours randomly). SDK config is attached below. I have raised similar issue but with different IDFv 4.3 https://github.com/espressif/esp-idf/issues/6073#issuecomment-743028301 . I have downloaded the latest master of v4.2 but in this facing the BLE stuck issue which is different from IDFv 4.3. It is difficult to get BLE logs of a particular time where its stops working but I am trying to get it ASAP. 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5794439/sdkconfig.txt)

ROYMURLI, 2021-01-18:
I am getting series of ELx40 and ELx200 and device got stuck after these. BLE stopped working and also other task stopped working. 

```
2021-01-16 21:51:14ELx200
2021-01-16 21:51:14ELx200
2021-01-16 21:51:14ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx200
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15scan_evt timeout
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15ELx40
2021-01-16 21:51:15ELx40
```

prasad-alatkar, 2021-01-21:
@ROYMURLI Can you please share the commit head ? I can see the fix being merged into `release/v4.2` with commit id: 98a53689. Can you please confirm the issue was apparent even with this commit ?

ROYMURLI, 2021-01-21:
[image]

Hi @prasad-alatkar  The screenshot is attached here.
IDFv- v4.2-238-g8cd16b60f

blueMoods, 2021-02-05:
@ROYMURLI Please call this function after `esp_bt_controller_enable()`
```
extern int esp_bt_controller_set_log_level(int level);
esp_bt_controller_set_log_level(0);
```
Then test the `device got stuck` issue.

Alvin1Zhang, 2021-03-16:
@ROYMURLI Thanks for reporting, would you please help if any updates for this issue? Thanks.

ROYMURLI, 2021-03-16:
@Alvin1Zhang , Currently not facing this issue after updating the above lines in my code. Thanks for your help.

closed 2021-03-16 by ROYMURLI