espressif/esp-idf #6380: esp_hid_host example crashes on stack overflow when receiving rapid inputs

LMBernardo, 2021-01-11:
- [ X ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ X ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ X ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      [ ESP32-DevKitC ]
- Kit version (for WroverKit/PicoKit/DevKitC): [ v1 ]
- Module or chip used:  [ ESP32-WROOM-32 ]
- 
- IDF version (run ``git describe --tags`` to find it):
    // v4.2-198-g2aafcd9b1
       ( What I'm currently using, but was also happening in 4.3 )
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0

- Operating System:     [ Windows ]
- (Windows only) environment type: [ PowerShell ].
- Using an IDE?: [ VSCode + ESP-IDF Plugin ]
- Power Supply:         [ USB ]

## Problem Description

The esp_hid_host example provided with the IDF is susceptible to a stack overflow on rapid events. The stack overflow occurs in the "esp_bt_hid_events" task defined in bt_hidh.c. It is possible to cause this overflow with a keyboard, but it is much more noticeable on a mouse where rapid mouse movements result in a crash and reboot of the ESP32.

### Expected Behavior

The bluetooth HID host example can handle rapid mouse movements and keyboard inputs without crashing.

### Actual Behavior

There is a stack overflow in the "esp_bt_hidh_events" task on rapid events that crashes the device.

### Steps to reproduce

1. Build the esp_hid_host example and flash it to an ESP32
2. Pair a bluetooth HID device
3. Give rapid inputs such as fast typing or rapid mouse movement

// If possible, attach a picture of your setup/wiring here.

It's just an ESP32-DEVKIT-V1 connected to my laptop via USB. Nothing else is connected.

### Code

The following code defines this task, starting at line 319 of bt_hidh.c:
```cpp
    esp_event_loop_args_t event_task_args = {
        .queue_size = 5,
        .task_name = "esp_bt_hidh_events",
        .task_priority = uxTaskPriorityGet(NULL),
        .task_stack_size = 2048,
        .task_core_id = tskNO_AFFINITY
    };
```

Setting the stack size to a higher value mitigates this issue. I have only experimented with 2048*10, which is probably overkill.
Notably, this task is defined in the function `esp_err_t esp_bt_hidh_init(const esp_hidh_config_t *config)`, which takes a config struct. This struct currently only contains a callback handler, but could be modified to take an (optional) stack size value.

## Debug Logs

```
...
W (35185) BT_HIDH: Wrong Data Len: 6 != 5
I (35195) ESP_HIDH_DEMO: 00 fa 2f 00 00 
I (35195) ESP_HIDH_DEMO: 5c:f6:dc:23:88:6d INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
W (35205) BT_HIDH: Wrong Data Len: 6 != 5
I (35205) ESP_HIDH_DEMO: 00 fe 18 00 00 

***ERROR*** A stack overflow in task esp_bt_hidh_eve has been detected.

Backtrace:0x4008e64f:0x3ffd5ad0 0x4008ecf9:0x3ffd5af0 0x4008efa1:0x3ffd5b10 0x4008fc0e:0x3ffd5b90 0x4008f094:0x3ffd5bb0 0x4008f04a:0x00000000 |<-CORRUPTED
0x4008e64f: panic_abort at C:/Users/Anon/Git/esp-idf/components/esp_system/panic.c:330

0x4008ecf9: esp_system_abort at C:/Users/Anon/Git/esp-idf/components/esp_system/system_api.c:106

0x4008efa1: vApplicationStackOverflowHook at C:/Users/Anon/Git/esp-idf/components/freertos/xtensa/port.c:436

0x4008fc0e: vTaskSwitchContext at C:/Users/Anon/Git/esp-idf/components/freertos/tasks.c:2770

0x4008f094: _frxt_dispatch at C:/Users/Anon/Git/esp-idf/components/freertos/xtensa/portasm.S:431

0x4008f04a: _frxt_int_exit at C:/Users/Anon/Git/esp-idf/components/freertos/xtensa/portasm.S:231

ELF file SHA256: 1e5de788efcddbb9

Rebooting...
```

## Other items if possible

- [  [sdkconfig.copy.txt](https://github.com/espressif/esp-idf/files/5794540/sdkconfig.copy.txt) ] sdkconfig
- [  [esp_hid_host.zip](https://github.com/espressif/esp-idf/files/5794543/esp_hid_host.zip) ] elf file in the ``build`` folder
- [ I can provide this later if necessary ] coredump (This provides stacks of tasks.)

LMBernardo, 2021-01-11:
I have created PR #6385 which aims to address this issue.

Alvin1Zhang, 2021-01-12:
Thanks for reporting and contributing in https://github.com/espressif/esp-idf/pull/6385, we will look into.

demik, 2021-04-18:
@LMBernardo I confirm that after testing your fix in #6385 with ESP-IDF v4.2-408-gc9cf7bcb0-dirty resolves the issue for me (I was getting reboot every few seconds with a mouse before).

Thanks for your work !

Alvin1Zhang, 2021-06-08:
Thanks for reporting and sharing updates, the fix is available at https://github.com/espressif/esp-idf/commit/67b0e0389fc09b4d4f9e17aaf802687513e0b316, feel free to reopen.

closed 2021-06-08 by Alvin1Zhang