espressif/esp-idf #6381: Flash external flash using LittleFS

tsctrl, 2021-01-11:
Hi,

Recently i discovered that spiffs read speed for external flash are a bit slow for my usage. So i am doing some poc if littlefs fix the issue but having roadblock to proceed.
I successfully flash the internal flash of my esp32 bin file but not be able to flash external flash using the generated bin using mklittlefs. I am successfully doing the same using mkspiffs though. 

`mklittlefs.exe -c data --size 16777216 --page 256 --block 4096 ext-spiffs.bin`

`esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5`
`esptool.py v3.1-dev`
`Serial port com6`
`Connecting........_`
`Chip is ESP32-D0WDQ6 (revision 1)`
`Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None`
`Crystal is 40MHz`
`MAC: 24:6f:22:c3:a3:54`
`Uploading stub...`
`Running stub...`
`Stub running...`
`Configuring SPI flash mode...`
`Configuring flash size...`
`Auto-detected Flash size: 16MB`
`Compressed 16777216 bytes to 499494...`
`Writing at 0x00004000... (6 %)`
`A fatal error occurred: Timed out waiting for packet header`

any help for me to investigate further or any support by espressif for little fs?

dobairoland, 2021-01-11:
Hi @tsctrl. Please try again with the latest version from https://github.com/espressif/esptool or the `--no-compress` argument.

projectgus, 2021-01-11:
(To give a bit of extra context:

The issue is probably that the littlefs bin file is very compressible, and the timeout for a single compressed packet is too short for the amount of uncompressed data that it contains. The esptool master branch contains a fix for this that was recently added, but we haven't add it to ESP-IDF yet. @dobairoland proposes either pulling the latest master commit of esptool (which should set the timeout correctly), or passing the --no-compress argument which will be slower to flash but works around the compression issue.)

projectgus, 2021-02-09:
Hi @tsctrl,

Did you get a chance to retry this? It should work with the latest master branch of ESP-IDF and/or esptool.

Angus

tsctrl, 2021-02-16:
Hi All, 

ya. need to use --no-compress to flash successfully. i am not really agree with the "very compressible" part. 
i have to wait few minutes for the flash to complete.? 

no chance to retest for now on the latest esptool/IDF. need to revisit the POC. will update soon

tsctrl, 2021-02-17:
Still having issue:

without --no-compress
`
esptool --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5
esptool.py v3.0
Serial port com6
Connecting........_____....._____..
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Compressed 16777216 bytes to 1456042...
Wrote 16777216 bytes (1456042 compressed) at 0x00000000 in 196.2 seconds (effective 684.2 kbit/s)...
File  md5: ad7b3998fcc208ba1a152a49b5a75f49
Flash md5: a0060da309921f9f2ce2f71dc4ab7ff2
MD5 of 0xFF is d5929b7a5ad99f511fc83d7d0b48b85f

A fatal error occurred: MD5 of file does not match data in flash!
`

with --no-compress success flashing in around 30 minutes 

`
esptool --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5 --no-compress
esptool.py v3.0
Serial port com6
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Wrote 16777216 bytes at 0x00000000 in 1521.6 seconds (88.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

dobairoland, 2021-02-17:
@tsctrl You are using now an even older version of esptool than in the first report. Please clone the https://github.com/espressif/esptool repo and try the latest master from there. The related change is very new and haven't been released yet.

tsctrl, 2021-02-18:
master branch:

esptool --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5
'esptool' is not recognized as an internal or external command,
operable program or batch file.

C:\Development\jam-waktu-solat\esp32-jws-espressif-idf>esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5
esptool.py v3.1-dev
Serial port com6
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Compressed 16777216 bytes to 1456042...
Writing at 0x00ae0fe7... (49 %)
A fatal error occurred: Timed out waiting for packet header

tsctrl, 2021-02-18:
to add more:
now the write checksum does not match 

```
esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5 --no-compress
esptool.py v3.1-dev
Serial port com6
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Wrote 16777216 bytes at 0x00000000 in 1492.3 seconds (89.9 kbit/s)...
File  md5: ad7b3998fcc208ba1a152a49b5a75f49
Flash md5: d2f5cd7676a765ea9adf5f9c4d462b54
MD5 of 0xFF is d5929b7a5ad99f511fc83d7d0b48b85f

A fatal error occurred: MD5 of file does not match data in flash!
````

tsctrl, 2021-02-18:
when i try to use mkspiffs to create bin files i am getting write issue.

mkspiffs.exe -c data -s 16777216 -p 256 -b 4096 ext-spiffs.bin
```
/web/assets/lib/i18n/locale/en.json
SPIFFS_write error(-10010): unknown
error adding file!
Error for adding content from locale!
```

i think this is because the folder length name to long. Is there a way to resolve this issue? do fatfs can resolve and do fatfs now supported in esptools? @petermccloud @projectgus

tsctrl, 2021-02-18:
tried using spiffsgen

```
spiffsgen.py --page-size 256 --block-size 4096 --obj-name-len 128 16777216 C:\Development\jam-waktu-solat\esp32-jws-espressif-idf\main ext-spiffs.bin
```
with the same issue:

```
esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5
esptool.py v3.1-dev
Serial port com6
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Compressed 16777216 bytes to 47404...
Wrote 16777216 bytes (47404 compressed) at 0x00000000 in 142.2 seconds (effective 943.9 kbit/s)...
File  md5: 4ae326e5ea9422b98e59682cb365faae
Flash md5: eae0cefe4a74fd42e447b0b00563d4ea
MD5 of 0xFF is d5929b7a5ad99f511fc83d7d0b48b85f
```

tsctrl, 2021-02-19:
Hi @dobairoland , 

i made a little progress today.  for the flash issue it is actually because of bad cabling. the fix is actually works and the flash are faster without --no-compress.
below is the summary of the flashing logs:

```
esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5
esptool.py v3.1-dev
Serial port com6
Connecting........_____.
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:6f:28:c3:a5:e4
Uploading stub...
Running stub...
Stub running...
Configuring SPI flash mode...
Configuring flash size...
Compressed 16777216 bytes to 1458781...
Wrote 16777216 bytes (1458781 compressed) at 0x00000000 in 201.4 seconds (effective 666.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

i actually end up buying internal 16MB chips from espressif today, will still using little fs as it 3 time faster than the spiffs and will use fatfs with sd for future proofing. closing this issue.

Thank you for you time and support!

closed 2021-02-19 by tsctrl