espressif/esp-idf #6382: BLE-MESH Vendor Callbacks

MrCurio, 2021-01-11:
**Is your feature request related to a problem? Please describe.**

Hello, 
I am currently developing an application based on Bluetooth Mesh and i have a few recommendations i would like you to hear to simplify development and programming with regards to Vendor Models.

**Describe the solution you'd like**

Currently, SIG models init and deinit their correspondent _model_ and _user_data_  by means of stack initialized callbacks. 
```
typedef struct {
    /** Callback used during model initialization. Initialized by the stack. */
    esp_ble_mesh_cb_t init_cb;

    /** Callback used during model deinitialization. Initialized by the stack. */
    esp_ble_mesh_cb_t deinit_cb;
} esp_ble_mesh_model_cbs_t;
```

However, Vendor Models are not allowed (or do not use) to access this callbacks. Therefore, Vendor Models _user_data_ must be deinit manually which may lead to Memory leaks. I'd propose to create some vendor model only API to register these callbacks into the Stack. This way, the deinit of the Bluetooth Mesh Stack will also help freeing user allocated memory.

In addition, i would also like to have separated messages callbacks for different vendor models. Currently, all Vendor Messages arrive to the same callback where messages should be filtered and linked to their correspondent response function. I'd propose an API call similar to the one below. This way, code understanding gets reduced.
```
//Propose API function call
esp_ble_mesh_register_custom_model_callback(esp_ble_mesh_model_t *vendor_model, void (*msg_cb_vendor_function)(esp_ble_mesh *param, ...));
```

**Additional context**

If there already exist features similar to the ones i explained, i would like to have the latest patch you have. It would be really interesting for my project.

Thanks!

Alvin1Zhang, 2021-01-11:
Thanks for raising this feature request.

Campou, 2021-01-11:
Hi @MrCurio 

Thanks for this.

Could this be achieved by initializing the `model->cb` when defining a vendor model? Then no need to add a new API for this.

Currently `ESP_BLE_MESH_VENDOR_MODEL` is used to define a vendor model, maybe we could add a new macro, e.g. `ESP_BLE_MESH_VENDOR_MODEL_CB` for defining a vendor model with the corresponding callbacks registered. And users could choose to use either of them to define a vendor model based on their requirements.

Thanks.

MrCurio, 2021-01-11:
Hi @Campou 
That would be great. I wish i could help developing those features, bit I'm afraid i don't have the time.
If anyone plans to develop it, please let me know with a comment here.

Thanks!

Campou, 2021-01-25:
Hi @MrCurio 

That's fine, thanks for the idea!

I will mark this as a feature in the next release version, maybe for `idf 4.4`, and I will try to implement this before the release.

Thanks.