espressif/esp-idf #6383: ADC values inverted when read out with ULP

ctag-fh-kiel, 2021-01-11:
In release/v4.2 branch ADC values are inverted when read out with ULP.
This may also be the case when reading using standard ADC API, I have not tested it.
Seems that there was an issue already like this but it is still persisting in release/v4.2

Alvin1Zhang, 2021-01-12:
@ctag-fh-kiel Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ctag-fh-kiel, 2021-01-12:
## Environment
- Module or chip used:  ESP32-WROVER-IB, 16MB Flash, 4MB SPIRAM
- IDF version release/v4.2
- Build System: idf.py
- Compiler version: Default of release/v4.2
- Operating System:     [Windows|macOS]
- Power Supply:        external 3.3V

## Problem Description

adc 1 readout using ULP is inverted

### Expected Behavior
not inverted with release/v4.1 branch

### Actual Behavior
inverted with release/v4.2 branch

### Code to reproduce this issue

check the ULP and driver code here:
https://github.com/ctag-fh-kiel/ctag-tbd/blob/master/components/drivers/adc.cpp
https://github.com/ctag-fh-kiel/ctag-tbd/blob/master/components/drivers/ulp/adc.S

## Other items if possible
 sdkconfig here:
https://github.com/ctag-fh-kiel/ctag-tbd/blob/master/sdkconfig.defaults