espressif/esp-idf #6384: Renaming and moving main component and adding a component makes main component not compile anymore

TobiasBoeschMiele, 2021-01-11:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used:  ESP32-WROOM-32
- IDF version
    v4.3-dev-2398-g2bfdd036b
- Build System:         CMake and idf.py
- Compiler version:
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes eclipse plugin
- Power Supply:         USB

## Problem Description

When the main component is renamed (moved to a subfolder) and another component is created, the main component is not compiled anymore resulting in ```undefined reference to `app_main'```

### Expected Behavior
main is compiled and `app_main` function is found

### Actual Behavior
main is not compiled and `app_main` function is not found

### Steps to reproduce

1. Create an empty project using eclipse "New esspressif project" wizard
2. Compile and find it compiles fine
3. Move main component to `source/startup` and [rename the main component according to the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#renaming-main-component)
4. Comile and check that it it compiles fine
5. Add another component to `source/testComponent`
6. Compile and see it doesn't compile

I've created a test project in a git repository that holds the history made in the steps mentioned above: 

### Code to reproduce this issue

[renamingMainAddingComponent.zip](https://github.com/espressif/esp-idf/files/5795803/renamingMainAddingComponent.zip)

## Debug Logs

build log is in the attached git repository.

## Other items if possible

- [X ] sdkconfig file (attach the sdkconfig file from your project folder) (In the attached git repo)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

TobiasBoeschMiele, 2021-01-11:
If an additional component is created while the main component is located in the standard location: `projectDir/main` the main component compiles fine.

Alvin1Zhang, 2021-01-12:
Thanks for reporting.

projectgus, 2021-01-12:
Hi @TobiasBoeschMiele,

Thanks for the comprehensive explanation of the problem, and the test project.

In CMake, EXTRA_COMPONENT_DIRS is a variable. It's set in the CMakeLists.txt file, and then read inside the implementation of `project`.

So when doing this in the top-level CMakeLists.txt file:

```
set(EXTRA_COMPONENT_DIRS "./source/startup")                                                                                         
set(EXTRA_COMPONENT_DIRS "./source/testComponent")                                                                                   
```

The second statement replaces the value of `EXTRA_COMPONENT_DIRS` with the new value, the old value is not saved anywhere.

You can set the variable to a list of values in multiple ways. For example, this can work:

```
set(EXTRA_COMPONENT_DIRS "./source/startup" "./source/testComponent")                                                                                   
```                                                                     

However, would recommend doing this instead:

```
set(EXTRA_COMPONENT_DIRS "source")
```

The build system will automatically look in every subdirectory of `source` for a component CMakeLists.txt file.

TobiasBoeschMiele, 2021-01-12:
Thanks @projectgus not overwriting previous values solves the problem.
The reason I used `EXTRA_COMPONENT_DIRS` that way might be that `REQUIRES` behaves that way.

```
REQUIRES foo
REQUIRES bar
```

closed 2021-01-12 by TobiasBoeschMiele