espressif/esp-idf #6387: ESP32-S2 idf building in eclipse with submodules

wojciech-angelo, 2021-01-12:
## Environment
- Development Kit:      [ESP32-S2-WROOM]
- Module or chip used:  [ESP32-S2-WROOM]
- IDF version (run ``git describe --tags`` to find it):  v4.3-dev-2399-g4206a0422
- Build System:        Official eclipse with plugin ESP-IDF Eclipse Plugin (Version: 1.2.3.202010091505)
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (esp-2020r3 8.4.0?)
- Operating System:     [Windows 10 Pro] 
- (Windows only) environment type: [eclipse built].
- Using an IDE?: [Yes] Eclipse IDE for C/C++ Developers (includes Incubating components)

>Version: 2020-09 (4.17.0)
>Build id: 20200910-1200

- Power Supply:         [problem with building idf...]

## Problem Description
Hello I've got a big problem with my eclipse IDE and esp-idf (latest version, yesterday fetching).
I cannot build program with Eclipse (this program works my colleague with Linux and vscode installed - I preffer eclipse because some months ago I successfully run DEBUGGER with this board)

First of all: I need USB component program so I need to update idf to latest. I done it but I have strange logs in "git status" command:

```
$ git status 
On branch master
Your branch is ahead of 'origin/master' by 1 commit.
  (use "git push" to publish your local commits)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   components/asio/asio (modified content)
        modified:   components/bootloader/subproject/components/micro-ecc/micro-ecc (modified content)
        modified:   components/bt/controller/lib (modified content)
        modified:   components/bt/host/nimble/nimble (modified content)
        modified:   components/cbor/tinycbor (modified content)
        modified:   components/coap/libcoap (modified content)
        modified:   components/esp_wifi/lib (modified content)
        modified:   components/esptool_py/esptool (modified content)
        modified:   components/expat/expat (modified content)
        modified:   components/json/cJSON (modified content)
        modified:   components/libsodium/libsodium (modified content)
        modified:   components/lwip/lwip (modified content)
        modified:   components/mbedtls/mbedtls (modified content)
        modified:   components/mqtt/esp-mqtt (modified content)
        modified:   components/nghttp/nghttp2 (modified content)
        modified:   components/protobuf-c/protobuf-c (modified content)
        modified:   components/tinyusb/tinyusb (modified content)
        modified:   components/unity/unity (modified content)
        modified:   examples/build_system/cmake/import_lib/main/lib/tinyxml2 (modified content)
        modified:   examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (mo

no changes added to commit (use "git add" and/or "git commit -a") 
```
and

`git  submodule update --init --recursive`

do nothing...

Second when I build my project (I click on build icon in eclipse IDE) everything is compiled (with some warnings - normal) but building stops when linker starts work:

OUTPUT with error:
```
[2/3] cmd.exe /C "cd . && D:\Programy\Eclipse_CPP\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\eink-esp-proj.elf.rsp  -o eink-esp-proj.elf  && cd ."
FAILED: eink-esp-proj.elf 
cmd.exe /C "cd . && D:\Programy\Eclipse_CPP\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\eink-esp-proj.elf.rsp  -o eink-esp-proj.elf  && cd ."
d:/programy/eclipse_cpp/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(phy_init.c.obj):(.literal.esp_phy_load_cal_and_init+0x10): undefined reference to `get_phy_version_str'
d:/programy/eclipse_cpp/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(phy_init.c.obj): in function `esp_phy_load_cal_and_init':
D:/Programy/esp-idf-v4.2-rc/components/esp_wifi/src/phy_init.c:474: undefined reference to `get_phy_version_str'
d:/programy/eclipse_cpp/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj):(.literal.wpa_supplicant_install_ptk+0x4): undefined reference to `esp_wifi_get_sta_hw_key_idx_internal'
d:/programy/eclipse_cpp/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `wpa_supplicant_install_ptk':
D:/Programy/esp-idf-v4.2-rc/components/wpa_supplicant/src/rsn_supp/wpa.c:688: undefined reference to `esp_wifi_get_sta_hw_key_idx_internal'
d:/programy/eclipse_cpp/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/wpa_supplicant/libwpa_supplicant.a(wpa.c.obj): in function `wpa_supplicant_install_gtk':
D:/Programy/esp-idf-v4.2-rc/components/wpa_supplicant/src/rsn_supp/wpa.c:824: undefined reference to `esp_wifi_get_sta_hw_key_idx_internal'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### Expected Behavior
I need to build project in eclipse...
### Actual Behavior
Building stops on linking...
### Steps to reproduce

 as mention above... (Install eclipse, install plugin, update idf to latest, do a hello world, build everything together...

### Code to reproduce this issue
any Hello world example in eclipse...

Please help me because this is very important project in my current job...
Greetings

wojciech-angelo, 2021-01-13:
I tried linux and the same problem. do you have any ideas?

Windoofs, 2021-01-14:
Hi,
got the error `/components/esp_wifi/src/phy_init.c:474: undefined reference to `get_phy_version_str'` as well. I just re-run `git submodule update --init --recursive`  and did a `fullclean`. This fixed it for me at least.

projectgus, 2021-01-22:
Hi @wojciech-angelo ,

Sorry for the slow response. It looks like your git directory has gotten into a bad state somehow. What method did you use to update it?

Cloning ESP-IDF from github again may be the quickest way to get a working copy of the master branch.

wojciech-angelo, 2021-01-26:
> Hi @wojciech-angelo ,
> 
> Sorry for the slow response. It looks like your git directory has gotten into a bad state somehow. What method did you use to update it?
> 
> Cloning ESP-IDF from github again may be the quickest way to get a working copy of the master branch.

Ok, I handle this...
thanks...

projectgus, 2021-02-09:
@wojciech-angelo Thanks for letting me know. Is the issue resolved now?

demik, 2021-03-22:
> got the error `/components/esp_wifi/src/phy_init.c:474: undefined reference to `get_phy_version_str'`as well. I just re-run`git submodule update --init --recursive` and did a`fullclean`. This fixed it for me at least.

Same error here after pulling from release/v4.2. Running `git submodule update --init --recursive` and a fullclean resolves the issue.

rustyx, 2021-06-14:
This happens if you do a `git pull` without updating submodules or if updating submodules fails due to a network timeout, etc.

If `git status` shows any changes in submodule folders, then `git submodule update --init --recursive`  should resolve it.

Alvin1Zhang, 2021-06-15:
@Windoofs Thanks for reporting, would you please help share if any updates for this issue? Thanks.

Windoofs, 2021-06-15:
The error has not occurred since then.

projectgus, 2021-07-13:
Closing as it seems like this is resolved.

closed 2021-07-13 by projectgus