espressif/esp-idf #6392: Problem with installation latest esp-s2 idf on new clean ubuntu installation...

wojciech-angelo, 2021-01-13:
I cannot properly configure esp-idf. I do everything like in Getting started (https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/linux-setup.html):
Firstly: needed packages, then newest python, then just like said in instruction cloning esp-idf into clean ubuntu disk.

I stuck on install.sh script. 
Output:
````
Installing ESP-IDF tools
/usr/bin/env: "python": Nie ma takiego pliku ani katalogu 
````

I do an alias from python3 to python. But effect is the same.
Please help, because this is your product your website, and your responsible...

chegewara, 2021-01-13:
Maybe this will help:
https://github.com/espressif/esp-idf/issues/5908#issuecomment-701517116

wojciech-angelo, 2021-01-13:
> Maybe this will help:
> [#5908 (comment)](https://github.com/espressif/esp-idf/issues/5908#issuecomment-701517116)

Yes I found the solution here:
https://askubuntu.com/questions/320996/how-to-make-python-program-command-execute-python-3

exactly this command:
`$ sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10`

but, thank you @chegewara for a fast reply :)

closed 2021-01-14 by wojciech-angelo

dobairoland, 2021-01-14:
I'm glad that this got solved. Please note that this information is also available in the getting started guide of ESP-IDF: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#setting-up-python-3-as-default-for-ubuntu-and-debian

wojciech-angelo, 2021-01-14:
ok thank you :)