espressif/esp-idf #6394: lwIP: Server socket remains open for 120s after closing

CharlesSeartech, 2021-01-13:
## Environment

- Development Kit:      ESP32-Wrover
- Kit version (for WroverKit): [v1]
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):    // v4.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):    // (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: Yes, but only for editing: eclipse Version: 2020-12 (4.18.0))
- Power Supply:         USB, battery or external 5V

## Problem Description

I have  problem with closing a server socket on the ESP32 and then having to wait 120s before I can re-open the socket again.

### Expected Behavior

If a server socket is closed on the ESP32 then I should be able to reopen it again anytime after closing.

### Actual Behavior

When I close a server socket on the ESP32 and I try to reopen a socket within 120s I get an error "bind failed: Address already in use".
If the client closes the socket first and then the ESP32 closes the sockets then all works correctly and I can re-open the server socket and bind without having to wait the 120s.

Application: When the user requests the log file I start a new task and open a server socket using the following sequence:
Using tcp_server.c example code to reproduce the problem.
```cpp
    // Open socket
    int listen_sock = socket(addr_family, SOCK_STREAM, ip_protocol);
    int err = bind(listen_sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
    err = listen(listen_sock, 1);
    int sock = accept(listen_sock, (struct sockaddr *)&source_addr, &addr_len);
```
Once the socket has connected I upload the log file and close the socket:
```cpp
    // Close socket
    shutdown(sock, SHUT_RDWR);
    close(sock);
    close(listen_sock);
    ...
    vTaskDelete(NULL);
```
When the tcp_server_task() is restarted within 120s the bind request fails, "bind failed: Address already in use".

### Steps to reproduce

1. Using espV4.1\esp-idf\examples\protocols\sockets\tcp_server\main\tcp_server.c, attached.
[tcp_server.c.txt](https://github.com/espressif/esp-idf/files/5808783/tcp_server.c.txt)
2. Compile and flash ESP32.
3. Using example_test.py or any other TCP client software, Hercules SETUP utility by https://www.hw-group.com as it has a send box to do quick testing.
[example_test.py.txt](https://github.com/espressif/esp-idf/files/5808947/example_test.py.txt)
4. Send "TestMessage Test" and close client socket on Hercules or example_test.py will close the socket after receiving the echo from the ESP32. The socket will close and loop around and call accept to accept the next connection. All as expected.
5. Send "TestMessage Restart" and close client socket on Hercules or example_test.py will close the socket after receiving the echo from the ESP32. The socket will close, the listen_sock will be closed, due to bTest flag, and the task deleted. The main loop will restart the tcp_server_task(), due to bTestEnd flag, and a new listen_sock will bind to the port number, 6001 in this case, and wait on accept for the next connection.  All as expected.
6. Send "TestMessage Close" and the ESP32 will close the server socket, due to the "Close"/bTest flag, Hercules or example_test.py will close due to the FIN ACK the socket after receiving the echo from the ESP32. The task will be deleted and the main loop will restart the tcp_server_task(). A new listen_sock will be acquired but when we try to bind we get the error "bind failed: Address already in use". I then retry every 5s and after 120s the bind will be successful.

In my development I have seen this issue before on port 80 before but because the client almost always closed the connection it was not an issue. Now that I wish to close the server socket and reopen it within 120s there seems to be a problem. I have tried calling shutdown on the listen_sock but it generates an error "shutdown failed: Invalid argument", so that isn't the issue. I have tried adding delays when closing the sock and listen_sock but that doesn't help.

Can you help please?

### Code to reproduce this issue

See attached "tcp_server.c.txt".

## Debug Logs

```
I (490) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 100 lines cut ...]
...
I (168724) example: Restarting tcp_server_task
I (168724) example: Socket created
**E (168725) example: Socket unable to bind: errno 112**
bind failed: Address already in use
I (168731) example: CLEAN_UP
I (169734) example: CLEAN_UP 2
I (169824) example: Wait 5000
I (174824) example: Restarting tcp_server_task
I (174824) example: Socket created
**I (174825) example: Socket bound, port 6001
I (174827) example: Timeout = 123051**
I (174831) example: Socket listening
I (1007464) example: Socket accepted ip address: 192.168.3.124
I (1007469) example: Received 15 bytes: TestMessge Test
W (1008583) example: Connection closed
I (1008584) example: do_retransmit end
I (1008584) example: Close Socket
I (1008587) example: Timeout = 2
I (1008589) example: Socket listening
```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5809032/sdkconfig.txt)

Alvin1Zhang, 2021-01-14:
@CharlesSeartech Thanks for the very detailed reporter and letting us know, we will look into.

CharlesSeartech, 2021-01-18:
Good day,

I solved the problem.
It's a TIME_WAIT issue that the socket stays open for TIME_WAIT time before allowing bind again. You can get around it with SO_REUSEADDR.
https://hea-www.harvard.edu/~fine/Tech/addrinuse.html

Regards,
Charles

david-cermak, 2021-01-21:
@CharlesSeartech Thanks for sharing the solution! 
Could you please close the issue?

closed 2021-01-21 by CharlesSeartech