espressif/esp-idf #6395: parallel builds

edbek, 2021-01-13:
1. Please, give advice on how to make a parallel builds ?
2. In addition, for some reason, compilation in Eclipse for the target ESP32-S2 does not work (works only ESP32) !

projectgus, 2021-01-14:
H @edbek,

Can you please give some more information about which ESP-IDF version you are using, how you are building your project, etc?

Some tips in the meantime:

* If you're using CMake (or the [ESP-IDF Eclipse Plugin](https://github.com/espressif/idf-eclipse-plugin/) which uses CMake on the backend) then you should be getting parallel builds already, automatically.
* You need to use ESP-IDF v4.2 or newer, and the CMake build system, in order to build for ESP32-S2.

edbek, 2021-01-14:
Thank you, everything works with ESP-IDF v4.2 !

projectgus, 2021-01-22:
Glad to hear!

closed 2021-01-22 by projectgus